esphome/issues #6268: W5500 SPI-based Ethernet without interrupt pin causes crash at init

slakichi, 2024-09-22:
### The problem

W5500 can build without interrupt_pin configuration, but actually, ESPHome firmware will setup W5500 ethernet driver with uninitialized interrupt pin config. This cause infinite crash at initialize ethernet component.
Confirmed with [M5Stack M5AtomS3 Lite](https://docs.m5stack.com/en/core/AtomS3%20Lite) with [M5Stack ATOMIC PoE Base W5500](https://docs.m5stack.com/en/atom/Atomic%20PoE%20Base)

### Which version of ESPHome has the issue?

2024.9.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

M5AtomS3 Lite

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
esphome:
  name: m5atoms3-lite
  platformio_options:
    board_build.f_cpu: 80000000L
    board_build.f_flush: 80000000L

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 8MB
  variant: esp32s3
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: "6.8.1"

network:
  enable_ipv6: False

ethernet:
  type: W5500
  clk_pin: GPIO5
  mosi_pin: GPIO8
  miso_pin: GPIO7
  cs_pin: GPIO6
  # W5500(SPI) optional pins
  #interrupt_pin:
  #reset_pin:
  clock_speed: 10MHz
  manual_ip:
    static_ip: 192.168.xxx.xxx
    gateway: 192.168.xxx.xxx
    subnet: 255.255.255.0
    dns1: 192.168.xxx.xxx

api:
  encryption:
    key: !secret home_assistant_api_password
  reboot_timeout: 300s

ota:
  platform: esphome
  password: !secret ota_password
[... 146 lines cut ...]
+  }

   esp_eth_mac_t *mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
 #else
@@ -314,7 +323,7 @@
   ESP_LOGCONFIG(TAG, "  MISO Pin: %u", this->miso_pin_);
   ESP_LOGCONFIG(TAG, "  MOSI Pin: %u", this->mosi_pin_);
   ESP_LOGCONFIG(TAG, "  CS Pin: %u", this->cs_pin_);
-  ESP_LOGCONFIG(TAG, "  IRQ Pin: %u", this->interrupt_pin_);
+  ESP_LOGCONFIG(TAG, "  IRQ Pin: %d", this->interrupt_pin_);
   ESP_LOGCONFIG(TAG, "  Reset Pin: %d", this->reset_pin_);
   ESP_LOGCONFIG(TAG, "  Clock Speed: %d MHz", this->clock_speed_ / 1000000);
 #else
--- ethernet_component.h        2024-09-22 14:35:21.067973882 +0900
+++ ethernet_component.h        2024-09-21 23:35:05.449671506 +0900
@@ -107,7 +107,7 @@
   uint8_t miso_pin_;
   uint8_t mosi_pin_;
   uint8_t cs_pin_;
-  uint8_t interrupt_pin_;
+  int interrupt_pin_{-1};
   int reset_pin_{-1};
   int phy_addr_spi_{-1};
   int clock_speed_;
```

ssieb, 2024-09-22:
I started to create a PR and then I noticed that you had already shown the required change.  Why don't you make a PR?

slakichi, 2024-09-22:
1. I am new to git
2. `poll_period_ms` should be configurable (but I'm hassle with it)

ivanduka, 2024-09-25:
@slakichi , it's very simple to create a PR, you'll need to fork the repo, create a new branch from main, do the changes and commit them, then push the branch to GitHub, create a new Pull Request and choose your branch as the source.

cortices, 2024-09-27:
I was just coming here to make the same change, @slakichi if you don't mind I am happy to open this PR?

slakichi, 2024-09-27:
Everyone seemed to want me to submit a PR, so I submitted a Draft, putting the document out of my mind.
... There seemed to be an error immediately.

cortices, 2024-09-28:
There's no error, it's just waiting for review :) Thanks!

closed 2024-10-29 by jesserockz