esphome/issues #6273: BL0942 Reports wrong energy accumulation KWh

yiannosi, 2024-09-24:
### The problem

I have two ESPhome modules with BL0942.
Both module correctly report voltage,current and frequency but the Energy register seems to increase exponentially.

[image]

### Which version of ESPHome has the issue?

2024.

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

theanch92, 2024-09-24:
I have the same issue, solved for now like this:

  energy:
    name: Energy   
    accuracy_decimals: 2
    filters:
      multiply: 0.000001

yiannosi, 2024-09-24:
Hmm..What makes you think its a simple matter of scaling?

theanch92, 2024-09-24:
I have made a comparison with the history, I have practically a small constant load, I thought it was just a multiplication problem but in fact it is higher by about 3 times even with the multiply: 0.000001. Today 1.21kwh, before update 0.35khw

Then I don't know, let's wait for someone to know more

yiannosi, 2024-09-26:
Update: A third BL0942 based ESPHome device been in operation for 2 months and was so far reporting energy correctly, now exhibits the same wrong energy.
I am now wondering if this has anything to do with any recent firmware / software upgrade.
Till a solution is found, I am now using an Integral Helper in HA to calculate Energy from Power..

yiannosi, 2024-09-28:
Update: I have a total of 4 EspHome devices with BL0942 three of which give wrong energy indication. The 4th one appears to be OK. I just noticed that the one that is ok is still on ESPHome firmware 24.9.0 while the rest were upgraded tro 4.9.1.
Just to check I have now reverted back to ESPHome firmware 4.8.3 by restoring a backup (didnt backup on 4.9.0), recompiled and uploaded one of the devices which for now looks to be recording energy correctly again.
So this points to a firmware issue that appeared probably in version 4.9.1.
Will continue monitoring and post updates.

Can anyone else confirm this ?

yiannosi, 2024-09-28:
I can now confirm that all ESPHome modules with BL0942 downgraded to firmware version 8.3 (as well as one with version 9.0) report energy correctly.

The bug was introduced with the latest version 9.1
If anyone can look into the code, it would be nice if we could have fixed by next release...

theanch92, 2025-01-14:
Sorry for the delay in providing an update. In my case, the issue turned out to be hardware-related. A 470uF capacitor failed, causing an unstable power supply that led to measurement errors and random reboots. However, in my case, the device is an "Aubess WiFi Smart Switch with Power Monitoring" featuring a BK7231N chip. I’ve already had to replace this capacitor in three devices! I’m not sure if the BL0942 is under increased stress with the latest updates, which might be contributing to the capacitor failures.

[image]