esphome/issues #6275: Addressable lights inside a "on_value" cause ESP32 to not boot

agl1tch3dpir8, 2024-09-25:
### The problem

I'm writing code that illuminates addressable LED's based on a number input. If I use "on_value_range" the code compiles, the ESP boots, and operates correctly. However, since I need 10 different LEDS, I cannot use "on_value_range" as it seems to be restricted to just 3 separate zones.

When I change the automation to "on_value," the code compiles and uploads, but then the ESP fails to boot. I can still upload new code by connecting via USB and using ESPWEB. No logs available because devices fails to boot.

### Which version of ESPHome has the issue?

2024.9.1 (Issue since 2024

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Display Backlight"
    id: back_light
    internal: true
    restore_mode: ALWAYS_ON
    default_transition_length: 1s
    effects:
      - pulse:
          name: "Armed Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
          min_brightness: 0%
          max_brightness: 100%
  - platform: rgb
    name: "Front Indicator LED"
    red: led_red
    green: led_green
    blue: led_blue
  - platform: esp32_rmt_led_strip
    id: ringlight
    rgb_order: GRB
    pin: 22
    num_leds: 12
    rmt_channel: 1
    chipset: ws2812
    name: "Light Ring"
    #max_refresh_rate: 1s
    effects:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 50

switch:
  - platform: template
    id: autoback
    name: "Automatic Backlight"
    optimistic: True
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_off: 
      then:
        - light.turn_on: 
            id: back_light
            brightness: "1"

number:
  - platform: template
    name: "Living Room Volume"
    min_value: 0
    max_value: 10
    step: 1
    optimistic: true
    on_value:
      then:
        - light.addressable_set:
            id: ringlight
            range_from: 0
            range_to: 10
            red: 0%
            green: 0%
            blue: 0%
```

### Anything in the logs that might be useful for us?

```txt
Device is a DIYmall's ESP32 with a 2.4" display. Found on Amazon here: https://www.amazon.com/dp/B0CLGD2DG6?ref=ppx_yo2ov_dt_b_fed_asin_title
```

ssieb, 2024-09-27:
`on_value_range` can have as many ranges as you want.  The reason it works though is because it won't trigger on the first value since the value has to change from outside to inside the range.  `on_value` is trying to control the light before it's initialized.
Try adding `setup_priority: 700` to the number.

agl1tch3dpir8, 2024-09-27:
Oh wow thank you! I didn't know about the `setup_priority` but that was exactly what I needed! Closing as not a issue.

closed 2024-09-27 by agl1tch3dpir8