esphome/issues #6276: BLE_rssi now only reports "nan" instead of value

AndersRipa, 2024-09-26:
### The problem

I have been using ESP32 and ESPHome and its esp32_ble_tracker and sensors ble_rssi and ble_presence since august 16 this year and both have worked OK. I have the habit of upgrading the ESPHome integration and recompiling the ESP32s when new releases becomes available. Starting on the evening of 19 september the ble_rssi stopped working and as of yesterday all the 6+ BLE ESP32 have stopped reporting ble_rssi values and only giving the value "nan". The last node that worked stopped reporting after it was updated to 2024.9.1 (not sure what version was on it before that). Please note that ble_presence have been and is still working correctly its only the ble_rssi value that have stopped working.

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

AZDelivery 5 x ESP32 Dev Kit C V4 NodeMCU WLAN/WiFi Development Board

### Component causing the issue

ble_rssi

### Example YAML snippet

```yaml
substitutions:
  name: esphome-web-e62c70
  friendly_name: Test BLE e62c70
  
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: ripa.BLE_tracker
    version: 1.0.0

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

# Example configuration entry for finding
# Service UUIDs and iBeacon UUIDs and identifiers
esp32_ble_tracker:

binary_sensor:
    # Presence based on iBeacon UUID
    - platform: ble_presence
      ibeacon_uuid: '6a3d4446-2467-4bc1-8458-e3a63b09ee97'
      #ibeacon_uuid: '$(mobil_anders)'
      name: "Anders Mobil BLE"

sensor:
    # RSSI based on iBeacon UUID
    - platform: ble_rssi
      #ibeacon_uuid: $(mobil_anders)
      ibeacon_uuid: '6a3d4446-2467-4bc1-8458-e3a63b09ee97'
      name: "Anders Mobil BLE RSSI"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/esphome-web-e62c70.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-e62c70 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/esphome-web-e62c70/src/main.cpp.o
Linking .pioenvs/esphome-web-e62c70/firmware.elf
RAM:   [==        ]  16.1% (used 52904 bytes from 327680 bytes)
Flash: [========= ]  86.8% (used 1592973 bytes from 1835008 bytes)
Building .pioenvs/esphome-web-e62c70/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esphome-web-e62c70/firmware.bin"], [".pioenvs/esphome-web-e62c70/firmware.elf"])
Wrote 0x196520 bytes to file /data/build/esphome-web-e62c70/.pioenvs/esphome-web-e62c70/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/esphome-web-e62c70/firmware.bin"], [".pioenvs/esphome-web-e62c70/firmware.elf"])
========================= [SUCCESS] Took 98.91 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.107
INFO Uploading /data/build/esphome-web-e62c70/.pioenvs/esphome-web-e62c70/firmware.bin (1598752 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 19.16 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.107 using esphome API
INFO Successfully connected to esphome-web-e62c70 @ 192.168.0.107 in 4.111s
INFO Successful handshake with esphome-web-e62c70 @ 192.168.0.107 in 0.016s
[10:26:10][I][app:100]: ESPHome version 2024.9.1 compiled on Sep 26 2024, 10:24:15
[10:26:10][I][app:102]: Project ripa.BLE_tracker version 1.0.0
[10:26:10][C][wifi:600]: WiFi:
[10:26:10][C][wifi:428]:   Local MAC: D8:BC:38:E6:2C:70
[10:26:10][C][wifi:433]:   SSID: 'RIPA_privat'[redacted]
[10:26:10][C][wifi:436]:   IP Address: 192.168.0.107
[10:26:10][C][wifi:440]:   BSSID: 50:46:5D:5E:07:70[redacted]
[10:26:10][C][wifi:441]:   Hostname: 'esphome-web-e62c70'
[10:26:10][C][wifi:443]:   Signal strength: -43 dB ▂▄▆█
[10:26:10][C][wifi:447]:   Channel: 11
[10:26:10][C][wifi:448]:   Subnet: 255.255.255.0
[10:26:10][C][wifi:449]:   Gateway: 192.168.0.1
[10:26:10][C][wifi:450]:   DNS1: 192.168.0.1
[10:26:10][C][wifi:451]:   DNS2: 0.0.0.0
[10:26:10][C][logger:185]: Logger:
[10:26:10][C][logger:186]:   Level: DEBUG
[10:26:10][C][logger:188]:   Log Baud Rate: 115200
[10:26:10][C][logger:189]:   Hardware UART: UART0
[10:26:10][C][ble_presence:011]: BLE Presence 'Anders Mobil BLE'
[10:26:10][C][ble_rssi:011]: BLE RSSI Sensor 'Anders Mobil BLE RSSI'
[10:26:11][C][ble_rssi:011]:   Device Class: 'signal_strength'
[10:26:11][C][ble_rssi:011]:   State Class: 'measurement'
[10:26:11][C][ble_rssi:011]:   Unit of Measurement: 'dBm'
[10:26:11][C][ble_rssi:011]:   Accuracy Decimals: 0
[10:26:11][C][esp32_ble:391]: ESP32 BLE:
[10:26:11][C][esp32_ble:393]:   MAC address: D8:BC:38:E6:2C:72
[10:26:11][C][esp32_ble:394]:   IO Capability: none
[10:26:11][C][esp32_ble_server:200]: ESP32 BLE Server:
[10:26:11][C][esp32_improv.component:261]: ESP32 Improv:
[10:26:11][C][esp32_ble_tracker:655]: BLE Tracker:
[10:26:11][C][esp32_ble_tracker:656]:   Scan Duration: 300 s
[10:26:11][C][esp32_ble_tracker:657]:   Scan Interval: 320.0 ms
[10:26:11][C][esp32_ble_tracker:658]:   Scan Window: 30.0 ms
[10:26:11][C][esp32_ble_tracker:659]:   Scan Type: ACTIVE
[10:26:11][C][esp32_ble_tracker:660]:   Continuous Scanning: True
[10:26:11][C][captive_portal:089]: Captive Portal:
[10:26:11][C][web_server:145]: Web Server:
[10:26:11][C][web_server:146]:   Address: esphome-web-e62c70.local:80
[10:26:11][C][mdns:116]: mDNS:
[10:26:11][C][mdns:117]:   Hostname: esphome-web-e62c70
[10:26:11][C][esphome.ota:073]: Over-The-Air updates:
[10:26:11][C][esphome.ota:074]:   Address: esphome-web-e62c70.local:3232
[10:26:11][C][esphome.ota:075]:   Version: 2
[10:26:11][C][safe_mode:018]: Safe Mode:
[10:26:11][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:26:11][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:26:11][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:26:11][C][api:139]: API Server:
[10:26:11][C][api:140]:   Address: esphome-web-e62c70.local:6053
[10:26:11][C][api:144]:   Using noise encryption: NO
[10:26:11][C][improv_serial:032]: Improv Serial:
[10:26:35][D][binary_sensor:036]: 'Anders Mobil BLE': Sending state ON
[10:27:06][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[10:27:06][D][esp32.preferences:114]: Saving 1 preferences to flash...
[10:27:06][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:31:07][D][esp32_ble_tracker:270]: Starting scan...
[10:31:07][D][sensor:094]: 'Anders Mobil BLE RSSI': Sending state nan dBm with 0 decimals of accuracy
```

### Additional information

Please let me know what other information is needed to debug this issue. I can supply other files from the system, but I have only been using HA and ESPHome since mars this year so a bit uncertain what would be useful.

jpenrodST, 2024-10-01:
Recently updated to 2024.9.2 and having the same issue as well.

ssieb, 2024-10-01:
Duplicate of #6259

closed 2024-10-01 by ssieb