esphome/issues #6277: M5Stack POECam-W is 20x slower with esphome

archef2000, 2024-09-26:
### The problem

I have the M5Stack POECam-W that I connected over ethernet both times and the [example sketch](https://github.com/m5stack/M5PoECAM/blob/5dece159b5f86b52b9eaaba1f0fbc889cb6f3bf9/examples/web_cam/eth/eth.ino) takes about 50-100ms per frame.

<details>
<summary>log from example:</summary>

JPG: 42115B 20ms, speed: 16164 kbps/s
JPG: 41981B 70ms, speed: 4779 kbps/s
JPG: 11600B 65ms, speed: 1414 kbps/s
JPG: 11121B 45ms, speed: 1964 kbps/s
JPG: 11933B 63ms, speed: 1513 kbps/s
JPG: 10942B 45ms, speed: 1938 kbps/s
JPG: 12889B 15ms, speed: 6835 kbps/s
JPG: 13153B 35ms, speed: 2925 kbps/s
JPG: 12502B 35ms, speed: 2780 kbps/s
JPG: 26674B 77ms, speed: 2763 kbps/s
JPG: 12220B 30ms, speed: 3200 kbps/s
JPG: 12518B 84ms, speed: 1188 kbps/s
JPG: 12741B 46ms, speed: 2190 kbps/s
JPG: 12526B 18ms, speed: 5340 kbps/s
JPG: 12501B 73ms, speed: 1368 kbps/s
JPG: 12745B 46ms, speed: 2188 kbps/s
JPG: 13038B 18ms, speed: 5517 kbps/s
JPG: 12641B 35ms, speed: 2826 kbps/s
JPG: 12686B 36ms, speed: 2817 kbps/s
JPG: 12506B 35ms, speed: 2787 kbps/s
JPG: 27115B 77ms, speed: 2805 kbps/s
JPG: 27786B 72ms, speed: 3070 kbps/s
JPG: 12622B 79ms, speed: 1263 kbps/s
JPG: 13101B 5ms, speed: 19073 kbps/s
JPG: 12641B 52ms, speed: 1916 kbps/s
JPG: 12858B 35ms, speed: 2862 kbps/s
JPG: 12567B 36ms, speed: 2786 kbps/s
JPG: 13114B 85ms, speed: 1228 kbps/s
JPG: 12552B 5ms, speed: 18832 kbps/s
JPG: 12464B 102ms, speed: 971 kbps/s
JPG: 13457B 47ms, speed: 2289 kbps/s
JPG: 12973B 11ms, speed: 9123 kbps/s
JPG: 26397B 77ms, speed: 2726 kbps/s
JPG: 13206B 30ms, speed: 3428 kbps/s
JPG: 12511B 35ms, speed: 2804 kbps/s
JPG: 12798B 36ms, speed: 2786 kbps/s
JPG: 13108B 88ms, speed: 1182 kbps/s
JPG: 12666B 5ms, speed: 18820 kbps/s
JPG: 13781B 49ms, speed: 2227 kbps/s
JPG: 12485B 35ms, speed: 2807 kbps/s
JPG: 12779B 77ms, speed: 1314 kbps/s
JPG: 12505B 5ms, speed: 18615 kbps/s
JPG: 12378B 96ms, speed: 1024 kbps/s
JPG: 12575B 84ms, speed: 1192 kbps/s
JPG: 26433B 29ms, speed: 7244 kbps/s
JPG: 12669B 30ms, speed: 3282 kbps/s
JPG: 13776B 35ms, speed: 3073 kbps/s
JPG: 28040B 77ms, speed: 2899 kbps/s
JPG: 27914B 72ms, speed: 3100 kbps/s
JPG: 12407B 77ms, speed: 1284 kbps/s
JPG: 13248B 49ms, speed: 2160 kbps/s
JPG: 12539B 11ms, speed: 8421 kbps/s
JPG: 12920B 38ms, speed: 2719 kbps/s
JPG: 12424B 74ms, speed: 1342 kbps/s
JPG: 12913B 46ms, speed: 2216 kbps/s
JPG: 12376B 21ms, speed: 4645 kbps/s
JPG: 12130B 86ms, speed: 1116 kbps/s
JPG: 12852B 46ms, speed: 2196 kbps/s
JPG: 13139B 10ms, speed: 10197 kbps/s
JPG: 13273B 35ms, speed: 2967 kbps/s
JPG: 13453B 36ms, speed: 2969 kbps/s
JPG: 12965B 35ms, speed: 2898 kbps/s
JPG: 11862B 35ms, speed: 2663 kbps/s
JPG: 12871B 80ms, speed: 1280 kbps/s
JPG: 12120B 5ms, speed: 19067 kbps/s
JPG: 13433B 59ms, speed: 1821 kbps/s
JPG: 12590B 35ms, speed: 2830 kbps/s
JPG: 13039B 36ms, speed: 2892 kbps/s
JPG: 13064B 35ms, speed: 2903 kbps/s
JPG: 13043B 36ms, speed: 2894 kbps/s
JPG: 12989B 35ms, speed: 2890 kbps/s
JPG: 12461B 35ms, speed: 2781 kbps/s
JPG: 26680B 77ms, speed: 2766 kbps/s
JPG: 13003B 30ms, speed: 3356 kbps/s
JPG: 13294B 35ms, speed: 2956 kbps/s
JPG: 13053B 35ms, speed: 2910 kbps/s
JPG: 13042B 35ms, speed: 2906 kbps/s
JPG: 12862B 36ms, speed: 2856 kbps/s
JPG: 13156B 35ms, speed: 2935 kbps/s
JPG: 12873B 36ms, speed: 2854 kbps/s
JPG: 12674B 38ms, speed: 2648 kbps/s
JPG: 12894B 33ms, speed: 3059 kbps/s
JPG: 12800B 35ms, speed: 2847 kbps/s
JPG: 13634B 36ms, speed: 3011 kbps/s
JPG: 13523B 35ms, speed: 3010 kbps/s
JPG: 12402B 35ms, speed: 2788 kbps/s
</details>

The same config in esphome is taking 2s per frame even when the image is only twice as big.
Tested with esp32_camera_web_server

<details>
<summary>log from esphome:</summary>
<code>
  [D][esp32_camera:196]: Got Image: len=19809
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19809B 3013ms (0.3fps)
  [D][esp32_camera:196]: Got Image: len=19901
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19901B 3013ms (0.3fps)
  [D][esp32_camera:196]: Got Image: len=19765
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19765B 2012ms (0.5fps)
  [D][esp32_camera:196]: Got Image: len=19672
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19672B 3013ms (0.3fps)
  [D][esp32_camera:196]: Got Image: len=19818
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19818B 3014ms (0.3fps)
  [D][esp32_camera:196]: Got Image: len=19846
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19846B 2014ms (0.5fps)
  [D][esp32_camera:196]: Got Image: len=19782
  [D][esp32_camera_web_server:198][httpd]: MJPG: 19782B 3008ms (0.3fps)
  [D][esp32_camera:196]: Got Image: len=62292
  [D][esp32_camera_web_server:198][httpd]: MJPG: 62292B 6049ms (0.2fps)
</code>
</details>

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not HA

### What platform are you using?

ESP32

### Board

M5Stack POECam-W

### Component causing the issue

esp32_camera, ethernet (W5500)

### Example YAML snippet

```yaml
esphome:
      name: esp-poe
      friendly_name: esp-poe
    
    esp32:
      board: esp32dev
      framework:
        type: arduino
    
    logger:
    
    ethernet:
      type: W5500
      clk_pin: GPIO23
      mosi_pin: GPIO13
      miso_pin: GPIO38
      cs_pin: GPIO4
      clock_speed: 23Mhz
    
    esp32_camera:
      external_clock:
        pin: GPIO27
        frequency: 20MHz
      i2c_pins:
        sda: GPIO14
        scl: GPIO12
      data_pins: [GPIO32, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
      vsync_pin: GPIO22
      href_pin: GPIO26
      pixel_clock_pin: GPIO21
      reset_pin: GPIO15
      resolution: SXGA
      name: PoECam
      jpeg_quality: 16
      frame_buffer_count: 2
    
    psram:
      mode: octal
      speed: 80MHz
    
    esp32_camera_web_server:
      - port: 8080
        mode: stream
      - port: 8081
        mode: snapshot
    
    binary_sensor:
      - platform: gpio
        pin:
          number: GPIO37
          inverted: true
        name: "Button"
```

cortices, 2024-09-27:
This is caused by the lack of support for direct polling of SPI Eth chipsets, which causes erratic behaviour when no interrupt pin is configured.
Will be solved by the PR being proposed in #6268

cortices, 2024-11-23:
Unfortunately there's still not an Arduino framework which supports the necessary ESP feature for polling mode.
And for some reason the ESP_IDF fixed version will just not initialise on M5 POECam-W, I think because the Reset pin isn't accessible.