esphome/issues #6281: RGBWW Light + MQTT - Invalid color mode 'cwww' received

nima-1102, 2024-09-27:
### The problem

RGB works without any problems with the RGBWW Light, but as soon as you want to set the cold/warm white temperature via the Homeassistant GUI, the lamp temperature changes as desired, but a warning appears. The last RGB color is then displayed as the current light color and not the current cold/warm white temperature.

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

BK72XX

### Board

WBLC5

### Component causing the issue

mqtt / rgbww

### Example YAML snippet

```yaml
bk72xx:
  board: generic-bk7231t-qfn32-tuya

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

sm2135:
  clock_pin: P26
  data_pin: P24
  rgb_current: 15mA
  cw_current: 35mA
  separate_modes: false

output:
  - platform: sm2135
    id: output_blue
    channel: 0
    min_power: 0.2
    max_power: 0.9
    zero_means_zero: true
  - platform: sm2135
    id: output_green
    channel: 1
    min_power: 0.2
    max_power: 0.9
    zero_means_zero: true
  - platform: sm2135
    id: output_red
    channel: 2
    min_power: 0.2
    max_power: 0.9
    zero_means_zero: true
  - platform: sm2135
    id: output_warm
    channel: 3
    min_power: 0.2
    max_power: 0.9
    zero_means_zero: true
  - platform: sm2135
    id: output_cold
    channel: 4
    min_power: 0.2
    max_power: 0.9
    zero_means_zero: true

light:
  - platform: rgbww
    id: light_rgbww
    name: "Test Light"
    color_interlock: true
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3000 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm
    default_transition_length: 0s
```

### Anything in the logs that might be useful for us?

```txt
logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:365
integration: MQTT (documentation, issues)
First occurred: September 27, 2024 at 23:56:21 (3 occurrences)
Last logged: 00:00:00

Invalid color mode 'cwww' received for entity light.test_light
```

### Additional information

MQTT Payload of the Warm / Cold Temp change: 
`{"color_mode":"cwww","state":"ON","brightness":133,"color":{"c":0,"w":255}}`

color_interlock is required for the SM2135 LED driver.

ssieb, 2024-09-27:
We need more information.  That error message is from HA, the mqtt integration.  It's not an esphome error.
Is that mqtt payload what is getting sent to esphome?
Is the light changing to what you want, but the color displayed in HA is wrong?

nima-1102, 2024-09-27:
> We need more information. That error message is from HA, the mqtt integration. It's not an esphome error. Is that mqtt payload what is getting sent to esphome? Is the light changing to what you want, but the color displayed in HA is wrong?

This is the payload which, after I have changed the light from e.g. OFF or RGB to cold/warm white temperature via the Home Assistant GUI, ESPHome stores as a state in mqtt under "lamp-gu10-2/light/test-light/state". Yes, the light changes color temperature, but color displayed in HA is wrong because the data is not changed due to "Invalid color mode".

I don't know if ESPHome sets the color_mode incorrectly or if the color_mode is just not defined in Home Assistant.

nima-1102, 2025-01-29:
That's still a problem.

kquinsland, 2026-01-18:
Related / possibly a dupe: https://github.com/esphome/issues/issues/6130