esphome/issues #6283: Api not connecting after update

zumbefin, 2024-09-28:
### The problem

Hi, i have one esp32 device that after every ota update losts api connection, if i update that specific device from esp home ui.
it gives error: 

"WARNING Can't connect to ESPHome API for takapiha @ 10.1.1.40: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.1.1.40', port=6053))]: [Errno 111] Connect call failed ('10.1.1.40', 6053) (SocketAPIError)"

All other devices are working after updates.

i can ping the device and update with OTA so only api goes broken somehow.

i have swapped same type of esp32 to rule out hardware issue.

Usually it gets fixed when i do the download with usb cable.

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: "takapiha"
  friendly_name: Takapiha 

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:  
  #encryption: 
    #key: "H1b619zclEtTHFKv0ElSZgfY78MqJJg6J9XbgkbQmEU="

ota:
  - platform: esphome
    password: salasana

safe_mode:
  boot_is_good_after: 90s
  num_attempts: 3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Takapiha Fallback Hotspot"
    password: "w7TadNQo9DpS"

  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.1.1.40
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.1.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0  

captive_portal:

#web_server:
 # port: 80
  #version: 3    

# i2C pins and setup
i2c:
  sda: GPIO21 # D21
  scl: GPIO22 # D22
  scan: true
  id: bus_a  

#Valoisuus anturi BH1750
sensor:
  - platform: bh1750
    name: "takapiha valoisuus bh1750"
    address: 0x23
    update_interval: 60s

  - platform: sht4x
    temperature:
      name: "takapiha lämpötila sht40"
    humidity:
      name: "takapiha kosteus sht40"
    heater_max_duty: 0.05       

#sisäinen lämpötila
  - platform: internal_temperature
    name: "takapiha ESP lämpötila"    

  # BME280 Temperature, Pressure, Humidity sensor
  - platform: bme280_i2c
    temperature:
      name: "takapiha lämpötila bme280"
    pressure:
      name: "takapiha ilmanpaine bme280"
    humidity:
      name: "takapiha kosteus bme280"
    address: 0x76  

#restart nappi
switch:
  - platform: restart
    name: "Restart ESP32"
    id: restart_switch
```

MiniBlister, 2024-09-29:
I have the same issue. Going back to 2024.7 and the device is online again.

spudje, 2024-09-29:
I also have the exact same issue. Reverted back to 2024.6.6 in my case and everything works again.

ssieb, 2024-09-29:
That indicates that it is bootlooping.  Why do you have serial logs disabled?  You will need to get serial logs to find out why it's crashing.

siegeld, 2024-10-06:
Same issue here:

INFO ESPHome 2024.9.2
WARNING Can't connect to ESPHome API for esphome-web-edb248 @ 10.11.2.62: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.11.2.62', port=6053))]: [Errno 111] Connect call failed ('10.11.2.62', 6053) (SocketAPIError)

siegeld, 2024-10-06:
Downgrading to 2024.7.3 did not fix the problem.

chrishoage, 2024-10-07:
I am having similar issues. I can ping, I can upload but only sometimes. 

The API can't connect. 

I'm using esp8266 esp01_1m (a sonoff s31) 

Oddly when I flash the esp over serial everything  works while it's powered by the flasher. 

When I plug the S31 into the wall I get the issues.

spudje, 2024-10-07:
I really wouldn't know what change in my yaml to get it working again. I don't have a sensor in there.

`substitutions:
  devicename: ventilatie
  upper_devicename: Ventilatie

esphome:
  name: ventilatie
  comment: Fan controller

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO
#  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: !secret esphome_ventilatie_api_encryption_key

  services:
    - service: set_speed
      variables:
        run_speed: int
        run_time: int
      then:
        - lambda: |-
            zehnder_fan->setSpeed(run_speed, run_time);  

ota:
  password: !secret esphome_ventilatie_ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "utility-bridge Fallback"
    password: !secret esphome_utility_bridge_ap_password

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: Europe/Amsterdam

button:
  - platform: restart
    id: ${devicename}_esphome_restart
    name: ${upper_devicename} herstart

# Load external components
external_components:
  - source:
      type: local
      path: custom_components

# SPI
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

# nRF905 config
nrf905:
  id: "nrf905_rf"
  # cs_pin: GPIO23
  cs_pin: GPIO15
  cd_pin: GPIO33
  ce_pin: GPIO27
  pwr_pin: GPIO26
  txen_pin: GPIO25
  # We don't need AM and DR at the moment as they are read from the inernal registers
  am_pin: GPIO32
  dr_pin: GPIO35

# The FAN controller
fan:
  - platform: zehnder
    id: zehnder_fan
    name: "Ventilation"
    nrf905: nrf905_rf
    update_interval: "30s"`

MiniBlister, 2024-10-20:
on my system it seems to be the SPI that cause bootloop

spudje, 2024-10-20:
Did you fix it and how?

alwarrior, 2024-11-16:
> on my system it seems to be the SPI that cause bootloop

Same problem, I confirm that only the module that spi uses has the problem.

andyfrei, 2024-11-28:
same issue here iwth HA 2024.11.3 and ESPHome 2024.11.2

rlesac, 2024-12-03:
I have a similar issue, esp32 cams, I just updated the camera settings for one, it worked fine with OTA install, made the same changes to the other esp32 cam, after a successful uploading it would not connect to the API again, but was pignable. Tried a few more time with same results, until it became unreachable. Which is shit, as I use HA in a remote location.

spudje, 2024-12-04:
Hope someone could pick this up, or at least give us clues on how to fix.

benjaminalbers, 2024-12-10:
Same issue with ESPHome 2024.10.4 and 2024.11.3. Going back to version 2024.6.6 fix the Problem for me.

There might be a problem with the SPI. If the on_value part in the number component is commented out everything works

`spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19
  interface: hardware
  
spi_device:
  id: spidev
  cs_pin: 5
  spi_mode: mode3

number:
  - platform: template
    name: "Cannel 1"
    optimistic: true
    min_value: 0
    max_value: 255
    step: 1
    mode: box
    id: Poti
    on_value: 
      then:
        - lambda: !lambda |-
            id(spidev).enable();
            id(spidev).write_byte(0b00000000);
            id(spidev).write_byte(x);
            id(spidev).disable();

button:
  - platform: template
    name: Increment
    on_press: 
      then:
        - logger.log: Button PLUS Pressed
        - number.set: 
            id: Poti
            value: !lambda "return id(Poti).state+1;"
        - lambda: !lambda |-
            id(spidev).enable();
            id(spidev).write_byte(0b00000100);
            id(spidev).disable();
                          
  - platform: template
    name: Decrement
    on_press: 
      then:
        - logger.log: Button MINUS Pressed
        - number.set: 
            id: Poti
            value: !lambda "return id(Poti).state-1;"
        - lambda: !lambda |-
            id(spidev).enable();
            id(spidev).write_byte(0b00001000);
            id(spidev).disable();
`

magverhoeven, 2024-12-11:
I have the same problem, if I comment out this on_value my display works but offcourse 'im missing some stats I like to display.

Cjkeenan, 2024-12-21:
Anyone find a resolution to this? Still seems to be an issue in 2024.12.1 with my Feit Outdoor plug that has no SPI or on_value parts in the yaml.

Cjkeenan, 2024-12-21:
Anyone find a resolution to this? Still seems to be an issues in 2024.12.1 with my Feit Outdoor plug that has no SPI or on_value parts in the yaml.

jawilczek, 2024-12-21:
update esphome 2024.12.2 don't fix problem

"
**WARNING Can't connect to ESPHome API for display02 @ 10.xx.xx.xx: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6,**
"

ssieb, 2024-12-23:
@spudje you have at least one external component there and maybe others.  You really shouldn't use the `custom_components` directory.
@benjaminalbers can you get serial logs of the crash?

spudje, 2024-12-24:
> Quote reply

@ssieb No clue how to change it then. This is an ESP32 with Radio module to control ventilation, which is not supported by ESPHome and not created by me.

spudje, 2024-12-25:
OK, I got it working again hanks to this fix: https://github.com/Sanderhuisman/ESPHome-Zehnder-RF/issues/7

benjaminalbers, 2024-12-27:
@ssieb this is the log I get after uploading hope this will help

[ESP32-NodeMCU_logs.txt](https://github.com/user-attachments/files/18263839/ESP32-NodeMCU_logs.txt)

ssieb, 2024-12-27:
@spudje rename the `custom_components` directory to something else or you will have the potential for random errors in the future.

ssieb, 2024-12-27:
@benjaminalbers try adding `setup_priority: 599` to the template number.

benjaminalbers, 2025-01-01:
@ssieb setting setup_priority to 599 is working for me. I'm now running version 2024.12.2 and everything works as expected. Many thanks to you.

closed 2025-01-02 by ssieb