esphome/issues #6286: LVGL animimg widget causes excessive blocking errors.

Littleislandbrewing, 2024-09-29:
### The problem

When using a animimg widget that is full frame, it causes 100ms+ blocking errors on each change of the image.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### Component causing the issue

LVGL - animimg:

### Example YAML snippet

```yaml
switch:
  - platform: template
    name: "test anim"
    id: test_anim
    optimistic: true
    turn_on_action:
      - lvgl.animimg.update:
          id: [ page1_error_background, ss_error_background ]
          duration: 1s
          hidden: false  

  pages:
    - id: page1
      widgets:
        - animimg:
            id: page1_error_background
            src: [ display_error, display_cooling ]
            duration: 1000ms
            hidden: true

    - id: page2
      widgets:
        - animimg:
            id: ss_error_background
            src: [ display_error, display_cooling ]
            duration: 1000ms
            hidden: true

image:
  - file: tank_cooling.png
    id: display_cooling
    resize: 480x480
    type: RGB565
  - file: tank_error.png
    id: display_error
    resize: 480x480
    type: RGB565
```

### Anything in the logs that might be useful for us?

```txt
[15:51:14][D][sensor:094]: 'Heap Free': Sending state 121548.00000 B with 0 decimals of accuracy
[15:51:14][D][sensor:094]: 'Loop Time': Sending state 80.00000 ms with 0 decimals of accuracy
[15:51:14][D][sensor:094]: 'Heap Max Block': Sending state 83968.00000 B with 0 decimals of accuracy
[15:51:14][D][sensor:094]: 'Free PSRAM': Sending state 3567467.00000 B with 0 decimals of accuracy
[15:51:19][D][sensor:094]: 'Heap Free': Sending state 121576.00000 B with 0 decimals of accuracy
[15:51:19][D][sensor:094]: 'Loop Time': Sending state 20.00000 ms with 0 decimals of accuracy
[15:51:19][D][sensor:094]: 'Heap Max Block': Sending state 83968.00000 B with 0 decimals of accuracy
[15:51:19][D][sensor:094]: 'Free PSRAM': Sending state 3567467.00000 B with 0 decimals of accuracy
[15:51:21][W][web_server_idf:070][httpd]: Only application/x-www-form-urlencoded supported for POST request
[15:51:21][D][switch:012]: 'test anim' Turning ON.
[15:51:21][D][switch:055]: 'test anim': Sending state ON
[15:51:21][W][component:237]: Component lvgl took a long time for an operation (111 ms).
[15:51:21][W][component:238]: Components should block for at most 30 ms.
[15:51:21][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:21][W][component:238]: Components should block for at most 30 ms.
[15:51:22][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:22][W][component:238]: Components should block for at most 30 ms.
[15:51:22][W][component:237]: Component lvgl took a long time for an operation (111 ms).
[15:51:22][W][component:238]: Components should block for at most 30 ms.
[15:51:22][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:22][W][component:238]: Components should block for at most 30 ms.
[15:51:23][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:23][W][component:238]: Components should block for at most 30 ms.
[15:51:23][W][component:237]: Component lvgl took a long time for an operation (111 ms).
[15:51:23][W][component:238]: Components should block for at most 30 ms.
[15:51:24][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:24][W][component:238]: Components should block for at most 30 ms.
[15:51:24][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:24][W][component:238]: Components should block for at most 30 ms.
[15:51:24][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:24][W][component:238]: Components should block for at most 30 ms.
[15:51:25][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:25][W][component:238]: Components should block for at most 30 ms.
[15:51:25][D][sensor:094]: 'Heap Free': Sending state 114636.00000 B with 0 decimals of accuracy
[15:51:25][D][sensor:094]: 'Loop Time': Sending state 128.00000 ms with 0 decimals of accuracy
[15:51:25][D][sensor:094]: 'Heap Max Block': Sending state 83968.00000 B with 0 decimals of accuracy
[15:51:25][D][sensor:094]: 'Free PSRAM': Sending state 3567467.00000 B with 0 decimals of accuracy
[15:51:25][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:25][W][component:238]: Components should block for at most 30 ms.
[15:51:25][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:25][W][component:238]: Components should block for at most 30 ms.
[15:51:26][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:26][W][component:238]: Components should block for at most 30 ms.
[15:51:26][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:26][W][component:238]: Components should block for at most 30 ms.
[15:51:26][W][component:237]: Component lvgl took a long time for an operation (111 ms).
[15:51:26][W][component:238]: Components should block for at most 30 ms.
[15:51:27][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:27][W][component:238]: Components should block for at most 30 ms.
[15:51:27][W][component:237]: Component lvgl took a long time for an operation (109 ms).
[15:51:27][W][component:238]: Components should block for at most 30 ms.
[15:51:27][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:27][W][component:238]: Components should block for at most 30 ms.
[15:51:28][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:28][W][component:238]: Components should block for at most 30 ms.
[15:51:28][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:28][W][component:238]: Components should block for at most 30 ms.
[15:51:28][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:28][W][component:238]: Components should block for at most 30 ms.
[15:51:29][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:29][W][component:238]: Components should block for at most 30 ms.
[15:51:29][W][component:237]: Component lvgl took a long time for an operation (110 ms).
[15:51:29][W][component:238]: Components should block for at most 30 ms.
[15:51:29][W][component:237]: Component lvgl took a long time for an operation (111 ms).
[15:51:29][W][component:238]: Components should block for at most 30 ms.
```

clydebarrow, 2024-10-04:
What display are you using? It's the update of the entire display that is taking a long time. If it's causing a problem I'd suggest using a smaller image. If it's not causing a problem, just ignore the log messages.

Littleislandbrewing, 2024-10-04:
The display is the esp32-s3-4848s040 that everyone is using.

The reasoning for using the 480x480 image size is because I couldnt get the transparency to work properly on a PNG (using photoshop) so in a situation where there are multiple images, stacked on top of each other on different parts of the screen Ive had to make individual images.

clydebarrow, 2024-10-04:
That's nearly half a megabyte of data for each image. It's going to take a while to copy - that's just the nature of the beast.  Are you using ESP-IDF 5.x? That will speed things up, but you will still see log messages.

closed 2024-10-04 by clydebarrow

Littleislandbrewing, 2024-10-04:
No im not but.....

Each image is only 15kb in size as a PNG. What makes you say 500kb???

clydebarrow, 2024-10-04:
480 squared times 2 is the number of bytes of the resulting image. It's uncompressed in flash and in the display RAM.

Littleislandbrewing, 2024-10-12:
how do I make sure Im on ESP-IDF is 5.X

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: dev
    
    or
    
    version: latest
    
both have compile errors

clydebarrow, 2024-10-12:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    platform_version: 6.8.1
    version: 5.3.0
```