esphome/issues #6287: Lvgl arc widget in reverse mode issue

plplaaa2, 2024-09-29:
### The problem

I think the touch response was wrong when using lvgl arc widget in reverse mode.

If you touch the top of Arc, the value goes down
Touch the bottom of Arc to increase the value.

It is difficult to explain in writing, so please refer to the video below for details.

[https://youtu.be/HyvZdVLVdoI](url)

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

ESP32 S3 N8R8

### Component causing the issue

LVGL

### Example YAML snippet

```yaml
- arc:
    x: -30
    y: 15
    width: 280
    height: 280
    border_width: 0
    bg_opa: TRANSP
    align: RIGHT_MID
    adjustable: true
    min_value: 0
    max_value: 100
    start_angle: 180
    end_angle: 0
    arc_width: 30
    rotation: 60
    mode: REVERSE
    id: r1_set_volume
    indicator:
      arc_color: 0x007BB0
      arc_width: 30

    on_release:
      then:
        - homeassistant.action:
            action: media_player.volume_set
            data: 
              entity_id: media_player.bedroom_spk_media_player
              volume_level: !lambda 'return (x / 100);'                                    
    widgets:
      - label:
          align: BOTTOM_MID
          text: "볼  륨"
          text_font: regular_40
          text_color: 0xE5E5E5
          text_align: CENTER
```

clydebarrow, 2024-10-04:
It's a bug in the LVGL library. Will be resolved when the ESPHome library is migrated to LVGL V9. Nothing else can be done about it at present.

closed 2024-10-04 by clydebarrow

plplaaa2, 2024-10-04:
> It's a bug in the LVGL library. Will be resolved when the ESPHome library is migrated to LVGL V9. Nothing else can be done about it at present.

Thank you for your reply.

I solved it by reversing it. :)

[image]