esphome/issues #6291: ESPHome fails to compile for Raspberry Pi Pico W

kcajjones, 2024-09-30:
### The problem

ESPHome 2024.9.1 fails to compile for Raspberry Pi Pico W.
Running HomeAssistant Core 2024.9.3, Supervisor 2024.09.1, Operating System 13.1, Frontend 20240909.1 on a Raspberry Pi 3b.
The Pico W works fine on 2024.8.5 firmware but will not update.

There seems to be other reports of issues with the recent updates to ESPHome here:
[https://community.home-assistant.io/t/esphome-problems-building-for-rp2040-rpipicow/777019](url)

```
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/lounge-subwoofer-pico.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing lounge-subwoofer-pico (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30904.0 (3.9.4) 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.6
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Linking .pioenvs/lounge-subwoofer-pico/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pioenvs/lounge-subwoofer-pico/firmware.elf] Error 1
========================= [FAILED] Took 44.73 seconds =========================
```

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Example YAML snippet

```yaml
esphome:
  name: lounge-subwoofer-pico
  friendly_name: Lounge Subwoofer Pico

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "cQSsoMBtPdQ2XYpWEWp7+2YqIwvBkV1/O1W7jCZ5f9E="

ota:
  - platform: esphome
    password: "e35bb9b6cf370418167faf19e9552dd9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Lounge-Subwoofer-Pico"
    password: "Qr0Ix36qGneY"
    
# Switch component for subwoofer control
switch:
  - platform: gpio
    name: Subwoofer
    pin: 16
    inverted: true  # This line inverts the logic
  # Optional: Set an icon for the switch in HA
    icon: mdi:subwoofer
    restore_mode: ALWAYS_OFF  # Ensures relay starts off on boot
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/lounge-subwoofer-pico.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing lounge-subwoofer-pico (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30904.0 (3.9.4) 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.6
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Linking .pioenvs/lounge-subwoofer-pico/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `std':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x124): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0x17c): undefined reference to `__retarget_lock_init_recursive'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: /data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/lib/thumb/libc.a(libc_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'
collect2: error: ld returned 1 exit status
*** [.pioenvs/lounge-subwoofer-pico/firmware.elf] Error 1
========================= [FAILED] Took 44.73 seconds =========================
```

ssieb, 2024-09-30:
There's ongoing discussion about how to resolve this better.  For now, use this instead:
```yaml
platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c
```

ssieb, 2024-09-30:
After the next point release, you can remove that line completely.

closed 2024-10-01 by jesserockz

kcajjones, 2024-10-01:
> After the next point release, you can remove that line completely.

Just updated ESPHome to 2024.9.2 and the platform_version line is still required for the yaml to be valid.

lsellens, 2024-10-01:
> > After the next point release, you can remove that line completely.
> 
> Just updated ESPHome to 2024.9.2 and the platform_version line is still required for the yaml to be valid.
@kcajjones 
It's working fine for me. Did you leave the "framework:" line by mistake?