esphome/issues #6293: CST816S - component setup fails if the touchscreen controller is in standby state

dsolera, 2024-10-01:
### The problem

The CST816 touchscreen component, during setup() and particularly continue_setup(), reads a register from the I2C bus asking the touchsreen controller its chip ID register. If this read fails, the component is marked as failed and no further action can be done to recover it.

The problem is that the CST816S chip (the one I'm using) enters a **standby state** after 2 seconds of touch inactivity. When in this mode, the chip **does not** respond to I2C commands until it is awaken by a touch ([see datasheet](https://files.waveshare.com/upload/5/51/CST816S_Datasheet_EN.pdf)).

Thus, initialization of the component **always fails**, unless you **keep tapping on the screen until the component is loaded**. If the component is initialized correctly, then touch works fine indefinitely. Moreover, when the chip is in standby, it doesn't even appear in I2C bus scans.

I tried checking the source code of the component and it appears that the chip ID is not really used, so maybe it could be possible to accept it via config and, if provided, to avoid asking the chip ID on setup or maybe just issuing a warning without marking the component as failed. Unfortunately I'm not skilled enough to provide a MR.

I'm not sure if other variants of this chip have the same issue.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

cst816

### Example YAML snippet

```yaml
i2c:
  - id: i2c_bus0
    sda: GPIO21
    scl: GPIO4
    scan: true

touchscreen:
  platform: cst816
  id: touchscreen0
  i2c_id: i2c_bus0
  interrupt_pin: GPIO23
  reset_pin: GPIO32
  address: 0x15
```

clydebarrow, 2024-10-04:
Where in the datasheet does it say that it does not respond to i2c commands in standby mode?

dsolera, 2024-10-04:
It doesn't say that explicitly, but the behavior is exactly that and I
tested two different units (thinking one could be defective).

Moreover, the datasheet says that the chip enters standby mode after 2s of
inactivity and that standby current draw is less than 6 uA. It is
reasonable to say that the I2C bus is disabled at such a low current draw.

The interesting thing is that the datasheet says:
Standby mode - In this mode, the touch chip scans the touch screen at a
lower frequency, and enters the dynamic
mode after detecting the finger. At the same time,
*the touch chip wakes up the host through theIRQ pin or switches to dynamic
mode through the reset pin*.

The source code actually flips the reset pin before initialization, but it
doesn't seem to wake the chip up (my reset pin is configured correctly).
Maybe it's some kind of race condition during init. If we can't remove the
chip ID request during setup, maybe it's possible to add a method to reset
the component state afterwards?

clydebarrow, 2024-10-04:
I did find some information elsewhere that supports what you're suggesting - that it does not respond to i2c in standby state. That's a nuisance since that means there is no way to reliably check that it even exists on the bus (which is the main reason for reading the chip ID.) It will need an additional config option to tell the driver not to expect to be able to read a chip ID.

clydebarrow, 2024-10-07:
@dsolera Please test the PR and see if it fixes your issue. It tests fine for me on a CST816T.

dsolera, 2024-10-07:
I did a quick test and it works perfectly! Thank you very much.

closed 2024-10-09 by jesserockz