esphome/issues #6296: Deepsleep INVERT_WAKEUP fundamentaly broken

geludwig, 2024-10-02:
### The problem

A reed switch is used to wakeup an ESP32 from deepsleep. The ESP might go into deepsleep when the switch is closed or open, but should wake up every time a level change occurs (HIGH -> LOW or LOW -> HIGH).

May I cite from the docs:
_INVERT_WAKEUP: When deep sleep was set up to wake up on a HIGH signal, but the wakeup pin is already HIGH, then re-configure deep sleep to wake up on a LOW signal and vice versa. Useful in situations when you want to use observe the state changes of a pin using deep sleep and the ON/OFF values last longer._

Issue:
- ESPHome assigns pull-down/pull-up resistors when entering deepsleep automatically.
- When entering deepsleep with pull-up (internally or externally), INVERT_WAKEUP and the pin pulled to ground, the pull-down resistor is assigned automatically (why?).

Solve:
- Do NOT assign any automatic pullup/pulldown when entering deepsleep when using INVERT_WAKEUP. Or give the ability to wakeup on EDGE_CHANGE.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

XIAO ESP32C3

### Component causing the issue

deepsleep

### Example YAML snippet

```yaml
deep_sleep:
  run_duration: 2min
  #sleep_duration: 5min
  wakeup_pin:
    number: GPIO4
    allow_other_uses: true
    mode: INPUT_PULLUP
  wakeup_pin_mode: INVERT_WAKEUP
```

### Additional information

[image]

geludwig, 2024-10-03:
Addition: This does not work ...
```
esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y
      ESP_SLEEP_GPIO_ENABLE_INTERNAL_RESISTORS: n
    advanced:
      ignore_efuse_mac_crc: false
```

closed 2024-10-03 by geludwig

moesterheld, 2024-10-31:
But what if I **want** to use an internal resistor for pull-up? I think the setting won't be helping me there since it also disables my pull-up, right?

geludwig, 2024-11-01:
"CONFIG_ESP_SLEEP_GPIO_ENABLE_INTERNAL_RESISTORS" disables internal pull-ups and pull-downs during deep-sleep for all pins which are configured for deep-sleep.

So yes, if the internal resistors are mandatory, "CONFIG_ESP_SLEEP_GPIO_ENABLE_INTERNAL_RESISTORS: n" cant be used.