esphome/issues #6301: OTA: ERROR Error receiving acknowledge binary size: timed out USB: ok

Littleislandbrewing, 2024-10-04:
### The problem

I have 8 units of 4" LVGL screens that have "progressively" shown this issue as I have updated the functionality in them via OTA.

Originally this issue started and would fail but give the option to "retry" which would work, but, possibly, as the memory allocation has increased, this issue has progressed to only being able to flash via USB on all of them except for one unit.

The sketch is approx 3500 lines (yes) and works without issue. (5119552 bytes) and they all have the same code on them except for the name and friendly name. 

From what I can see in the logger, ESPHome creates the binary. 

Is it possible that ESPHome cant upload a binary of that size via OTA???

### Which version of ESPHome has the issue?

2024.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

board: esp32-s3-devkitc-1

### Example YAML snippet

```yaml
this is a OTA issue
```

### Anything in the logs that might be useful for us?

```txt
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- lvgl @ 8.4.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/opbd1-b069660/src/main.o
Linking .pioenvs/opbd1-b069660/firmware.elf
RAM:   [=         ]  11.1% (used 36264 bytes from 327680 bytes)
Flash: [======    ]  63.0% (used 5119181 bytes from 8126464 bytes)
Building .pioenvs/opbd1-b069660/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/opbd1-b069660/firmware.bin"], [".pioenvs/opbd1-b069660/firmware.elf"])
Wrote 0x4f1e40 bytes to file /data/build/opbd1-b069660/.pioenvs/opbd1-b069660/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/opbd1-b069660/firmware.bin"], [".pioenvs/opbd1-b069660/firmware.elf"])
========================= [SUCCESS] Took 45.03 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.xx.xx
INFO Uploading /data/build/opbd1-b069660/.pioenvs/opbd1-b069660/firmware.bin (5119552 bytes)
ERROR Error receiving acknowledge binary size: timed out
```

j4c3, 2024-10-16:
I have run into this same issue with the same board definition.  I found that a sketch that is 2735465 bytes will upload via ota okay, but bumping the size to 3196224 bytes will fail.   Logging the esp side, when it works, I see: 

`08:54:23
[D]
[esphome.ota:117]
Starting update from 192.168.1.xx...`

`08:54:32
[D]
[esphome.ota:293]
Progress: 0.0%`

`08:54:32
[D]
[esphome.ota:...`

When it fails, it shows:

`08:58:11
[D]
[esphome.ota:117]
Starting update from 192.168.1.xx...`

`08:58:23
[W]
[esphome.ota:370]
Remote closed connection`

`08:58:23
[W]
[esphome.ota:244]
Reading binary MD5 checksum failed`

`08:58:23
[W]
[esphome.ota:398]
Failed to write 1 bytes of data, errno 104`

`08:58:23
[E]
[component:164]
Component esphome.ota set Error flag: unspecified`

`08:58:28
[E]
[component:176]
Component esphome.ota cleared Error flag`

ssied on discord mentioned there may be an issue with it timing out, potentially in erasing the flash?

msinn, 2024-10-19:
Same problem here. Same board definition as the two above.

ciberio, 2024-11-26:
mismo problema, después de varias compilaciones seguidas, repentinamente aparece este error.
"ERROR Error receiving acknowledge binary size: timed out"

instalacion por usb sin problemas.

pixelwave, 2025-01-15:
Today I got the same error trying to compile via OTA:

`ERROR Error receiving acknowledge binary size: timed out`

Full Log:

```

INFO ESPHome 2024.12.2
INFO Reading configuration /config/lilygo-t4-s3.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing lilygo-t4-s3 (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 512KB RAM, 8MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
|-- lvgl @ 8.4.0
Compiling .pioenvs/lilygo-t4-s3/src/main.cpp.o
Linking .pioenvs/lilygo-t4-s3/firmware.elf
RAM:   [=         ]   7.3% (used 38272 bytes from 524288 bytes)
Flash: [====      ]  43.4% (used 3522849 bytes from 8126464 bytes)
Building .pioenvs/lilygo-t4-s3/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/lilygo-t4-s3/firmware.bin"], [".pioenvs/lilygo-t4-s3/firmware.elf"])
Wrote 0x36c290 bytes to file /config/.esphome/build/lilygo-t4-s3/.pioenvs/lilygo-t4-s3/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/lilygo-t4-s3/firmware.bin"], [".pioenvs/lilygo-t4-s3/firmware.elf"])
========================= [SUCCESS] Took 13.58 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.136 port 3232...
INFO Connected to 192.168.178.136
INFO Uploading /config/.esphome/build/lilygo-t4-s3/.pioenvs/lilygo-t4-s3/firmware.bin (3523216 bytes)
ERROR Error receiving acknowledge binary size: timed out
```

It works no issue via cable/USB but OTA is always throwing that error?

I also tried "Clean build files" and to reboot ESPHome. No effect.

pixelwave, 2025-01-20:
UPDATE: In my case that fixed it -> https://github.com/esphome/issues/issues/6695

closed 2025-02-10 by jesserockz