esphome/issues #6303: 2024.9.2: Substitution must only consist of upper/lowercase characters, the underscore and numbers. The character '-' cannot be used.

532910, 2024-10-04:
### The problem

  
  Substitution must only consist of upper/lowercase characters, the underscore and numbers. The character '-' cannot be used.
  eugene-phone_ibeacon_uuid: UUID

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Additional information

Such breaking changes MUST NOT happen on minor update. 2024.9.1 worked fine

ssieb, 2024-10-04:
That code hasn't changed in years.

532910, 2024-10-05:
hmm, indeed, my fault

closed 2024-10-05 by 532910