esphome/issues #6304: radon_eye_ble doesn't connect to Radon Eye RD200 V3

baldisos, 2024-10-05:
### The problem

I am trying to connect my Radon Eye RD200 V3 sensor with the radon_eye_ble component.
I can read and write data with the python script from [https://github.com/sormy/radoneye](sormy).
The output i get from the python script when using _radoneye list_:
`94:B5:55:7F:51:26	FR:HG04RE000465`
And when using _radoneye status 94:B5:55:7F:51:26_       
```
alarm_enabled	no
alarm_interval_minutes	60
alarm_level_bq_m3	300
alarm_level_pci_l	8.11
counts_current	85
counts_previous	90
counts_str	85/90
day_avg_bq_m3	584
day_avg_pci_l	15.78
display_unit	bq/m3
firmware_version	V3.0.1
latest_bq_m3	452
latest_pci_l	12.22
model	D200V3
month_avg_bq_m3	0
month_avg_pci_l	0.0
peak_bq_m3	961
peak_pci_l	25.97
serial	000HG04RE465
uptime_minutes	9729
uptime_str	6d18h09m

```
But when using the radon_eye_ble within esphome on a Wemos Mini D1 ESP32 (ESP32-Wroom-32) i don't get any output. No logs on connection and no data.

I've only ever read about V2 versions online, the device i got is a V3. But it does work with the python script, so i guess there's something missing within the esphome component.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.1

### What platform are you using?

ESP32

### Board

Wemos Mini D1 ESP32 (ESP32-Wroom-32)

### Component causing the issue

radon_eye_ble

### Example YAML snippet

```yaml
esp32_ble_tracker:

ble_client:
  - mac_address: 94:b5:55:7f:51:26
    id: radon_eye_ble_id

sensor:
  - platform: radon_eye_rd200
    ble_client_id: radon_eye_ble_id
    update_interval: 5min # default
    radon:
      name: "Radon"
    radon_long_term:
      name: "Radon Langzeit"
```

### Anything in the logs that might be useful for us?

```txt
[14:51:09][W][radon_eye_rd200:141]: Connection in progress
[14:53:09][W][radon_eye_rd200:141]: Connection in progress
[14:54:57][D][esp32_ble_tracker:270]: Starting scan...
[14:55:09][W][radon_eye_rd200:141]: Connection in progress
[14:57:09][W][radon_eye_rd200:141]: Connection in progress
[14:59:09][W][radon_eye_rd200:141]: Connection in progress
[14:59:58][D][esp32_ble_tracker:270]: Starting scan...
[15:01:09][W][radon_eye_rd200:141]: Connection in progress
[15:03:09][W][radon_eye_rd200:141]: Connection in progress
[15:04:57][D][esp32_ble_tracker:270]: Starting scan...
[15:05:09][W][radon_eye_rd200:141]: Connection in progress
[15:07:09][W][radon_eye_rd200:141]: Connection in progress
[15:09:09][W][radon_eye_rd200:141]: Connection in progress
[15:09:57][D][esp32_ble_tracker:270]: Starting scan...
[15:11:09][W][radon_eye_rd200:141]: Connection in progress
[15:13:09][W][radon_eye_rd200:141]: Connection in progress
[15:14:57][D][esp32_ble_tracker:270]: Starting scan...
[15:15:09][W][radon_eye_rd200:141]: Connection in progress
[15:17:09][W][radon_eye_rd200:141]: Connection in progress
```

### Additional information

[image]

ckuethe, 2024-10-08:
maybe a duplicate of #5655 ?

baldisos, 2024-10-08:
As per the instructions from https://esphome.io/components/sensor/radon_eye_ble.html, if i understand it correctly, the _radon_eye_ble_ component is just used to find the mac address of the Radon Eye? The _radon_eye_ble_ isn't referenced in the _radon_eye_rd200_ component which is used to read the values, right? So the issue should be within the _radon_eye_rd200_ component. The referenced issue is within the _radon_eye_ble_ component.

ckuethe, 2024-10-09:
For fun, I patched radon_eye_rd200 to use the UUIDs from [sormy/radoneye/KNOWLEDGE_V2.md](https://github.com/sormy/radoneye/blob/main/KNOWLEDGE_V2.md) ... which also doesn't work, but at least gets a little further. Far enough to say whether protocol v1 or v2 is in use.

I haven't tried updating the decoder to work with v2 responses.

```
[C][esp32_ble:032]: Setting up BLE...
[D][esp32_ble:284]: Enabling BLE...
[C][esp32_ble:032]: Setting up BLE...
[C][esp32_ble:032]: Setting up BLE...
[W][radon_eye_rd200:141]: Connection in progress
[D][esp32_ble:284]: Enabling BLE...
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 0
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=0 gattc_if=4
[V][esp32_ble_client:127]: [0] [30:C9:22:A3:7E:CE] gattc registered app id 1
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] Found device
[D][esp32_ble_tracker:671]: Found device 30:C9:22:A3:7E:CE RSSI=-57
[D][esp32_ble_tracker:692]:   Address Type: PUBLIC
[D][esp32_ble_tracker:694]:   Name: 'FR:ID16RE000939'
[D][esp32_ble_tracker:219]: Pausing scan to make connection...
[D][esp32_ble_tracker:219]: Pausing scan to make connection...
[D][esp32_ble_tracker:219]: Pausing scan to make connection...
[I][esp32_ble_client:067]: [0] [30:C9:22:A3:7E:CE] 0x00 Attempting BLE connection
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 40
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=40 gattc_if=4
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] ESP_GATTC_CONNECT_EVT
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 2
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=2 gattc_if=4
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] ESP_GATTC_OPEN_EVT
[I][radon_eye_rd200:015]: Connected successfully!
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 46
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=46 gattc_if=4
[D][esp32_ble_client:306]: [0] [30:C9:22:A3:7E:CE] Event 46
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=7 gattc_if=4
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=7 gattc_if=4
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=7 gattc_if=4
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 6
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=6 gattc_if=4
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] ESP_GATTC_SEARCH_CMPL_EVT
[V][esp32_ble_client:223]: [0] [30:C9:22:A3:7E:CE] Service UUID: 0x1801
[V][esp32_ble_client:225]: [0] [30:C9:22:A3:7E:CE]  start_handle: 0x1  end_handle: 0x5
[V][esp32_ble_client:223]: [0] [30:C9:22:A3:7E:CE] Service UUID: 0x1800
[V][esp32_ble_client:225]: [0] [30:C9:22:A3:7E:CE]  start_handle: 0x14  end_handle: 0x1c
[V][esp32_ble_client:223]: [0] [30:C9:22:A3:7E:CE] Service UUID: 0x1523
[V][esp32_ble_client:225]: [0] [30:C9:22:A3:7E:CE]  start_handle: 0x28  end_handle: 0xffff
[I][esp32_ble_client:227]: [0] [30:C9:22:A3:7E:CE] Connected
[V][esp32_ble_client:069]: [0] [30:C9:22:A3:7E:CE]  characteristic 0x1524, handle 0x2a, properties 0x6
[V][esp32_ble_client:069]: [0] [30:C9:22:A3:7E:CE]  characteristic 0x1525, handle 0x2c, properties 0x12
[V][esp32_ble_client:069]: [0] [30:C9:22:A3:7E:CE]  characteristic 0x1526, handle 0x2f, properties 0x12
[V][radon_eye_rd200:147]: writing 0x50 to write service
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 18
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=18 gattc_if=4
[D][esp32_ble_client:188]: [0] [30:C9:22:A3:7E:CE] cfg_mtu status 0, mtu 517
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 9
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=9 gattc_if=4
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] ESP_GATTC_WRITE_DESCR_EVT
[V][esp32_ble:359]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[V][esp32_ble_client:121]: [0] [30:C9:22:A3:7E:CE] gattc_event_handler: event=3 gattc_if=4
[D][esp32_ble_client:110]: [0] [30:C9:22:A3:7E:CE] ESP_GATTC_READ_CHAR_EVT
[V][radon_eye_rd200:081]: result bytes: 0000 00000000 00000000 00000000 0000 0000 0000
```

baldisos, 2024-10-17:
I tried the same thing, but i can't wrap my head around how the decoding works. I can't get this to work.
The data is read, but as you say, it's not decoded properly.

ckuethe, 2024-10-17:
first problem is [radon_eye_rd200.cpp#L148](https://github.com/esphome/esphome/blob/dev/esphome/components/radon_eye_rd200/radon_eye_rd200.cpp#L148) where it's sending a `0x50` to poll the device and according to [KNOWLEDGE_V2.md](https://github.com/sormy/radoneye/blob/main/KNOWLEDGE_V2.md) the newer version seems to use `0x40`.

after I 3d print some round tuits, I'm going to see about adding another check in [here](https://github.com/esphome/esphome/blob/dev/esphome/components/radon_eye_rd200/radon_eye_rd200.cpp#L25) to try detect characteristics with the new UUIDs and use that to set some sort of version flag. Once the protocol version is known, the right command can be send and its response can be decoded. of course, there will need to be a decoder written for the new protocol...

ckuethe, 2024-10-23:
Yay, a different error! I'm still not getting meaningful data from my RD200 - getting back a 20-byte message full of `0x00`, but at least I can now make my detector beep when the ESP32 connects.

baldisos, 2024-10-29:
I wonder if @sormy could be of help here. This is beyond my capabilities.

ckuethe, 2024-10-31:
Sure, that sounds helpful.interesting

My RD200V3 works with https://github.com/sormy/radoneye and https://github.com/sormy/radoneye-reader - I'm able to poll the full measurements as described in https://github.com/sormy/radoneye/blob/main/KNOWLEDGE_V2.md

sormy, 2024-11-09:
Here is a branch: https://github.com/sormy/esphome/tree/rd200v2
Feel free to polish if needed

paukstis, 2024-12-09:
Hi, thanks @sormy for the branch, testing it, and I'm stuck with last message from radon_eye_rd200 saying "writing 0x40 to write service", and then silence, nothing more happens.

With radoneye-reader it works, I can see all sensor data.

Here is the log from testing the branch, I added a few custom "debug1" messages in the code, just to see a bit more what's going on, any idea where I could start digging further ?

```
[07:26:14][D][esp32_ble:292]: Enabling BLE...
[07:26:15][D][esp32_ble_tracker:273]: Starting scan...
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 2
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 7
[07:26:15][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 0
[07:26:15][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=0 gattc_if=4
[07:26:15][V][esp32_ble_client:127]: [0] [24:4C:AB:E6:FF:86] gattc registered app id 1
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:15][D][esp32_ble_client:110]: [0] [24:4C:AB:E6:FF:86] Found device
[07:26:15][D][esp32_ble_tracker:677]: Found device 24:4C:AB:E6:FF:86 RSSI=-73
[07:26:15][D][esp32_ble_tracker:698]:   Address Type: PUBLIC
[07:26:15][D][esp32_ble_tracker:700]:   Name: 'FR:GJ17RE000945'
[07:26:15][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[07:26:15][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[07:26:15][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 18
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 18
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 18
[07:26:15][I][esp32_ble_client:067]: [0] [24:4C:AB:E6:FF:86] 0x00 Attempting BLE connection
[07:26:15][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 40
[07:26:15][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=40 gattc_if=4
[07:26:15][D][esp32_ble_client:110]: [0] [24:4C:AB:E6:FF:86] ESP_GATTC_CONNECT_EVT
[07:26:15][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 2
[07:26:15][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=2 gattc_if=4
[07:26:15][D][esp32_ble_client:110]: [0] [24:4C:AB:E6:FF:86] ESP_GATTC_OPEN_EVT
[07:26:15][I][radon_eye_rd200:031]: Connected successfully!
[07:26:15][D][esp32_ble_tracker:273]: Starting scan...
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 2
[07:26:15][V][esp32_ble:339]: (BLE) gap_event_handler - 7
[07:26:16][V][esp32_ble:339]: (BLE) gap_event_handler - 20
[07:26:16][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:16][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 46
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=46 gattc_if=4
[07:26:17][D][esp32_ble_client:306]: [0] [24:4C:AB:E6:FF:86] Event 46
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=7 gattc_if=4
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=7 gattc_if=4
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 7
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=7 gattc_if=4
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 6
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=6 gattc_if=4
[07:26:17][D][esp32_ble_client:110]: [0] [24:4C:AB:E6:FF:86] ESP_GATTC_SEARCH_CMPL_EVT
[07:26:17][V][esp32_ble_client:223]: [0] [24:4C:AB:E6:FF:86] Service UUID: 0x1801
[07:26:17][V][esp32_ble_client:225]: [0] [24:4C:AB:E6:FF:86]  start_handle: 0x1  end_handle: 0x5
[07:26:17][V][esp32_ble_client:223]: [0] [24:4C:AB:E6:FF:86] Service UUID: 0x1800
[07:26:17][V][esp32_ble_client:225]: [0] [24:4C:AB:E6:FF:86]  start_handle: 0x14  end_handle: 0x1c
[07:26:17][V][esp32_ble_client:223]: [0] [24:4C:AB:E6:FF:86] Service UUID: 0x1523
[07:26:17][V][esp32_ble_client:225]: [0] [24:4C:AB:E6:FF:86]  start_handle: 0x28  end_handle: 0xffff
[07:26:17][I][esp32_ble_client:227]: [0] [24:4C:AB:E6:FF:86] Connected
[07:26:17][V][esp32_ble_client:069]: [0] [24:4C:AB:E6:FF:86]  characteristic 0x1524, handle 0x2a, properties 0x6
[07:26:17][V][esp32_ble_client:069]: [0] [24:4C:AB:E6:FF:86]  characteristic 0x1525, handle 0x2c, properties 0x12
[07:26:17][V][esp32_ble_client:069]: [0] [24:4C:AB:E6:FF:86]  characteristic 0x1526, handle 0x2f, properties 0x12
[07:26:17][W][radon_eye_rd200:077]: ***** ESP_GATTC_SEARCH_CMPL_EVT debug1, read_handle=44 write_handle=42
[07:26:17][W][radon_eye_rd200:194]: ***** esp_ble_gattc_register_for_notify debug1, 44 0
[07:26:17][I][radon_eye_rd200:180]: writing 0x40 to write service
[07:26:17][D][ble_client:058]: All clients established, services released
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 18
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=18 gattc_if=4
[07:26:17][D][esp32_ble_client:188]: [0] [24:4C:AB:E6:FF:86] cfg_mtu status 0, mtu 517
[07:26:17][V][esp32_ble:367]: (BLE) gattc_event [esp_gatt_if: 4] - 38
[07:26:17][V][esp32_ble_client:121]: [0] [24:4C:AB:E6:FF:86] gattc_event_handler: event=38 gattc_if=4
[... 6 lines cut ...]
[07:26:17][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[... cut ...]
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:21][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:22][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:22][V][esp32_ble:339]: (BLE) gap_event_handler - 3
[07:26:22][V][esp32_ble:339]: (BLE) gap_event_handler - 3
```

sormy, 2024-12-09:
I guess there is a bug in the code - response listener is attached after response has arrived.

Sam-C-Dev, 2024-12-14:
I took am having similar issues. But different errors. Posting here in case it helps. 
v3.0.1, purchased last week from the US Amazon Store. 
Name: 'FR:IF03RE000702'
Using a C3 board, not sure if that makes a different. I'll try a plain esp32 board soon, I just ordered some (I'm very new to esp32)

Important pieces from my config:
```
esphome:
  name: radoneyemonitor
  friendly_name: RadonEyeMonitor

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG # Required for the tracker to show the device

esp32_ble_tracker:

ble_client:
  - mac_address: 30:C9:22:E8:BD:4A
    id: radon_eye_ble_id

sensor:
  - platform: radon_eye_rd200
    ble_client_id: radon_eye_ble_id
    update_interval: 5min # default
    radon:
      name: "Radon"
      unit_of_measurement: "pCi/L"
      accuracy_decimals: 2
      filters:
        - lambda: return x / 37;
    radon_long_term:
      name: "Radon Long Term"
      unit_of_measurement: "pCi/L"
      accuracy_decimals: 2
      filters:
        - lambda: return x / 37;
```

In the logs:
```
[12:45:26][D][esp32_ble_client:110]: [0] [30:C9:22:E8:BD:4A] Found device
[12:45:26][D][esp32_ble_tracker:674]: Found device 30:C9:22:E8:BD:4A RSSI=-73
[12:45:26][D][esp32_ble_tracker:695]:   Address Type: PUBLIC
[12:45:26][D][esp32_ble_tracker:697]:   Name: 'FR:IF03RE000702'
[12:45:26][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[12:45:26][I][esp32_ble_client:067]: [0] [30:C9:22:E8:BD:4A] 0x00 Attempting BLE connection
[12:45:27][D][esp32_ble_client:172]: [0] [30:C9:22:E8:BD:4A] ESP_GATTC_DISCONNECT_EVT, reason 62
[12:45:27][W][radon_eye_rd200:021]: Disconnected!
[12:45:27][D][esp32_ble_client:110]: [0] [30:C9:22:E8:BD:4A] ESP_GATTC_OPEN_EVT
[12:45:27][W][esp32_ble_client:143]: [0] [30:C9:22:E8:BD:4A] Connection failed, status=133
[12:45:27][D][esp32_ble_tracker:273]: Starting scan...
[12:46:16][D][esp32_ble_client:110]: [0] [30:C9:22:E8:BD:4A] Found device
[12:46:16][D][esp32_ble_tracker:674]: Found device 30:C9:22:E8:BD:4A RSSI=-75
[12:46:16][D][esp32_ble_tracker:695]:   Address Type: PUBLIC
[12:46:16][D][esp32_ble_tracker:697]:   Name: 'FR:IF03RE000702'
[12:46:16][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[12:46:16][I][esp32_ble_client:067]: [0] [30:C9:22:E8:BD:4A] 0x00 Attempting BLE connection
[12:46:17][D][esp32_ble_client:172]: [0] [30:C9:22:E8:BD:4A] ESP_GATTC_DISCONNECT_EVT, reason 62
[12:46:17][W][radon_eye_rd200:021]: Disconnected!
```

paukstis, 2024-12-14:
Bluetooth events jungle is a mess 😱 .. I have code working now, I'll clean it a bit tomorrow and will do a PR to @sormy 's branch.

sormy, 2024-12-14:
It is definitely messed up code, i will have a follow up fix but I have no way to test it :-) so it might work or not

sormy, 2024-12-14:
@paukstis Feel free to submit PR to upstream, Im glad that this draft helped to make it work. I guess I have missing esp_ble_gattc_write_char_descr call to enable actual notification on handle.

paukstis, 2024-12-14:
I used `esp_ble_gattc_write_char` but in ESP_GATTC_WRITE_DESCR_EVT event to write `0x40` command, and for notifications, I did indeed used at first the `esp_ble_gattc_write_char_descr` writing `0x01` to trigger notifications, like in examples code, but then removed it after seeing it works without.

Also had to change read_sensors_ params to:
`read_sensors_(param->notify.value, param->notify.value_len);`

And to disable this line in ESP_GATTC_SEARCH_CMPL_EVT:
`//this->node_state = esp32_ble_tracker::ClientState::ESTABLISHED;`

sormy, 2024-12-14:
I think "is_valid_radon_value_" can be dropped, it was used as workaround for buggy previous implementation

paukstis, 2024-12-15:
@sormy current code, before you created branch V2, uses gattc_read_char, but branch V2 is now based on gattc_register_for_notify. In the PR maybe it's better to leave code for V1 as it is, based on gattc_read_char ?

My device is V2 and I have no way of testing the new code with V1.

paukstis, 2024-12-15:
@sormy hi, juste created PR to your branch, for your inspection, please have a look, if ok, I'll submit it to upstream. I left all V1 and V2 with _register_for_notify mode, maybe someone can retest it with V1 device, tests with V2 are ok.

s1r0x, 2024-12-27:
> Bluetooth events jungle is a mess 😱 .. I have code working now, I'll clean it a bit tomorrow and will do a PR to @sormy 's branch.

Hi @paukstis, 
where can i find the working code? i've the same problem regarding "esp_ble_gattc_get_descr_by_char_handle error, status=10"

paukstis, 2024-12-27:
I did a PR to ESPHome dev branch here: https://github.com/esphome/esphome/pull/7962

s1r0x, 2024-12-27:
thx found everything, but as i am not familar with git and as there are old files in dev branch as far as i can see, i took over your changes manually . Now everything is working

ckuethe, 2024-12-27:
@s1r0x you can reference this PR in your device config by including a chunk like this. 

```yaml
external_components:
  - source: github://pr#7962
    components: [ radon_eye_rd200 ]
```

https://esphome.io/components/external_components.html is useful.

baldisos, 2024-12-27:
I didn't have much time the last weeks and months, i just want to confirm that this technically works by referencing the PR with my FR:HG0 device. But i don't get any readings on the Long Term Entity.
[image]

ckuethe, 2024-12-27:
If you've rebooted your RD200 recently, you won't get long term measurements. Mine rebooted about 10 days ago and currently the display says the long term (monthly) radon is 0.000.

Maybe contact RadonEye support and ask them to update the firmware to estimate long term radon if there are enough short term measurements available?

baldisos, 2024-12-27:
Nah, there is a valid value. I'm currently reading it through a Python script and a RPi.

I just don't get it when using the ESP.

On the other hand it isn't that big of a problem because I'm calculating the mean values through HA anyways.

[image]

sormy, 2024-12-27:
Side note. Technically, RadonEye can be also used as smart alarm device given that it has alarm command.

paukstis, 2024-12-28:
@baldisos yep, and it's in the code, just not published, I updated now the PR with commit that will push daily average if monthly is not there yet.

baldisos, 2024-12-28:
> @baldisos yep, and it's in the code, just not published, I updated now the PR with commit that will push daily average if monthly is not there yet.

Yes, this works! Excellent work!

DelSerag, 2025-01-25:
Hi! Is there any way to implement this using Arduino? I'm trying to connect my ESP32 to the to RD200 V3.0.1

cubertt, 2025-03-18:
> [@s1r0x](https://github.com/s1r0x) you can reference this PR in your device config by including a chunk like this.
> 
> external_components:
>   - source: github://pr#7962
>     components: [ radon_eye_rd200 ]
> 
> https://esphome.io/components/external_components.html is useful.

This works with my V3 Device. Will this be included in the standard ESPhome code someday?
My code looks like that on an ESP32 lan device
`logger:
  level: VERBOSE
external_components:
  - source: github://pr#7962
    components: [ radon_eye_rd200 ]  
#esp32_ble_tracker:
#radon_eye_ble: 
  # Mac adresse ist: 24:4C:AB:33:56:1A
#esp32_ble_tracker:
#radon_eye_ble:
#  ble_client_id: radon_eye_ble_id
esp32_ble_tracker:
  #scan_parameters:
  #  interval: 110ms
 #   window: 99ms
   # active: true 
ble_client:
  - mac_address: 00:3D:AA:32:26:1A # change to your Adress
    id: radon_eye_ble_id
 #   auto_connect: true
    #passive: false
sensor:
  - platform: radon_eye_rd200
    ble_client_id: radon_eye_ble_id
    update_interval: 5min # default
    radon:
      name: "Radon"
   #   unit_of_measurement: "pCi/L"
      unit_of_measurement: "Bq/m³"
      accuracy_decimals: 2
 #     filters:
 #       - lambda: return x / 37;
    radon_long_term:
      name: "Radon Long Term"
    #  unit_of_measurement: "pCi/L"
      unit_of_measurement: "Bq/m³"
      accuracy_decimals: 2
 #     filters:
#        - lambda: return x / 37;`

jruben4, 2025-06-17:
@sugarpines anything special I need to do for the initial Bluetooth address discovery phase with a v3.0.1 device?

The recommended code block is not finding any Bluetooth devices.

`logger:
  level: DEBUG # Required for the tracker to show the device

esp32_ble_tracker:
radon_eye_ble:`

sugarpines, 2025-06-17:
You need to specify your device’s MAC address as shown above

ble_client:
  - mac_address: your-address

your-address can be found by following Device Discovery instructions at https://esphome.io/components/sensor/radon_eye_ble.html

The ESPHome log then contains the MAC address. If none are found, then I suspect either the radon sensor is not sending it or the ESP device is not receiving it. You can check sending by using the radon sensor’s app. Receiving by bringing other known BLE devices near your ESP.

jruben4, 2025-06-18:
I'm following those instructions - see the code I posted - I get no BLE hits - even when I am accessing the RD200 via it's app.  I would think I would see all sorts of BLE in the area - iphones, etc.

rbertking, 2025-08-25:
works for me too

`  - source: github://pr#7962
    components: [ radon_eye_rd200, radon_eye_ble ]`

docpayce, 2026-01-18:
Few months later, I just came across this thread and would like to strongly encourage a merge this into main. I was only able to get the RD200 up and running after using 
`  - source: github://pr#7962
    components: [ radon_eye_rd200 ]`

Would be great for the module to work out of the gate.