esphome/issues #6307: ILI9xxx breaks deep sleep

TobiasPSP, 2024-10-06:
### The problem

Using the ILI9xxxx display driver breaks deep sleep on ESP32: 

Once the driver has been initialized - since there is no way of sending it back to sleep - it keeps running even when esp32 is sent to deep sleep. 

This causes such ridiculously high power consumption that deep sleep is no longer making sense in battery-driven devices.

We investigated cases of ESP32 with built-in displays that consumed 9mA in deep sleep. The culprit was the ILI9xxx driver. 

Once we sent the 0x10 SLEEPIN command to the display via SPI *before* invoking deep sleep, the current consumption dropped from 9mA to 80uA.

Please add a way to send command bytes to the device, possibly similar to how you do it with the initialization bytes.  Or better yet, implement a Sleep() method for the display that internally sends the appropriate byte command.

Since ESP32 fully re-initialize after deep sleep, there is no need to issue a wakeup command.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ili9xxx

### Anything in the logs that might be useful for us?

```txt
Code works flawlessly, no exceptions, display looks good.
It is the **power consumption** that is concerning:
In deep sleep, the board consumes 9mA with the issue, and 80uA with the issue fixed.
```

### Additional information

The Adafruit display libraries on platformio implement this, and that's why they work flawlessly with deep sleep.

This should be a really low hanging fruit to fix as it basically is just sending a command byte to the device.

TobiasPSP, 2024-10-07:
I investigated further and managed to send a command code to the display like so:

````
- lambda: |-
          // Send display to sleep before deep sleep
          uint8_t command = 0x10;  
          id(display1).command(command);
````

This did turn off the display. However, deep sleep consumption just dropped from **9mA** to **2.1mA**. 

When I compare this to the control setup using custom C++ code with the Adafruit display library in platformio, I have no clue what causes this significant discrepancy (550% higher power consumption, reducing battery life to 20% of what platformio code can do). With platformio code, deep sleep consumption is just **370uA** after sending the display to sleep using the same control code.

So it cannot be the dev board hardware design. What else could be kept alive by *esphome*? I am testing a bare *TTGO T-Display board* with no peripherals connected. For consumption testing, I am using *Power Profiler Kit II*.

Bottom line is we need to find a way to get *esphome firmware* to the same deep sleep consumption that can be achieved by platformio code. Even a consumption of *2.1mA* is magnitudes too high for meaningful deep sleep.

Any ideas?

clydebarrow, 2024-10-07:
> Please add a way to send command bytes to the device

This already exists  and can be called in lambdas - e.g.

```
id(display_id).command(0x10);
```

The main issue would be where to call it from. But I raised a PR to send a SLPIN command on shutdown.

TobiasPSP, 2024-10-07:
Here's a quick write-up of the findings and steps to get deep sleep right with display boards: [https://done.land/components/microcontroller/families/esp/esp32/lilygot-display/t-display/programming/usingesphome/addingdeepsleep#efficient-deep-sleep](https://done.land/components/microcontroller/families/esp/esp32/lilygot-display/t-display/programming/usingesphome/addingdeepsleep#efficient-deep-sleep)

Maybe it helps someone save some time.

closed 2024-10-08 by jesserockz