esphome/issues #6311: ESP32 devices show offline, mDNS not working correctly

tomm1ed, 2024-10-07:
### The problem

Running ESPHome 2024.9.2 in a docker. 9 devices connected. The ones based on ESP8266 stay online/mDNS keeps on working, the ESP32 all drop off after some amount of time. When I reboot the ESP32 they work for some time again but then drop off again. Connection to HA keeps on working and when I put the [name].local in my /etc/hosts file the ESP32 are Updateable and Logs also work. There seems to be a difference between the ESP8266 and the ESP32 mDNS code because ESP8266 based boards stay online.

Running aioesphomeapi-discover from within the docker appears to show working mDNS:
```
ONLINE |vindriktning-keuken             |192.168.12.50  |e8db84d079c4|2024.8.3        |ESP8266   |d1_mini
ONLINE |p1-reader                       |192.168.12.14  |98cdac3191aa|2024.8.3        |ESP8266   |d1_mini
ONLINE |vindriktning-garage             |192.168.12.49  |e8db84dc374e|2024.8.3        |ESP8266   |d1_mini
ONLINE |everything-presence-lite-4f0df0 |192.168.12.241 |08a6f74f0df0|2024.9.2        |ESP32     |esp32dev
```

5 other ESP32 boards are offline and the EPL will be offline soon enough (was booted about 15 minutes ago), any idea how to fix (bar using the PING setting) Happens with both framework arduino and esp-idf. Devices do not go down, are pingable and usable in Home Assistant without issue.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

Mini D1 ESP8266, Custom Onju Voice PCB, Custom Everything Presence Lite board, M5 Stamp S3, 

### Example YAML snippet

```yaml
substitutions:
  name: btproxy-1
  friendly_name: btproxy-1

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: n

dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    name: Safe Mode Boot
    entity_category: diagnostic

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx="

ota:
  platform: esphome
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Btproxy-1 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.9.2
INFO Reading configuration /config/onju-voice-woonkamer.yaml...
INFO Starting log output from onju-voice-woonkamer.local using esphome API
WARNING Can't connect to ESPHome API for onju-voice-woonkamer.local: Error resolving IP address: [Errno -2] Name or service not known (APIConnectionError)

Directly after reboot of Onju Voice it shows up in the aioesphomeapi-discover
ONLINE |onju-voice-woonkamer            |192.168.12.16  |ecda3b5480e0|2024.8.3        |ESP32     |esp32-s3-devkitc-1

and logs work:
INFO ESPHome 2024.9.2
INFO Reading configuration /config/onju-voice-woonkamer.yaml...
INFO Starting log output from 192.168.12.16 using esphome API
INFO Successfully connected to onju-voice-woonkamer @ 192.168.12.16 in 0.060s
```

ssieb, 2024-10-08:
Can you resolve the mdns name from another computer on the network?

tomm1ed, 2024-10-08:
> Can you resolve the mdns name from another computer on the network?

Yes, tested from Ubuntu 22.04, Ubuntu 24.04, MacOS 15.0.1 and Windows Server 2019. All with the same ESP8266 work, ESP32 don't unless shortly after reboot.

Example from MacOS:
```
tom@MacBook-Air-van-Tom ~ % ping p1-reader.local
PING p1-reader.local (192.168.12.14): 56 data bytes
[...]
tom@MacBook-Air-van-Tom ~ % ping btproxy-1.local
ping: cannot resolve btproxy-1.local: Unknown host
```

Immediately after reboot of btproxy-1:
```
tom@MacBook-Air-van-Tom ~ % ping btproxy-1.local
PING btproxy-1.local (192.168.12.13): 56 data bytes
```

ssieb, 2024-10-08:
I've heard of this happening before, but it was a wifi or router firewall issue in that case.  I've never seen it happen and it hasn't come up on discord.  What is your wifi?

tomm1ed, 2024-10-08:
> I've heard of this happening before, but it was a wifi or router firewall issue in that case. I've never seen it happen and it hasn't come up on discord. What is your wifi?

I use Aruba InstantAP for WiFi and Fortigate as a firewall. 

Doesn't the fact that the ESP8266's (that are on the exact same network as the ESP32's) don't have the same issue suggest that there is a definitive difference between the two implementations and that the ESP32 one is somehow 'flawed' (for lack of a better term)?

(Also FWIW; my HomePods, AppleTVs, my Macs, my Shairport-sync Raspberry Pi's and my printer all also work without issue on the same WiFi using Bonjour/mDNS)

ssieb, 2024-10-08:
Clearly there's some sort of difference, but if no one else can reproduce the issue, it suggests something specific to your network.  Can you do any network sniffing to see what's happening?  I suggest coming to the esphome discord server for further discussion.

tomm1ed, 2024-10-08:
> 

> Clearly there's some sort of difference, but if no one else can reproduce the issue, it suggests something specific to your network. Can you do any network sniffing to see what's happening? I suggest coming to the esphome discord server for further discussion.

Definitely not the only one that has this, so here:
https://github.com/esphome/issues/issues/3003#issuecomment-1047746276

But tcpdumping on the docker host of the esphome installation shows me that all WiFi mDNS queries come from the AP that is the virtual controller. Following the link to https://github.com/espressif/esp-idf/issues/7453 
Is there any way to force esphome to enable the [CONFIG_MDNS_STRICT_MODE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mdns-strict-mode) option?

ssieb, 2024-10-08:
It looks to me like all the problems listed in that issue were due to network equipment.
As far as I can tell from the esp-idf link, it is already fixed in the current version.

tomm1ed, 2024-10-09:
> It looks to me like all the problems listed in that issue were due to network equipment. As far as I can tell from the esp-idf link, it is already fixed in the current version.

So it happens on Ubiquity, FritzBox and Aruba WiFi networks and only with ESPhome ESP32’s but the problem is with the network? (my network is riddled with mDNS devices and only the ESPHome ESP32s stop working)
This logically cannot be the conclusion. Is there anything I can do to supply you with info to get to the bottom of this?

randybb, 2024-10-09:
Was not reading linked issues, but on Ubiquity Unify is mDNS working fine, even between VLAN's - have been using only mDNS names (HA is connecting to esp8266 and esp32 MCUs via mDNS names, everything is connecting to mqtt and other services via their mDNS names too) and never had any issues.

ssieb, 2024-10-09:
I think everyone in that linked issue resolved their issues by fixing their network equipment or at least conclusively proved that it was the cause.  We don't know of any issues and can't reproduce it, so you'll have to find some proof that it's an esphome issue because there's nothing we can do without that.

tomm1ed, 2024-10-09:
> I think everyone in that linked issue resolved their issues by fixing their network equipment or at least conclusively proved that it was the cause. We don't know of any issues and can't reproduce [...]

So as I have also a lot of Tasmota devices I enabled mDNS on one of the ESP32 versions and they don't have the issue:

```
11:26:36.160  Add        2  11 local.               _esphomelib._tcp.    onju-voice-woonkamer
11:29:30.316  Rmv        0  11 local.               _esphomelib._tcp.    onju-voice-woonkamer
```
After the Remove nothing is heard from this device until I reboot it.

Tasmota32 works:
```
11:26:36.160  Add        2  11 local.               _http._tcp.          M5stack-temp-woonkamer
11:29:30.316  Rmv        0  11 local.               _http._tcp.          M5stack-temp-woonkamer
11:30:27.308  Add        2  11 local.               _http._tcp.          M5stack-temp-woonkamer
```

As do the ESP8266 ESPHome devices (and _e very other device_ that uses mDNS, from my Hikvision cams to my OTGW to my Laserjet to my Apple devices):
```
11:26:36.159  Add        3  11 local.               _esphomelib._tcp.    vindriktning-garage
11:29:30.316  Rmv        1  11 local.               _esphomelib._tcp.    vindriktning-garage
11:30:27.308  Add        3  11 local.               _esphomelib._tcp.    vindriktning-garage
```

Other examples:
```
11:26:36.160  Add        3  11 local.               _http._tcp.          HP LaserJet 500 MFP M525 [3DFDE6]
11:29:30.316  Rmv        1  11 local.               _http._tcp.          HP LaserJet 500 MFP M525 [3DFDE6]
11:30:27.308  Add        3  11 local.               _http._tcp.          HP LaserJet 500 MFP M525 [3DFDE6]
```

```
11:26:36.160  Add        3  11 local.               _http._tcp.          OTGW
11:29:30.316  Rmv        1  11 local.               _http._tcp.          OTGW
11:30:27.308  Add        3  11 local.               _http._tcp.          OTGW
```

So it is literally just ESPHome ESP32 devices that stop working after one announcement, so how can the conclusion not be that they do something different from every other device on my network.

ssieb, 2024-10-09:
Did you add the devices to HA by name or IP address?

I guess I need to clarify my previous comment.  It's not enough to know that esphome seems to be doing something different on your network, you need to know why or what.  Without that, there's nothing we can do.

tomm1ed, 2024-10-09:
> Did you add the devices to HA by name or IP address?
> 
> I guess I need to clarify my previous comment. It's not enough to know that esphome seems to be doing something different on your network, you need to know why or what. Without that, there's nothing we can do.

Some by IP, some automatically detected.

I am more than happy to try and get more info, tell me how I can help to get it the bottom of it and I’ll do my best.

ssieb, 2024-10-09:
If you added the device to HA by IP, then of course it's going to work because it doesn't need to do the MDNS lookup.
I don't really know how to test this with your hardware.  I use openwrt on all my APs, so I can just run tcpdump on the AP and find out what's happening.
Can another wifi device see the MDNS requests?
Can you take one of the devices to another network to see if it still happens there?
Do you have another AP you can setup independently to test with?

KodinLanewave, 2024-10-16:
I'd start by grabbing packet capture of your network when the ESPHome devices hop on your wifi via a tool such as wireshark; this will allow you to compare the mDNS hostname registration to see if it is different between esphome and other devices; my best guess here based on symptoms is that your network, like mine, is using dynamic dns (very short TTL's for DHCP lease, as well as DNS,) and esphome isn't periodically updating the DNS server with it's hostname once the IP lease expires.  Dev's, can you please verify my assumption?

DavesCodeMusings, 2024-10-25:
For what it's worth, I am experiencing the same issue with ESPHome and 8266-based Sonoff S31 smart plugs. Here's what I'm seeing:

* The smart plug shows online in ESPHome dashboard after it is first plugged in and powered up.
* It shows up in the output of avahi-browse (e.g. `avahi-browse -alrt | grep s31`).
* ESPHome can no longer resolve the smart plug name after a short time (maybe ten or fifteen minutes) when I try to fetch logs or update firmware, though it still shows as "online".
* After that time, avahi-browse shows the device (cached?), but complains "failed to resolve" after a "timeout reached".
* The smart plug is always controllable from home assistant. Automations and manual control work regardless of avahi-browse not resolving or ESPHome dashboard showing it as offline.
* When the smart plug is offline in the ESPHome dashboard, it no longer appears in the avahi-browse output, not even with a "timeout reached".
* All other non-ESPHome mdns devices show up in avahi-browse.

My setup:

* I am running ESPHome in a docker container, like the author of this issue.
* ESPHome and Home Assistant are both running with "network_mode: host"
* ESPHome and Home Assistant containers are from :latest, pulled Oct 24, 2024.
* Smart plugs updated to latest firmware as of Oct 24, 2024.
* Networking gear is Ubiquiti Unifi.
* Docker and avahi-browse are running on Alpine Linux x86_64 3.20.3 (apk updated on Oct 20, 2024.)
* ESPHome, Home Assistant, and smart plugs are all on the same default VLAN.
* No firewall between any of the devices.

After plugging in the smart plug, there is a short window of time (10 to 15 minutes) that I can interact with it normally in the ESPHome dashboard. After that...
* The smart plug still shows as online in the ESPHome dashboard for a while longer (cached info, maybe?)
* Attempting to fetch logs results in: "WARNING Can't connect to ESPHome API for s31-4.local: Error resolving IP address: [Errno -2] Name or service not known (APIConnectionError)"
* Output from avahi-browse shows the device, but fails to resolve (see below.)

```
$ avahi-browse -alrt | grep s31
+   eth0 IPv4 s31-4                                         _esphomelib._tcp     local
Failed to resolve service 's31-4' of type '_esphomelib._tcp' in domain 'local': Timeout reached
```

Interesting aside concerning the avahi-browse output above... I have five Sonoff s31 smart plugs. Only the s31-4 that I recently unplugged and re-inserted shows up. Any of the others are not in the avahi-browse output, show offline in ESPHome dashboard, but are still controllable in Home Assistant.

Current workaround:

Unplugging the smart plug and re-powering seems to work and the device appears "online" long enough to get logs or OTA update the firmware.

Other thoughts:

This seems like it's related to the firmware on the device, though why Home Assistant is not affected, I'm not sure. Perhaps it caches last known addresses for a longer period of time than avahi-browse.

I do not remember this happening before my recent docker pull of the latest ESPHome container and subsequent firmware updates on the devices. Though I did both updates in a short time span, so I can't say for sure which it was, if either.

If there's anything you would like me to try, I'm happy to post results here and help get this resolved. Otherwise, just chiming in to confirm the original poster's statements, let you know it's happening on ESP8266 as well, and suggest a workaround of unplugging/replugging.

DavesCodeMusings, 2024-10-25:
Additional information and a more effective workaround...

As a troubleshooting step, I rolled my ESPHome docker container back to image: 2024.9.2

Everything is fine in the ESPHome dashboard. Devices are showing up as online. `avahi-browse -alrt | grep s31` also shows all my Sonoff smart plug devices now.

I have not changed the device firmware at all, only the docker container image. Why this would affect avahi-browse, I have no idea.

Again, these are all 8266 devices. Perhaps if @tomm1ed were to roll back to an even earlier Docker container, it may fix the issue for their ESP32 devices.

DavesCodeMusings, 2024-10-25:
UPDATE: And now I'm back to using container image :latest with results similar to what I got by pinning it to 2024.9.2. So perhaps it is the act of restarting the container that makes the difference and not necessarily the version. Apologies to anyone who read this far expecting a solution.

tomm1ed, 2024-10-28:
> For what it's worth, I am experiencing the same issue with ESPHome and 8266-based Sonoff S31 smart plugs. Here's what I'm seeing:
> 
> * The smart plug shows online in ESPHome dashboard after it is first plugged in and powered up.
> * It shows up in the output of avahi-browse (e.g. `avahi-browse -alrt | grep s31`).
> * ESPHome can no longer resolve the smart plug name after a short time (maybe ten or fifteen minutes) when I try to fetch logs or update firmware, though it still shows as "online".
> * After that time, avahi-browse shows the device (cached?), but complains "failed to resolve" after a "timeout reached".
> * The smart plug is always controllable from home assistant. Automations and manual control work regardless of avahi-browse not resolving or ESPHome dashboard showing it as offline.
> * When the smart plug is offline in the ESPHome dashboard, it no longer appears in the avahi-browse output, not even with a "timeout reached".
> * All other non-ESPHome mdns devices show up in avahi-browse.
> 
> My setup:
> 
> * I am running ESPHome in a docker container, like the author of this issue.
> * ESPHome and Home Assistant are both running with "network_mode: host"
> * ESPHome and Home Assistant containers are from :latest, pulled Oct 24, 2024.
> * Smart plugs updated to latest firmware as of Oct 24, 2024.
> * Networking gear is Ubiquiti Unifi.
> * Docker and avahi-browse are running on Alpine Linux x86_64 3.20.3 (apk updated on Oct 20, 2024.)
> * ESPHome, Home Assistant, and smart plugs are all on the same default VLAN.
> * No firewall between any of the devices.
> 
> After plugging in the smart plug, there is a short window of time (10 to 15 minutes) that I can interact with it normally in the ESPHome dashboard. After that...
> 
> * The smart plug still shows as online in the ESPHome dashboard for a while longer (cached info, maybe?)
> * Attempting to fetch logs results in: "WARNING Can't connect to ESPHome API for s31-4.local: Error resolving IP address: [Errno -2] Name or service not known (APIConnectionError)"
> * Output from avahi-browse shows the device, but fails to resolve (see below.)
> 
> ```
> $ avahi-browse -alrt | grep s31
> +   eth0 IPv4 s31-4                                         _esphomelib._tcp     local
> Failed to resolve service 's31-4' of type '_esphomelib._tcp' in domain 'local': Timeout reached
> ```
> 
> Interesting aside concerning the avahi-browse output above... I have five Sonoff s31 smart plugs. Only the s31-4 that I recently unplugged and re-inserted shows up. Any of the others are not in the avahi-browse output, show offline in ESPHome dashboard, but are still controllable in Home Assistant.
> 
> Current workaround:
> 
> Unplugging the smart plug and re-powering seems to work and the device appears "online" long enough to get logs or OTA update the firmware.
> 
> Other thoughts:
> 
> This seems like it's related to the firmware on the device, though why Home Assistant is not affected, I'm not sure. Perhaps it caches last known addresses for a longer period of time than avahi-browse.
> 
> I do not remember this happening before my recent docker pull of the latest ESPHome container and subsequent firmware updates on the devices. Though I did both updates in a short time span, so I can't say for sure which it was, if either.
> 
> If there's anything you would like me to try, I'm happy to post results here and help get this resolved. Otherwise, just chiming in to confirm the original poster's statements, let you know it's happening on ESP8266 as well, and suggest a short-term workaround of unplugging/replugging.

For me it is just the ESP32s, ESP8266 always shows up on ESPHome, avahi on Linux, macOS and windows. Have not had time to deep dive yet (my workaround for the time being is add the .local addresses of the ESPHome ESP32 devices to /etc/hosts so I can at least update them wirelessly and see the logs) but glad that I am not alone in this.

jankornblum, 2024-11-22:
I can confirm this behaviour. First, i thought there might be anything wrong with my mDNS and VLAN setup, but that's not the case, because all other devices work fine! Even the ESP8266 works fine. But not ESP32. The ESP32 name resolution works just a few seconds after rebooting the device, then the name resolution gets lost.

Masterz69, 2024-12-03:
> I can confirm this behaviour. First, i thought there might be anything wrong with my mDNS and VLAN setup, but that's not the case, because all other devices work fine! Even the ESP8266 works fine. [...]

Just reflashed ESP-WROOM-32 device with 2024.10.1 ESP-IDF framework. Previously it used Arduino for ~year.
And now it not provide mDNS information.

Update: flashed Arduino framework again - mDNS works.

kelvtech-co-uk, 2024-12-09:
So my problem statment was almost identical except for me it was both ESP32 and ESP8266 chips which were affected in the same way @tomm1ed described.  I use a Unifi NanoHD access point with a Unifi Dream Machine Special Edition.  I enabled IGMP Snooping on both my networks configured in UDM, mDNS was already enabled but this still experienced the issues.

Resolution was to reboot the AP, I'm anticpating this might be temporary, its been working fine now for 24+ hours after fighting it days looking for the problem.

FWIW prior to rebooting the AP I was convinced this was ESPHome as other devices in my LAN use mDNS which worked then and now flawlessly e.g. a Denon AVR and my HP printer.  Both these devices are in different subnets but I could resolve them from each so it was just the ESP devices I was struggling with.  I was reading the other issues where Unifi kept popping up which led me to my solution.

I run ESPHome as a docker container at version 2024.11.1

jankornblum, 2024-12-09:
Im am quite sure that ESP firmware itself is the issue. As already stated above, other devices run fine and their mDNS resolution works reliable over subnets. Especially I can confirm this for an Onkyo TX-8150 audio receiver (don’t know its chip) as well as for Shelly relais devices (last one is ESP8266).

rmaeder, 2024-12-21:
Similar here. A couple of weeks ago more and more of the devices showed "OFFLINE", and were not reachable via mDNS.
The only two that always worked were two Athom smart plugs, based on ESP8266. All the others were ESP32, some Apollo MSR-1, EPLite, m5stack-atom-lite, m5stamp-c3u, and the like.
Sometimes they would be OK for a few minutes after restart, but never for very long.

Based on the suspected relation with the Wifi AP, I rebooted mine, a TP-Link Archer C6, with OpenWRT 23.05.5, and then all devices showed back up, I was able to do a "update all"!
So, let me see for how much longer things are working. Definitely very strange.

jankornblum, 2024-12-21:
I forgot to mention that the issue exists for WiFi devices (Apollo AIR-1) as well as PoE devices (Olimex ESP32 PoE). 

The working ESP8266 devices are WiFi (Shelly) as well as LAN (Onkyo audio receiver).

Metallica22, 2024-12-23:
Hi everyone,  

I’m facing an issue where several ESPHome devices appear as “online” in ESPHome but frequently show as “offline” in Home Assistant.  

### Setup  
- **Devices:**  
  - ESP8266 (relay board with 8 relays)  
  - ESP8266 (standalone)  
  - ESP32 (temperature sensors)  
  - ESP32 (Bluetooth proxy)  
  - ESP32 Ethernet (Bluetooth proxy)  
- **Network:** UniFi with multiple VLANs (correctly configured; the issue started later).  

### Issues  
- Devices, especially both ESP8266s and one ESP32, regularly go offline.  
- One ESP32 became more stable after being reflashed via USB, but the issue persists for others.  

### Attempted Solutions  
1. **Restarting Access Points:** Provides temporary stability (minutes to hours).  
2. **Multiple OTA installations:** No noticeable improvement.  
3. **Static IPs in the ESP code:**  
   - OTA installations ran faster.  
   - Devices stayed online for over 24 hours but eventually went offline again.  

### Current Status  
I was on ESPHome version **2024.12.2** and recently downgraded to **2024.9.2**. Long-term results are not yet available.  

### Questions  
- Has anyone experienced similar issues?  
- Any recommendations for ESPHome in VLAN setups or with UniFi networks?  

Thanks for your help!

tmchow, 2024-12-30:
I've been pulling my hair out all day today trying to resolve this. The "offline" issue impacted a few of my devices about a month ago, and I never got around to digging into it until now. After mucking with it for hours and hours, I finally stumbled upon this issue.  My 4 devices have been working flawless for 6+ months, then as I said, about a month ago 2 of them went offline and nothing I could do could fix them.  My Unifi dash shows they are connected and I can browse their ESP web UI via IP but mDNS was failing.

One of my Sonoff S31 was working and one was failing.  I reflashed the working on, and it suddenly doesn't work anymore.

It does seems like there is something in the ESP firmware that's impacting these devices.  For the records, it's impacitng me for Sonoff S31 (where I'm using `esp32dev` in the board config) and an ESP01 device (where i'm using `esp01_1m` for the config).

Metallica22, 2024-12-30:
> Hi everyone,
>
> I’m facing an issue where several ESPHome devices appear as “online” in ESPHome but frequently show as “offline” in Home Assistant.
>
> ### Setup
> - **Devices:**
>   - ESP8266 (relay [...]

Hi everyone,  

Update: After downgrading to version 2024.9.2, everything worked perfectly, and my devices stayed online consistently.  

However, since I didn’t want to stick with the old version, I ran some additional tests on the latest version (2024.12.2) with help from Simon42’s forum (a German YouTuber).  

It has now been running for two days without any issues on all my devices (even with three additional devices added via Tuya Cloudcutter).  

### Current Configuration  

Here’s my updated WiFi section from ESPHome:  

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  power_save_mode: NONE
  reboot_timeout: 900s
  manual_ip:
    static_ip: 10.10.20.8
    gateway: 10.10.20.1
    subnet: 255.255.255.0

I know that the `power_save_mode` section doesn’t actually do anything since it wasn’t configured before, but I included it regardless.  

Additionally, in the ESPHome Add-on configuration, I enabled the optional setting:  
USE PING for Status

### Observations  

What exactly fixed the issue, I’m not entirely sure, as I didn’t have the time to isolate the root cause. However, everything is now running without any problems.  

Hope this helps others experiencing similar issues!

ssieb, 2024-12-30:
@Metallica22 you've just bypassed MDNS entirely.  You set a static address and told the dashboard to use ping.

AndreKR, 2025-01-04:
FWIW I had the `Name or service not known` error today after updating from `esphome/esphome:2024.9.1` to `esphome/esphome:2024.12.2`. Initially I thought something had changed in the new version, but after rolling back to the old version the problem was still there. I then rebooted the Pi and now it works again, in both versions. 🤷

Metallica22, 2025-01-04:
@ssieb Okay, you're right. But at least it's a quick solution. My entire heating control runs via an ESP relay board. In winter it's bad if this fails.

@AndreKR How long has it been running stable since the reboot?

> What I noticed is that the mDNS version that is written to the ESP is always the same for versions 9.x and 12.x.

AndreKR, 2025-01-04:
@Metallica22 

> But at least it's a quick solution. My entire heating control runs via an ESP relay board. In winter it's bad if this fails.

You don't need the dashboard to be able to connect unless you want to make config changes, the rest will still work. If Home Assistant suffers from the same issue and you're using the native API, *that* might be a problem. I recommend you change to MQTT, it's rock solid and doesn't rely on stupid mDNS.

> How long has it been running stable since the reboot?

Well, I only rebooted yesterday and it still works.

On the same Pi I sometimes (every couple of months) have an issue where it stops accepting RAs. Since both RAs and mDNS are based on broadcast packets I suspect that the issues might be related. If that is true, then I expect mDNS to work for a couple of months. But as I said above, I wouldn't trust mDNS with anything even slightly critical.

tomm1ed, 2025-01-05:
Just a quick update, updated a couple of ESP32's and ESP8266 to  2012.12.2 to see if it's resolved but the issue persists. ESP8266 has working mDNS, ESP32 stops working after the first rmv 
(see https://github.com/esphome/issues/issues/6311#issuecomment-2401830550 )
As stated before all mDNS devices (Apple devices, Laserjet, Tasmota, ESPHome ESP8266) except the ESPHome ESP32 work.

sspaceman, 2025-01-18:
For the record / FWIW:
I'm seeing this problem with various ESP devices with esphome firmware.
Other platforms like Tasmota / Linux / printers has perfect mdns working.
Unifi network (UAP AC HD)

jrm523, 2025-01-19:
I have also had a lot of issues with esphome mdns on my network running opnsense firewall with unifi pro 6 WAPs. I have mdns repeater enabled on my firewall and other mdns devices work fine. However, esphome rarely ever works. I have not opened an issue because I have been slammed and have not had time to dig deeply into it. However, the earlier comment prompted me to post because it insinuated the OP was the only one with the issue.

ssieb, 2025-01-19:
The common thread seems to be Unifi across the various issues that have been filed about this.

jankornblum, 2025-01-20:
I don't think Unify itself is the problem. There are others here (like me) who don't use Unify at all. Also, VLANs themselves don't seem to be the problem. I have a fully working VLAN setup where the MDNS works fine for other devices (ESP8266 and others) but not for ESP32 devices.

jankornblum, 2025-01-20:
Oh, sorry, I forgot to mention that my working esp 8266 devices are not under esphome (shelly switch, onkyo receiver etc.), so I don’t know which implementation they are based on. But this assures that my network setup is fine. Under esphome I only have the esp 32 devices, where mdsn doesn’t work correctly.

Masterz69, 2025-01-20:
> I don't think Unify itself is the problem. There are others here (like me) who don't use Unify at all. Also, VLANs themselves don't seem to be the problem. I have a fully working VLAN setup where the MDNS works fine for other devices (ESP8266 and others) but not for ESP32 devices.

Which framework using on ESP32 ?
Me tried ESP-IDF once and that was a device disappeared from mDNS after couple of minutes.

Have ~20 ESP32 devices on Arduion framework in one segment (mesh by 3x Keenetic AP): almost no issues with mDNS.
Twice it was due to something on Keenetic, solved by AP restart.
Another time due to WiFi channel was overloaded, solved by channel change. 
That's all in 2 years, actually.

slongano, 2025-01-26:
BTW, I checked also with avahi, and the resolver times almost always. For the other devices the response is immediate.
There are some problems with ping too.
'''
100 packets transmitted, 99 received, 1% packet loss, time 99171ms
rtt min/avg/max/mdev = 19.086/74.192/213.885/32.424 ms
'''
The following is the response from another device with no mDNS problems.
'''
100 packets transmitted, 99 received, +3 duplicates, 1% packet loss, time 99135ms
rtt min/avg/max/mdev = 4.108/9.690/88.802/9.667 ms
'''
On average the ping response is 8 times slower.
Maybe its a problem with the network code that it's slowed by other tasks, like display rendering.

erkr, 2025-02-26:
I experience similar issues. I a a bunch of ESP32 C3 devices. After initial preparation (using esphome web tool over USB), discovery by the builder works. 
Than I flashed a Bluetooth proxy configuration using the esp-idf typr. From then on the devices are reported offline but the builder

bdraco, 2025-03-08:
You should be able to watch locally with https://github.com/python-zeroconf/python-zeroconf/blob/master/examples/async_browser.py to figure out if its an mDNS network issue or something preventing the mDNS traffic from getting to the container

Change https://github.com/python-zeroconf/python-zeroconf/blob/ea6905b1a0e6122e1baa0cbb39db1cb91ec0f310/examples/async_browser.py#L67 to `services = ["_esphomelib._tcp.local."]`

It might be a bit moot in 2025.3.x since we will have ping auto-fallback if mDNS doesn't work

kelvtech-co-uk, 2025-03-09:
> You should be able to watch locally with https://github.com/python-zeroconf/python-zeroconf/blob/master/examples/async_browser.py to figure out if its an mDNS network issue or something preventing [...]

For me when the problem presents I can still access the esp devices via their IP, its just mDNS resolution for esphome devices which doesn't work, mDNS for my HP printer or Denon AVR still works fine all the time.

I did try your suggestion, running async_browser.py doesn't see any esphome services until I've rebooted my Unifi AP and then esp using mDNS works for a period-of-time/until-some-unknown-event.

tomm1ed, 2025-03-09:
> You should be able to watch locally with https://github.com/python-zeroconf/python-zeroconf/blob/master/examples/async_browser.py to figure out if its an mDNS network issue or something preventing [...]

If you check my previous posts you can see that only ESP32 based ESPHOME devices (including S2 and S3 and now also confirmed with C3 variant because I got one of those as well) have a non functioning mDNS. Any other device (including Tasmota ESP32 devices!) work without a hitch, from my Apple devices to the HP printer to the Pioneer Receiver as well as the ESP8266 ESPHome devices work. So there is something in the ESPHome ESP32 implementation that does not work as expected in my network. 

See 
https://github.com/esphome/issues/issues/6311#issuecomment-2399362088 
And
https://github.com/esphome/issues/issues/6311#issuecomment-2401830550

jankornblum, 2025-03-09:
I totally agree! So please read previous comments first.

bdraco, 2025-03-09:
Which version of ESP-IDF are you using?

bdraco, 2025-03-09:
> So there is something in the ESPHome ESP32 implementation that does not work as expected in my network.

@tomm1ed 
It would be helpful if you tried what I suggested above in https://github.com/esphome/issues/issues/6311#issuecomment-2708521632 as it will help narrow down if its a problem in zeroconf, or a problem in ESP stack.

bdraco, 2025-03-09:
@kelvtech-co-uk 
> I did try your suggestion, running async_browser.py doesn't see any esphome services until I've rebooted my Unifi AP and then esp using mDNS works for a period-of-time/until-some-unknown-event.

Thanks for testing. Can you check the settings for your Unifi WiFi network. Here is a screenshot of known working settings for an IoT network:

[image]

erkr, 2025-03-09:
@bdraco 

I lack the knowledge to test your fix. If I add this to my yaml in  esphome builder, validation fails: 
```
external_components:
  - source: github://pr#8378
    components: [ mdns ]
    refresh: 0s
``` 
Sorry

bdraco, 2025-03-09:
> 
> I lack the knowledge to test your fix. If I add this to my yaml in esphome builder, validation fails:
>

Can you provide some details on how it fails validation / the error message shown?

bdraco, 2025-03-09:
> Mines slightly different, I only have a single AP so I disable `BSS Transition` and given I use a dedicated network for IoT devices I enable `UAPSD`. If you have a strong feeling over the settings I have being incompatible I'll schedule a test?

All looks good there.  Which firmware are you using on the AP itself?  Some of the older 7.x unifi releases drop multicast traffic after they have been running for an extended period of time.  I haven't seen the problem with the recent 8.x releases though.

kelvtech-co-uk, 2025-03-09:
> All looks good there. Which firmware are you using on the AP itself? Some of the older 7.x unifi releases drop multicast traffic after they have been running for an extended period of time. I [...]

So my AP is an old `UAP-nanoHD`, its running v6.6.78 which looks to be the latest version.  Now you should know, if you tell me that its possible/likely the old hardware+firmware are a concern then this will immediately valid my unqualifed urge to upgrade which in turn will give me no end of grief from my better half!

bdraco, 2025-03-09:
> So my AP is an old `UAP-nanoHD`, its running v6.6.78 which looks to be the latest version. Now you should know, if you tell me that its possible/likely the old hardware+firmware are a concern then [...]

I have some U6+'s and AC-PRO running that firmware version (AC-PROs using 6.6.77 actually) without an issue so sad to report that its probably not a good justification for upgrading.  The earlier 7.x series releases were the ones I saw all the painful multicast issues.

kelvtech-co-uk, 2025-03-09:
> I have some U6+'s and AC-PRO running that firmware version (AC-PROs using 6.6.77 actually) without an issue so sad to report that its probably not a good justification for upgrading. The earlier 7.x series releases were the ones I saw all the painful multicast issues.

You build me up and then you let me down hard!

> Are you able to try the potential fix in [esphome/esphome#8378](https://github.com/esphome/esphome/pull/8378) ?

Yes, assuming though as you haven't stated a trigger/cause for the problem this will just be an upgrade and wait for failure scenario?  I've experience working mDNS for my esphome devices for 10+ days before its stopped, any suggestions for monitoring and/or stressing it to failure quicker?

bdraco, 2025-03-09:
> You build me up and then you let me down hard!
>
>
> Yes, assuming though as you haven't stated a trigger/cause for the problem this will just be an upgrade and wait for failure scenario? I've [...]

10+ days could mean something is running out of memory so it could be the ESP or unifi AP (so maybe not out of the woods yet on that one)

Can you add a memory sensor to the ESP config and see if its slowly running out of RAM?

```yaml
sensor:
  - platform: template
    name: Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    unit_of_measurement: 'B'
    state_class: measurement
```

bdraco, 2025-03-09:
aioesphomeapi 29.5.1 just finished uploading https://github.com/esphome/aioesphomeapi/actions/runs/13751922938 so you can also try `aioesphomeapi-discover --verbose`. If your screen is flooded with traffic, its also possible you have something overloading multicast and traffic is getting dropped.  If there is too much mDNS traffic on the network, the ESP will likely not be able to see the question and answer.

kelvtech-co-uk, 2025-03-09:
@bdraco okay so ESPHome container bumped to 2025.2.2 and I've upgraded 2 devices, all very quick and simples.  I've also added [esphome/esphome#8378](https://github.com/esphome/esphome/pull/8378) and a free mem sensor to my ESP32 device.  I'm not sure if I referenced previously in much older post but my mDNS issue affects both ESP32 and ESP8266.  

My ESP8266 device is at 2025.2.2 but I've not modified the config as I don't think [esphome/esphome#8378](https://github.com/esphome/esphome/pull/8378) targets it?

My ESP32 device is resolvable via mDNS and its web server now shows the free memory on the device...  

[image]

My experiences of this bug are that my ESP32 and ESP8266 devices either both work or both do not so I guess the results here might be interesting...or not!

> aioesphomeapi 29.5.1 just finished uploading https://github.com/esphome/aioesphomeapi/actions/runs/13751922938 so you can also try aioesphomeapi-discover --verbose after upgrading with pip install --upgrade aioesphomeapi. If your screen is flooded with traffic, its also possible you have something overloading multicast and traffic is getting dropped. If there is too much mDNS traffic on the network, the ESP will likely not be able to see the question and provide the answer.

Given I'm using a container I'm assuming there is no way for me to force this upgrade and test?  I've only the 2 ESP devices in my network currently but I do have more mDNS clients, maybe 15+.

bdraco, 2025-03-09:
> [@bdraco](https://github.com/bdraco?rgh-link-date=2025-03-09T20%3A38%3A32.000Z) okay so ESPHome container bumped to 2025.2.2 and I've upgraded 2 devices, all very quick and simples. I've also added [...]

Yeah, we have at least 3 distinct potential issues going on in this issue report.   Still it will be useful to know if the new mdns stack from espressif that https://github.com/esphome/esphome/pull/8378 uses fixes the issue for the ESP32 side.  If not, I'm likely the AP a bit more as the cause for you.

> 
> > aioesphomeapi 29.5.1 just finished uploading [esphome/aioesphomeapi/actions/runs/13751922938](https://github.com/esphome/aioesphomeapi/actions/runs/13751922938) so you can also try aioesphomeapi-discover --verbose after upgrading with pip install --upgrade aioesphomeapi. If your screen is flooded with traffic, its also possible you have something overloading multicast and traffic is getting dropped. If there is too much mDNS traffic on the network, the ESP will likely not be able to see the question and provide the answer.
> 
> Given I'm using a container I'm assuming there is no way for me to force this upgrade and test? I've only the 2 ESP devices in my network currently but I do have more mDNS clients, maybe 15+.

You can install `aioesphomeapi` on your local system (I don't know what OS you are running though) and run `aioesphomeapi-discover --verbose` as its already available on PyPI.

kelvtech-co-uk, 2025-03-09:
> aioesphomeapi 29.5.1 just finished uploading https://github.com/esphome/aioesphomeapi/actions/runs/13751922938 so you can also try aioesphomeapi-discover --verbose after upgrading with pip install [...]

So I let this run for just over a minute and it generated 185 lines of output, not really sure how to interpret this.  Most lines look like google/chomecast chats but a 6 stand out...

`2025-03-09 21:28:58.741 DEBUG    Ignoring duplicate message with no unicast questions received from ('127.0.0.1', 5353) [socket 6 (('0.0.0.0', 5353))] (122 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x02\x00\x00\x00\x00\x0b_esphomelib\x04_tcp\x05local\x00\x00\x0c\x00\x01\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x93\x00\x1c\x19localbytes-plug-pm-8ca2b3\xc0\x0c\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x93\x00\x1e\x1btx-ultimate-t5-3c-86-6db700\xc0\x0c']`

`2025-03-09 21:29:02.745 DEBUG    Ignoring duplicate message with no unicast questions received from ('127.0.0.1', 5353) [socket 6 (('0.0.0.0', 5353))] (122 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x02\x00\x00\x00\x00\x0b_esphomelib\x04_tcp\x05local\x00\x00\x0c\x00\x01\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x90\x00\x1c\x19localbytes-plug-pm-8ca2b3\xc0\x0c\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x90\x00\x1e\x1btx-ultimate-t5-3c-86-6db700\xc0\x0c']`

`2025-03-09 21:29:07.658 DEBUG    Ignoring duplicate message with no unicast questions received from ('192.168.1.202', 5353) [socket 6 (('0.0.0.0', 5353))] (51 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x0fblackprostation\x06_dosvc\x04_tcp\x05local\x00\x00\xff\x00\x01']`

`2025-03-09 21:29:07.922 DEBUG    Ignoring duplicate message with no unicast questions received from ('192.168.1.202', 5353) [socket 6 (('0.0.0.0', 5353))] (51 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x0fblackprostation\x06_dosvc\x04_tcp\x05local\x00\x00\xff\x00\x01']`

`2025-03-09 21:29:11.750 DEBUG    Ignoring duplicate message with no unicast questions received from ('127.0.0.1', 5353) [socket 6 (('0.0.0.0', 5353))] (122 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x02\x00\x00\x00\x00\x0b_esphomelib\x04_tcp\x05local\x00\x00\x0c\x00\x01\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x87\x00\x1c\x19localbytes-plug-pm-8ca2b3\xc0\x0c\xc0\x0c\x00\x0c\x00\x01\x00\x00\x11\x8b\x00\x1e\x1btx-ultimate-t5-3c-86-6db700\xc0\x0c']`

`2025-03-09 21:29:50.441 DEBUG    Ignoring duplicate message with no unicast questions received from ('192.168.1.10', 5353) [socket 6 (('0.0.0.0', 5353))] (387 bytes) as [b'\x00\x00\x84\x00\x00\x00\x00\x01\x00\x00\x00\x03\x0b_googlecast\x04_tcp\x05local\x00\x00\x0c\x00\x01\x00\x00\x00x\x00.+Chromecast-916a532677485cdd4e7772f2bf65153f\xc0\x0c\xc0.\x00\x10\x80\x01\x00\x00\x11\x94\x00\xd2#id=916a532677485cdd4e7772f2bf65153f#cd=417F168FB62EB01C64C15EC395D85943\x03rm=\x05ve=05\rmd=Chromecast\x12ic=/setup/icon.png\x14fn=Master Bedroom TV\tca=465413\x04st=1\x0fbs=FA8FCA51C03A\x04nf=1\tct=CC7757\x13rr=AndroidNativeApp\x07rs=ITVX\xc0.\x00!\x80\x01\x00\x00\x00x\x00-\x00\x00\x00\x00\x1fI$916a5326-7748-5cdd-4e77-72f2bf65153f\xc0\x1d\xc1L\x00\x01\x80\x01\x00\x00\x00x\x00\x04\xc0\xa8\x01\n']`

bdraco, 2025-03-10:
> So I let this run for just over a minute and it generated 185 lines of output, not really sure how to interpret this. Most lines look like google/chomecast chats but a 6 stand out...

185 in a minute isn't great but probably not enough to overload the device. If you get it into the failure mode again, please run it, capture the full log and send it to my gdrive bdraco@gmail.com or dropbox nick@koston.org

[44 more replies]