esphome/issues #6314: Unable to complie with fonts

kyvaith, 2024-10-08:
### The problem

After many problems that I was able to solve by my own, now I'm in doubt. I'm unable to build the project in VSCode on Windows.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-devkit-c

### Component causing the issue

font

### Example YAML snippet

```yaml
..
font:
  - file: "https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/fonts/consola.ttf"
    id: my_font16
    size: 16
    bpp: 4
    glyphs:
      '${font_glyphs}'
..
```

### Anything in the logs that might be useful for us?

```txt
PS C:\Users\TomaszWitke\atmosfera-echo-hub> esphome compile .\atmosfera-echo-hub.yaml
INFO ESPHome 2024.9.2
INFO Reading configuration .\atmosfera-echo-hub.yaml...
INFO Updating https://github.com/gnumpi/esphome_audio@dev-next
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Scripts\esphome.exe\__main__.py", line 7, in <module>
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\__main__.py", line 1018, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\__main__.py", line 996, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\config.py", line 1091, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\config.py", line 945, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\config.py", line 935, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\config.py", line 859, in validate_config
[... 69 lines cut ...]
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\voluptuous\schema_builder.py", line 281, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\voluptuous\validators.py", line 251, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\voluptuous\validators.py", line 379, in _exec
    v = func(path, v)
        ^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\voluptuous\schema_builder.py", line 855, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\components\font\__init__.py", line 213, in download_web_font
    external_files.download_content(url, path)
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\site-packages\esphome\external_files.py", line 106, in download_content
    path.write_bytes(data)
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\pathlib.py", line 1036, in write_bytes
    with self.open(mode='wb') as f:
         ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\TomaszWitke\AppData\Local\Programs\Python\Python312\Lib\pathlib.py", line 1013, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: '.esphome\\font\\c9ee33c7\\font.ttf'
PS C:\Users\TomaszWitke\atmosfera-echo-hub>
```

durandm70, 2024-11-15:
Exact same error for me. All the sudden, my project, which compiled so far, does not anymore, with the "Invalid argument" related to font.ttf... (exact same error : "OSError: [Errno 22] Invalid argument: '.esphome\\font\\c9ee33c7\\font.ttf'")

clydebarrow, 2024-11-16:
> ### Which version of ESPHome has the issue?
> 2021.9.2

Might be time for an update I think.

durandm70, 2024-11-16:
> > ### Which version of ESPHome has the issue?
> > 2021.9.2
> 
> Might be time for an update I think.

Mine is 2024.10.3, so update already done... The problem has yet appeared :(

valor11, 2024-11-16:
Have a look here, it may help.  

https://community.home-assistant.io/t/compile-error-esphome-on-windows/764427/3

clydebarrow, 2024-11-16:
> Mine is 2024.10.3, so update already done... The problem has yet appeared :(

Try with the current beta version. There were changes to the font component since 2024.10.3. As far as I can tell there were no earlier changes since June.

durandm70, 2024-11-16:
Solved !
I first applied the upgrade suggested by cludebarrow ; same issue with version 2024.11.0b2...
Then, I have applied the solution suggested by jamescuber : it has solved the issue.

Practically speaking, I have downloaded the 2 fonts and replaced the https references by local references :
From:
```
...
- file: "https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/fonts/consola.ttf"
  id: my_font82
  size: 82
  glyphs:
    '${font_glyphs}'     

- file: "https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/raw/main/fonts/materialdesignicons-webfont.ttf"
...
```

to:
```
...
- file: "./fonts/consola.ttf"
  id: my_font82
  size: 82
  glyphs:
    '${font_glyphs}'     

- file: "./fonts/materialdesignicons-webfont.ttf"
...
```

Thanks to all for your help :-)

clydebarrow, 2024-11-17:
I tested this on Windows 11 from the command line and it worked just fine.