esphome/issues #6316: cse7766 platform says it requires even parity to compile and throws an error about parity on startup in the log

jweston2112, 2024-10-09:
### The problem

INFO ESPHome 2024.10.0b1
INFO Reading configuration /config/esphome/sonoff-1.yaml...
INFO Detected timezone 'America/New_York'
Failed config

uart: [source /config/esphome/sonoff-1.yaml:28]
  - rx_pin: 
      number: 3
      mode: 
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    baud_rate: 4800
    
    Component cse7766 requires parity EVEN for the uart referenced by uart_id.
    parity: NONE
    rx_buffer_size: 256
    stop_bits: 1

and in the log this shows up

[04:21:07][E][uart:028]:   Invalid parity: Integration requested parity NONE but you have EVEN!

### Which version of ESPHome has the issue?

2024.10.0b1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10

### What platform are you using?

ESP8266

### Board

Sonoff S31 

### Component causing the issue

cse7766

### Example YAML snippet

```yaml
uart:
  rx_pin: RX
  baud_rate: 4800
  parity: EVEN

sensor:
  - platform: cse7766
    current:
      name: "Current"
      accuracy_decimals: 1
      filters:
        - throttle_average: 60s
    voltage:
      name: "Voltage"
      accuracy_decimals: 1
      filters:
        - throttle_average: 60s
    power:
      name: "Power"
      accuracy_decimals: 1
      id: my_power
      filters:
        - throttle_average: 60s
```

### Anything in the logs that might be useful for us?

```txt
[04:21:07][E][uart:028]:   Invalid parity: Integration requested parity NONE but you have EVEN!
```

### Additional information

Sadly i Have no way to properly verify the values being reorted

ssieb, 2024-10-09:
There have been no changes in the data handling.

jweston2112, 2024-10-09:
probably has something to do with the sensor filter... best as I can tell - Thanks for looking into this....

closed 2024-10-09 by kbx81