esphome/issues #6319: Assist Satellite announce not available

hugobloem, 2024-10-10:
### The problem

Since 2024.10 Home Assistant has the `assist_satellite.announce` action, which should allow announcing messages on ESPHome devices given PR [#7377](https://github.com/esphome/esphome/pull/7377). 

However, when running the action in the HA developer tools none of the assist satellites show up. I am using the ESPHome provided [firmware for the esp32s3-box-3](https://github.com/esphome/wake-word-voice-assistants/tree/main/esp32-s3-box-3). Does anything need changed in the firmware to enable this?

### Which version of ESPHome has the issue?

2024.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10

### What platform are you using?

ESP32

### Board

esp32s3box3

### Component causing the issue

voice_assistant

bbaumg02, 2024-11-03:
i am having the same issue.

smirgol, 2024-11-22:
Same issue here. I'd like to set up an automation and have my S3 confirm with a message, but I cannot select the S3 as a target for the `assist_satellite: Announce`.

andreas-bulling, 2025-01-01:
Same problem here. Is anyone from ESPHome going to take care of this - rather serious I'd say - bug?

Cha0sMnky, 2025-01-01:
Same issue, but with m5stack atom echo hardware.

RichardA1, 2025-01-31:
I am having the same issue and I wanted to check to see what, if any, device will this work on? I can not find any information on how this is used.

hugobloem, 2025-01-31:
I think announce is only available for `voice_assist` configurations that use a `media_player` component instead of the `speaker` component. I don't understand why there is this limitation but it may be resolved soon with the move away from esp-adf in 2025.2.

RichardA1, 2025-03-26:
I have tried making a voice assistant with the speaker and media_player component but nothing pops up for the Assist Satellite announce.

Could someone please post some basic example code on how to do this?
The documentation is very has now explanation of how to create this entity... unless I am missing something, if so, please let us know.

AapoTahkola, 2025-03-30:
As far as I have understood there is currently no way to get speaker working with micro wake word unless you use android framework. https://gist.github.com/EverythingSmartHome/055fbdde31a607ef9d695d5cac780e94?permalink_comment_id=5380865#gistcomment-5380865 speaks of nabu speaker but as I looked at it the git repo removed it since then. If I were to try to fix it myself I would probably take a look at it with https://aider.chat/

ssieb, 2026-04-19:
I don't know about asking questions, but the announcing should certainly be working.  Check out the esphome ready-made projects for voice assistant.  If you don't have one of the example devices to try directly, you can look at the yaml for them.