esphome/issues #6321: Install fails on Python 3.13 (pillow needs to be updated)

ptr727, 2024-10-11:
### The problem

Installing using python 3.13 fails.

`pip wheel --no-cache-dir --progress-bar off --wheel-dir /wheels setuptools esphome[displays]`

```
Collecting pillow==10.2.0 (from esphome[displays])
  Downloading pillow-10.2.0.tar.gz (46.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      Traceback (most recent call last):
        File "/usr/local/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/usr/local/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/local/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-oxb4d18z/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-oxb4d18z/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/tmp/pip-build-env-oxb4d18z/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 318, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 31, in <module>
        File "<string>", line 28, in get_version
      KeyError: '__version__'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

Suspect issues in [pillow that needs to be updated to support Python 3.13](https://github.com/python-pillow/Pillow/issues/8075).

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

ptr727, 2024-10-11:
I see there is a [PR](https://github.com/esphome/esphome/pull/7566) to update `pillow` for vulnerability fixes, may also fix this issue.

stedaho, 2024-11-16:
I tested the patch from the PR with my current ESPHome `2024.10.3` installation and can confirm that the project compiles now! Before I had the same issues using Python `3.13.0`. Wonder why ESPHome pins a specific pillow version instead of allowing e.g. `10.x`.

unimatrixZxero, 2024-11-18:
Encountered the problem and updated to `2024.11.0b2` and now it works when the `requirements.txt` lists pillow with `10.4.0`