esphome/issues #6330: Sensirion SEN55 NoX reporting issues

chris-in-texas, 2024-10-15:
### The problem

For a while now the NoX values have increased to 100 from a value of 1 as a baseline, however those values do dip down towards mid-upper 60's on occasion.  After a power on reboot, this started happening in that the value went from 1 to 100.  If I take the SEN55 directly and use the Sensirion software it is reporting correctly with a value of 1, and if there is an engine running nearby it will increase as expected into the upper single digits.  If its plugged into the 8266 module the raw data feed is showing 100 now.  I have 8 of these sensors and all of them now are reporting 100 on the NOX levels.  Changing the index_offset has no effect on the unit.  There are other users experiencing the same issue.  It started with a single unit that went from 1 to 100 over a power on reboot, and now all of the sensors are doing the same thing.  It took about 6 months of various reboots to get them all now reporting 100 as the base value and will go down from there on occasion.  

### Which version of ESPHome has the issue?

2024

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

ryanwasalreadytaken, 2024-11-14:
If you need any help troubleshooting I have two units on two esp32 devices that have the same issue. Both were functioning and problem started with an update around the same timeline as chris-in-texas. 

I don't believe it should make a difference, but I'm using the sen5x adafruit breakout adapter on both.

chris-in-texas, 2024-11-30:
So it appears that there is an issue in the default values now that has started to conflict with things.  As it doesn't look like anyone was going to look at this issue I have put in a number of hours of troubleshooting and came up with a solution that does work for me.  I am currently running version 2024.11.2 and have updated all nodes with the same version.

How to fix;
1) in the yaml config for the node, simply adjust the "index_offset" from 100 to 1.  
2) Then save the config and install to the node.  
3) After that cycle power on the unit.  <-- this is important to do, without it the value will still be wrong.  I assume after 2x the learning time in hours it might eventually correct itself but simple power cycle will do it now.  This will remove any stored learning.  

Once the sensor starts reporting it should start to report the level of 1. (as long as there aren't any valid NOX levels).  I have verified this.  Some of the Sensirion documentation states that this value should default to 100 and other documentation states it should be default of 1.  The 100 value use to work just fine, but some change a while back and affected this.  All six of my units now are reporting a value of "1" and will increase when exposed to NOX and then return to "1" when aired out.

Here is the NOX section of the YAML config that does work at least for me with my SEN55's:

 nox:
      name: "Node 1 NOX"
      algorithm_tuning:
        index_offset: 1
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230

closed 2024-11-30 by chris-in-texas

Archer36, 2024-12-05:
Just wanted to say thanks for this, I wondered why it starting acting this way as well. It almost appears that someone thought the NOx and the VOC defaults were identical, but according to the SEN55 datasheet (https://sensirion.com/resource/datasheet/sen5x) index_offset and ating_max_duration_minutes are indeed different from their VOC counterparts. Should a PR be opened to update the esphome defaults to match the datasheet?

EinRainerZufall, 2024-12-24:
I've just fallen into this trap too. And I also think that this PR needs to be reopened so that the defaults in ESPHome match the defaults in the data sheet.