esphome/issues #6331: DS18B20 sensor not working after update from 2023.5.5 to 2024.9.2

vincent-tr, 2024-10-16:
### The problem

# Environment

I have 3 identical boxes, based on ESP32-POE from Olimex, with:
- a remote receiver
- 4 switchs
- a DS18B20 temperature sensor

I updated ESPHome from 2023.5.5 to 2024.9.2, and after the update:
- 2 boxes have the same issue with DS18B20 sensor, the 3rd one is working properly
- logging configuration level is set to INFO. If I switch it to DEBUG, the issue disappears

Here is the configuration part:
```
    one_wire:
      - platform: gpio
        pin: GPIO13

    sensor:
      - platform: dallas_temp
        id: temp
        name: bedplug-ch1-temp
        address: 0x3357ac4909646128
        update_interval: 10s
```

# The issue:

During setup, I get in the logs:
```
[08:13:54][W][gpio.one_wire:078]:   Found no devices!
```
Then, on each sensor update:
```
[08:13:58][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[08:13:59][W][dallas.temp.sensor:139]: 'bedplug-ch1-temp' - Scratch pad checksum invalid!
[08:13:59][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

None

### What platform are you using?

ESP32

### Board

ESP32-POE

### Component causing the issue

gpio.one_wire

m1ch, 2024-10-16:
I think this relates to https://github.com/esphome/issues/issues/6260 and the issues should be merged.

vincent-tr, 2025-01-28:
Fixed, cf here: https://github.com/esphome/issues/issues/6260#issuecomment-2618553096

closed 2025-01-28 by vincent-tr