esphome/issues #6332: File ownership and permissions issues with ESP-home web UI when using external editor (VS-Code Server)

m1ch, 2024-10-16:
### The problem

**Description:**  
When editing packages in the ESP-home web UI or working with Git, I encounter a problem that prevents me from accessing the edited files in VS-Code Server. The issue arises because the ESP-home web UI always resets the file ownership to root and permissions to 644 when modifying a file. 

This prevents me from continuing to edit the files in VS-Code Server, as the ownership and permissions changes cause my UID/GID (1000) to be lost. I've noticed that other Docker images allow setting the UID and GID of the running process, which would resolve this issue. 

**Expected behavior:**  
The ESP-home web UI should preserve the original file ownership and permissions when editing or working with Git, allowing seamless integration with external editors like VS-Code Server. 

**Steps to reproduce:**  

    Open the ESP-home web UI.
    Edit a package or work with Git in the editor.
    Save changes to a file.
    Observe that the file ownership is reset to root and permissions are set to 644.
     

**Reproducible on:**   

    Docker image used: ghcr.io/esphome/esphome:2024.9
    ESP-home version: 2024.9.2
     

I hope this improved bug report helps you get your issue addressed. Let me know if I can assist with anything else! 

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP32

### Board

NA

ptr727, 2024-10-16:
The ESPHome docker only runs as root, and will set file ownership/permissions to root.  
You can run your codeserver container as root, or add your vote for non-root support, or try a [version](https://github.com/ptr727/ESPHome-NonRoot) that does support non-root.

m1ch, 2024-10-18:
I will definitely add my vote for none root support, since there are already some working implementations out there. 

I also don't get, why the files can't be saved with the same permissions and owner as before. It's not hard to do in python.

triplepoint, 2025-02-24:
I believe this issue is still active.