esphome/issues #6335: 2024.10.0 breaks i2c on Emporia Vue2 devices

KodinLanewave, 2024-10-16:
### The problem

Upgrading my Emporia Vue2 to 2024.10.0 broke my device; it would constantly throw i2c timeouts:

```
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
^C[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][D][api.connection:199]: ESPHome Logs 2024.10.0 (192.168.0.254) requested disconnected
``` 

**I've tried flashing via home assistant integration as well as manually uploading via docker image** remote OTA, and 2024.10.0 has consistent results; Downgrading back to 2024.9.2 resolved the i2C issues.

Anything change with your i2c implementation in the latest release?

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Emporia Vue2

### Component causing the issue

i2c

### Anything in the logs that might be useful for us?

```txt
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
^C[17:50:30][E][emporia_vue:048]: Failed to read from sensor due to I2C error 3
[17:50:30][D][api.connection:199]: ESPHome Logs 2024.10.0 (192.168.0.254) requested disconnected
```

feisley, 2024-10-16:
Just to add here this 2024.10 seems to impact more than just Emporia Vue: https://community.home-assistant.io/t/display-not-working-any-more-after-update-to-esphome-2024-10-0/783022

g8ry, 2024-10-16:
Yeah, i2c is broken on Emporia Vue2 for me too on 2024.10, so had to downgrade to 2024.9. Same error as first posted.

craigwitter, 2024-10-16:
I had the same i2c error too. Reverting the firmware to the last 2024.9 version has no problem.

Paul0372, 2024-10-17:
also breaks i2c with BME680
[21:31:58][I][i2c.arduino:096]: Results from i2c bus scan:
[21:31:58][I][i2c.arduino:098]: Found no i2c devices!
[21:31:58][C][template.sensor:022]: Template Sensor 'BME680 Indoor Air Quality'
[21:31:58][C][template.sensor:022]:   State Class: 'measurement'
[21:31:58][C][template.sensor:022]:   Unit of Measurement: ''

clydebarrow, 2024-10-17:
try setting the i2c clock speed to something lower.

agoode, 2024-10-17:
I was finally able to resolve this by *increasing* the clock speed and also setting `timeout`:

```yaml
  frequency: 400kHz  # recommended range is 50-200kHz
  timeout: 1ms
```

Either one alone would not completely solve the problem.

genehand, 2024-10-17:
Nice, 400kHz plus the 1ms timeout fix the emporia-vue-local component for me as well.

KodinLanewave, 2024-10-17:
That said - _why_ has this changed in 2024.10.0 to require a different frequency?

thenoid, 2024-10-17:
Ditto, busted here too

wingrunr21, 2024-10-17:
> That said - _why_ has this changed in 2024.10.0 to require a different frequency?

This looks to be the only related I2C change: espressif/esp-idf#4999

Looks like the ESP-IDF 4.4.8 included this back port (which was included in ESPHome 2024.10). I didn't fully read through the code to validate that's what it was though. 

@agoode 
Did you try it keeping the frequency at 200 kHz but adding the timeout?

agoode, 2024-10-17:
I did try 200 kHz with even the max timeout (13 ms) but it did not work.

garylankford8, 2024-10-18:
Encountered the same issue with 10.0 and came here for a solution... 400kz and 1ms worked for me, too...

Thank you all!

weapons, 2024-10-18:
Hi Guys. I'm having the same issue with my Emporia Vue.  Where should I make the frequency and timeout changes to fix the problem?  TIA

feisley, 2024-10-18:
So just to summarize:

Emporia Vue 2 has a workaround by setting the i2c frequency to 400 kHz and 1 ms timeout.

Emporia Vue 3 works as long as the frequency is 200 kHz and the timeout can be left at the esp-idf default of 100 us (The only other frequencies i have tested are 100 kHz and 150 kHz, both of which do not work, but 150 kHz was working in 2024.09)

dmtalon, 2024-10-18:
I applied these changes after experiencing the problem OP did, and this seems to have corrected mine as well.  Thanks a ton. I was about to restore to the previous version.

veenstrahiem, 2024-10-18:
Same here, I can confirm this worked for me too: setting the i2c frequency to 400 kHz and 1 ms timeout

aminorjourney, 2024-10-19:
Works after a fashion. I'm not getting all my sensors populating properly from one of my Vue 2s even with the suggested values - rolling back to see if they re-appear.

kbrint, 2024-10-19:
Downgrade to 2024.9.2 worked for me.  New values (frequency and timeout) also worked, both on 2024.9.2 and now back on 2024.10.0.

burnski666, 2024-10-19:
Same problem, 400khz / 1ms also resolved!

echooffzack, 2024-10-19:
> I was finally able to resolve this by _increasing_ the clock speed and also setting `timeout`:
> 
> ```yaml
>   frequency: 400kHz  # recommended range is 50-200kHz
>   timeout: 1ms
> ```
> 
> Either one alone would not completely solve the problem.

Came here to say this worked for me as well.

Thank you!

tzapu, 2024-10-19:
both downgrade and upgrade + freq + timeout worked for me as well on a vue 2

cecheto, 2024-10-20:
Thanks, timeout and frequency change worked for me.

charettepa, 2024-10-21:
with recommended ranges at 50-200
is there any possibility of issues or damage to the board by running at 400?

clydebarrow, 2024-10-21:
> s there any possibility of issues or damage to the board by running at 400?

No, the worst that will happen is it won't work.

twister36, 2024-10-22:
Is this a band-aid fix or will changing the frequency and timeout settings be supported going forward? I haven't upgraded yet... I just saw this posted and wanted to validate before upgrading

agoode, 2024-10-22:
I am guessing this is the way going forward, since it seems the issue was caused by a longstanding bug that was fixed recently.

feisley, 2024-10-22:
I also just tested the Vue 3 on a frequency of 400 kHz and timeout of 1ms and that is working.

Additionally this has resolved the occasional errors I would see of malformed readings on my older setting of 200 kHz / default timeout.

ziptbm, 2024-10-22:
Has this issue been resolved for 10.1?  Or does it still require some manual manipulation as described above?

I rolled back to the 9.x version in the meantime.  Is upgrading advised at this time?  And if so, could someone point me to the specific steps to change?  Thanks.

agoode, 2024-10-23:
I think this issue can probably be closed here.

closed 2024-10-24 by clydebarrow

FredericMa, 2024-10-24:
> I am guessing this is the way going forward, since it seems the issue was caused by a longstanding bug that was fixed recently.

@agoode, do you have some additional info about the fixed bug that caused this problem? Just curious.

wingrunr21, 2024-10-24:
> 
> This looks to be the only related I2C change: [espressif/esp-idf#4999](https://github.com/espressif/esp-idf/issues/4999)
>

@FredericMa the one I linked earlier

RickSisco, 2024-10-27:
Also confirmed to fix vue2 on my end..

```
i2c:
  sda: 21
  scl: 22
  scan: false
  frequency: 400kHz  # recommended range is 50-200kHz
  timeout: 1ms
  id: i2c_a
```

murrayma, 2024-10-30:
> I also just tested the Vue 3 on a frequency of 400 kHz and timeout of 1ms and that is working.
> 
> Additionally this has resolved the occasional errors I would see of malformed readings on my older setting of 200 kHz / default timeout.

400kHz / 1ms fixed errors on my vue2 and also stopped occasional errors from my vue3.

seaverd, 2024-11-01:
I have this issue and updated per the fix recommended above.  However, now my log is filled with the following:

[09:12:16][W][component:237]: Component emporia_vue.sensor took a long time for an operation (57 ms).
[09:12:16][W][component:238]: Components should block for at most 30 ms.

Any suggestions?

VivantSenior, 2024-11-01:
> I have this issue and updated per the fix recommended above. However, now my log is filled with the following:
>
> [09:12:16][W][component:237]: Component emporia_vue.sensor took a long time for an [...]

[#4717](https://github.com/esphome/issues/issues/4717).

aviateur17, 2024-11-08:
I, too, had this issue right after upgrading the software.   I ordered 2 more units thinking I had a power issue on the board.  Thankfully this resolved the issue.

```
i2c:
  sda: 21
  scl: 22
  scan: false
  # frequency: 200kHz  # recommended range is 50-200kHz
  # Use below settings due to issue after upgrade.
  frequency: 400kHz  # recommended range is 50-200kHz
  timeout: 1ms
  id: i2c_a
```

mandreko, 2024-11-20:
Bummers. I had this affect me as well. However, after updating the frequency to 400khz and the timeout to 1ms, and pushing out a rebuild, I still get the error on both of my Vue2 boxes.