esphome/issues #6336: Custom boad json not found

xxLeoxx93, 2024-10-16:
### The problem

Running ESPHome as a stand alone docker it's fairly easy to put custom board jsons into "/config/esphome/.esphome/build/[your-esphome-device]/boards".
Now as a home assistant addon the folder struckture is completly different.
Also no documentation on how to include custom boards is available :(

### Which version of ESPHome has the issue?

2024.10

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10

### What platform are you using?

ESP32

### Board

lilygo-t5-47-plus

### Component causing the issue

board config

### Example YAML snippet

```yaml
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    board_dir: boards
    board: lilygo-t5-47-plus

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/esphome-web-77b900.yaml...
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING This board is unknown. Make sure the chosen chip component is correct.
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-77b900 (board: lilygo-t5-47-plus; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'lilygo-t5-47-plus'
```

ssieb, 2024-10-16:
You very rarely need a specific board config.  Try the generic one.

xxLeoxx93, 2024-10-17:
@ssieb Thanks fo the hint. The last time I used it it was needed but I tried a default board:

`esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: esp32-s3-devkitm-1 

esp32:
  variant: esp32s3
  board: esp32-s3-devkitm-1 `

to the same effect:

`INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/esphome-web-77b900.yaml...
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING This board is unknown. Make sure the chosen chip component is correct.
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-77b900 (board: esp32-s3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
UnknownBoard: Unknown board ID 'esp32-s3-devkitm-1'`

ssieb, 2024-10-17:
For S3, the generic board is `esp32-s3-devkitc-1` and you don't need to specify the variant.
Also remove the platformio options.

xxLeoxx93, 2024-10-17:
That worked, but the display stopped working after installing.
I managed to get the custom board file to work by installing the portainer addon, going into the esphome terminal, navigating to /data/build/<espname>/boards and placing the json there.

Seems very unpractical to me to have it this way but at least it works now...

ssieb, 2024-10-17:
Why would the display stop working?  You should come ask for help on the esphome discord server.

alexphredorg, 2024-11-20:
The instructions here solved my problem:
https://github.com/Januznl/esphome-lilygo-t5-s3-plus

It's the same basic approach, but using a different build_dir.  Note that the board_dir shown doesn't do anything, and using the proper boards_dir now creates a compile error.