esphome/issues #6338: Error uploading to devices in 2024.10.0

Sc00tz, 2024-10-17:
### The problem

When trying to update some devices, the output looks like this:

```
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/wyzeplug1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing wyzeplug1 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
RAM:   [====      ]  43.2% (used 35400 bytes from 81920 bytes)
Flash: [=====     ]  53.2% (used 544353 bytes from 1023984 bytes)
========================= [SUCCESS] Took 3.11 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.59
INFO Uploading /data/build/wyzeplug1/.pioenvs/wyzeplug1/firmware.bin (548512 bytes)
INFO Compressed to 379859 bytes
Uploading: [=                                                           ] 2% ERROR Error receiving acknowledge chunk OK: timed out
```

### Which version of ESPHome has the issue?

2024-10-0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.2

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
esphome:
  name: bedroom-minisplit

esp8266:
  board: esp01_1m

# Enable Home Assistant API
api:
  encryption:
    key: "-REDACTED-"

ota:
  - platform: esphome
    password: "-REDACTED-"
  
web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom-Minisplit"
    password: "-REDACTED-"

captive_portal:
    
substitutions:
  name: bedroom-minisplit
  friendly_name: Bedroom Minisplit

# Enable logging
logger:
  level: ERROR
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART0
```

Pat0856, 2024-10-17:
I have also a compiling error while uploading my ESP32 S3 Box3 with  ESPHome version 2024.10.0

Sc00tz, 2024-10-18:
> I have also a compiling error while uploading my ESP32 S3 Box3 with ESPHome version 2024.10.0

Just to clarify, it seems to compile fine, it's just not uploading.

Pat0856, 2024-10-19:
With HomeAssistant core 2024.10.3 the uploadind is working again

uspass, 2024-11-02:
With HomeAssistant core 2024.10.4 and ESPHome version 2024.10.2 uploading ESP32 S3 doesn't work. It looks like it uploads but flashing actually doesn't happen. The ESPHome version stays at 2024.10.0. More, if serial flashing on a new module using manual download  (ESP32 WROOM 06), the module doesn't do anything, like getting online.

uspass, 2024-11-02:
Another thing, all devices based on 8266 and those based on BK7231N are ok.