esphome/issues #6339: PN7160 post-read IRQ times out with 50kHz I2C bus

ducky64, 2024-10-17:
### The problem

When the I2C bus frequency is left at the default (50kHz), PN7160 fails on startup:
```
[W][pn7160:1162]: Timed out waiting for IRQ state
[W][pn7160_i2c:030]: read_nfcc_() post-read timeout waiting for IRQ line to clear
[W][pn7160:1111]: Error receiving message
[E][pn7160:1130]: Error in response to command: 60 00 09 02 01 20 04 04 61 12 50 05
[E][pn7160:244]: Error sending reset command
[E][pn7160:606]: Failed to reset NCI core
```

This repeats a few times before:
```
[E][pn7160:718]: Too many initialization failures -- check device connections
[E][component:119]: Component pn7160 was marked as failed.
[E][component:164]: Component pn7160 set Error flag: unspecified
```

Upping the frequency to 100kHz and beyond solves the problem. Perhaps there should be a note in the documentation about this?

Extending the wait timeout does not make a difference (other than making each cycle take longer). Perhaps it's the next read IRQ being detected?

Ignoring the results of the post-read check also makes things work. Perhaps it makes sense for the error to be non-fatal? If the expectation is the IRQ would have cleared in the specified timespan, maybe just log a warning?

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

custom ESP32-S3 + PN7160

### Component causing the issue

pn7160

### Example YAML snippet

```yaml
This breaks:
i2c:
  scl: GPIO17
  sda: GPIO18

pn7160_i2c:
  irq_pin: GPIO47
  ven_pin: GPIO39

This works:
i2c:
  scl: GPIO17
  sda: GPIO18
  frequency: 400kHz  # must be at least 100kHz or a fatal timeout happens

pn7160_i2c:
  irq_pin: GPIO47
  ven_pin: GPIO39
```

JohnMcLear, 2026-02-06:
Can you still replicate this issue?  I just tested it on 

```
[09:46:28.250][I][app:206]: ESPHome version 2026.1.3 compiled on 2026-02-06 09:45:43 +0000
[09:46:28.256][I][app:213]: ESP32 Chip: ESP32 r3.0, 2 core(s)
[09:46:28.256][W][app:222]: Set minimum_chip_revision: "3.0" to reduce binary size
```

no frequency item set - no errors and the pn7161 works.  Elechouse pn7161 fwiw on devkit board

JohnMcLear, 2026-02-25:
Feel free to test this repo https://github.com/JohnMcLear/esphome_pn7160

BlackEternity, 2026-09-11:
Just to chime in on this. I know it's old but someone like me might stumble upon it.
I encountered the same issue.
My Board is a MIKROE-6453 - a "Click System" board from MikroE that uses the PN7160 over I²C. 

My ESPHome error logs showed the same
`[E][pn7160:718]: Too many initialization failures -- check device connections
[E][component:119]: Component pn7160 was marked as failed.
[E][component:164]: Component pn7160 set Error flag: unspecified`

When I dug into the driver and troubleshooting to see the RAW NCI traffic values, I could read "61 23 00" as the constantly repeated error.
According  to [NXP User Manual (Hosted by Mouser)](https://www.mouser.com/pdfDocs/UM11495.pdf) it shows 0x23 as this error:

> Notification used to indicate that TxLdo (RF Transmitter)
> could not start. This is usually due to a missing or bad power
> supply on VUP/TVDD or a bad clock/power configuration

I removed the bridge from 3.3V to 5V and since then it works flawlessly.

TL;DR:
If you see 0.23 or 61 23 00 in your raw values, check the power supply. Chip Manufacturer recommends increasing Power stability / switching to 5V Jumpers when this occurs.