esphome/issues #6346: WeiKai WK2168 UARTs are not working with I2C

timdaman, 2024-10-18:
### The problem

I purchased a WeiKai [WK2168 breakout from AlieExpress](https://www.aliexpress.com/item/1005007605743585.html?spm=a2g0o.order_list.order_list_main.11.51921802UtddRR) and it has not been working as expected.

I am able to use  UARTs supplied by my device using a SPI connection however when I switch to using I2C to connect it to my esp32 I see no traffic over any UART. I have switched between SPI and I2C without touching any UART wires so I can confine the issue to using a I2C connection.

When connected via I2C I can see the device detected in the I2C bus scan. 

I have contacted DrCoolZic over email but I am posting here as well to make it findable. It have lost many hours to this and hopefully I will save others from struggling with any similar issue.

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.9.3

### What platform are you using?

ESP32

### Board

Basic esp32 dev board

### Component causing the issue

WeiKai

### Example YAML snippet

```yaml
substitutions:
  devicename: basement-bathroom-mmwave

esp32:
  board: esp32dev

esphome:
  name: ${devicename}

time:
  - platform: homeassistant
    id: my_time
    update_interval: 15min

logger:

wifi:
  ssid: !secret "wifi_ssid"
  password: !secret "wifi_password"

ota:
  - platform: esphome
    id: esphome_ota
    password: !secret "ota_pass"

safe_mode:

api:
  encryption:
    key: !secret "api_encryption_key"

button:
  - platform: restart
    name: Restart ${devicename}  

web_server:
  port: 80
logger:
  level: debug

i2c:
  - sda: GPIO33
    scl: GPIO25
    scan: true
    frequency: 100kHz
    id: wk2168_1

wk2168_i2c:
  - address: 0x70
    id: wk2168_bridge_i2c
    # test_mode: 1
    uart:
      - id: i2c_uart_0
        channel: 0
        baud_rate: 115200

dfrobot_sen0395:
  - id: mmWave_1
    uart_id: i2c_uart_0

switch:
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    type: sensor_active
    name: mmWave Active
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    type: presence_via_uart
    name: mmWave over UART
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    type: presence_via_uart
    name: mmWave over UART
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    type: turn_on_led
    name: mmWave LED
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    type: start_after_boot
    name: mmWave start on boot  

binary_sensor:
  # Information coming from uart bus
  - platform: dfrobot_sen0395
    dfrobot_sen0395_id: mmWave_1
    name: Mmwave Detected via UART

button:
  - platform: template
    name: "MMave Load Config"
    on_press:
        - dfrobot_sen0395.settings:
            factory_reset: true
            detection_segments:
              - [0cm, 3m]
            output_latency:
              delay_after_detect: 0s
              delay_after_disappear: 5s
            sensitivity: 7
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/basement-bathroom-mmwave.yaml...
INFO Detected timezone 'America/Toronto'
INFO Starting log output from 192.168.9.170 using esphome API
INFO Successfully connected to basement-bathroom-mmwave @ 192.168.9.170 in 0.045s
INFO Successful handshake with basement-bathroom-mmwave @ 192.168.9.170 in 0.096s
[22:17:55][I][app:100]: ESPHome version 2024.9.2 compiled on Oct  2 2024, 21:22:26
[22:17:55][C][wifi:600]: WiFi:
[22:17:55][C][wifi:428]:   Local MAC: 24:0A:C4:F9:E3:D0
[22:17:55][C][wifi:433]:   SSID: [redacted]
[22:17:55][C][wifi:436]:   IP Address: 192.168.9.170
[22:17:55][C][wifi:440]:   BSSID: [redacted]
[22:17:55][C][wifi:441]:   Hostname: 'basement-bathroom-mmwave'
[22:17:55][C][wifi:443]:   Signal strength: -82 dB ▂▄▆█
[22:17:55][C][wifi:447]:   Channel: 1
[22:17:55][C][wifi:448]:   Subnet: 255.255.255.0
[22:17:55][C][wifi:449]:   Gateway: 192.168.9.1
[22:17:55][C][wifi:450]:   DNS1: 149.112.121.20
[22:17:55][C][wifi:451]:   DNS2: 149.112.122.20
[22:17:55][C][logger:185]: Logger:
[22:17:55][C][logger:186]:   Level: DEBUG
[22:17:55][C][logger:188]:   Log Baud Rate: 115200
[22:17:55][C][logger:189]:   Hardware UART: UART0
[22:17:55][C][i2c.arduino:071]: I2C Bus:
[22:17:55][C][i2c.arduino:072]:   SDA Pin: GPIO33
[22:17:55][C][i2c.arduino:073]:   SCL Pin: GPIO25
[22:17:55][C][i2c.arduino:074]:   Frequency: 100000 Hz
[22:17:55][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[22:17:55][I][i2c.arduino:096]: Results from i2c bus scan:
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x70
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x71
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x72
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x73
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x74
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x75
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x76
[22:17:55][I][i2c.arduino:102]: Found i2c device at address 0x77
[22:17:55][C][weikai_i2c:163]: Initialization of wk2168_bridge_i2c with 2 UARTs completed
[22:17:55][C][weikai_i2c:164]:   Crystal: 14745600
[22:17:55][C][weikai_i2c:167]:   Transfer buffer size: 128
[22:17:55][C][weikai_i2c:169]:   Address: 0x70
[22:17:55][C][weikai:322]:   UART i2c_uart_0 ...
[22:17:55][C][weikai:323]:     Baud rate: 115200 Bd
[22:17:55][C][weikai:324]:     Data bits: 8
[22:17:55][C][weikai:325]:     Stop bits: 1
[22:17:55][C][weikai:326]:     Parity: NONE
[22:17:55][C][weikai:322]:   UART i2c_uart_1 ...
[22:17:55][C][weikai:323]:     Baud rate: 115200 Bd
[22:17:55][C][weikai:324]:     Data bits: 8
[22:17:55][C][weikai:325]:     Stop bits: 1
[22:17:55][C][weikai:326]:     Parity: NONE
[22:17:55][C][homeassistant.time:010]: Home Assistant Time:
[22:17:55][C][homeassistant.time:011]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[22:17:55][C][restart.button:017]: Restart Button 'Restart basement-bathroom-mmwave'
[22:17:55][C][restart.button:017]:   Icon: 'mdi:restart'
[22:17:55][C][dfrobot_sen0395:014]: Dfrobot Mmwave Radar:
[22:17:55][C][dfrobot_sen0395:016]:   Registered 'Mmwave Detected via UART'
[22:17:55][C][dfrobot_sen0395:016]:     Device Class: 'motion'
[22:17:55][C][dfrobot_sen0395:068]:   Sensor Active Switch 'mmWave Active'
[22:17:55][C][dfrobot_sen0395:091]:     Restore Mode: always OFF
[22:17:55][C][dfrobot_sen0395:068]:   Turn on LED Switch 'mmWave LED'
[22:17:55][C][dfrobot_sen0395:091]:     Restore Mode: always OFF
[22:17:55][C][dfrobot_sen0395:068]:   Presence via UART Switch 'mmWave over UART'
[22:17:55][C][dfrobot_sen0395:091]:     Restore Mode: always OFF
[22:17:55][C][dfrobot_sen0395:068]:   Start after Boot Switch 'mmWave start on boot'
[22:17:55][C][dfrobot_sen0395:091]:     Restore Mode: always OFF
[22:17:55][C][web_server:145]: Web Server:
[22:17:55][C][web_server:146]:   Address: basement-bathroom-mmwave.local:80
[22:17:55][C][mdns:116]: mDNS:
[22:17:55][C][mdns:117]:   Hostname: basement-bathroom-mmwave
[22:17:55][C][esphome.ota:073]: Over-The-Air updates:
[22:17:55][C][esphome.ota:074]:   Address: basement-bathroom-mmwave.local:3232
[22:17:55][C][esphome.ota:075]:   Version: 2
[22:17:55][C][esphome.ota:078]:   Password configured
[22:17:55][C][safe_mode:018]: Safe Mode:
[22:17:55][C][safe_mode:020]:   Boot considered successful after 60 seconds
[22:17:55][C][safe_mode:021]:   Invoke after 10 boot attempts
[22:17:55][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[22:17:55][C][api:139]: API Server:
[22:17:55][C][api:140]:   Address: basement-bathroom-mmwave.local:6053
[22:17:55][C][api:142]:   Using noise encryption: YES
[22:18:27][D][switch:012]: 'mmWave Active' Turning ON.
[22:18:28][D][dfrobot_sen0395.commands:050]: Command timeout
[22:18:28][D][dfrobot_sen0395.commands:054]: Retrying...
[22:18:30][D][dfrobot_sen0395.commands:050]: Command timeout
[22:18:30][D][dfrobot_sen0395.commands:054]: Retrying...
[22:18:31][D][dfrobot_sen0395.commands:050]: Command timeout
```

ScrotusGobbleBottom, 2024-10-20:
Did you figure this out?  Im in the exact same situation as you.... I have tried everything.
The instructions say to tie the IR pin to VCC,  well,  it doesn't boot and is apparently short to ground.

timdaman, 2024-10-21:
TLDR, SPI worked for me. I used Google translate on the datasheet and got all the details for that.

I have disassembled my experimental setup in preparation for installation but if memory is correct my wiring was very simple
* Connect the SPI ports(`MSO`, `MOSI`, `SCLK`, and `CS`) in the usual manner for SPI (CS is defined on the `wk2168_spi:` component.
* Connect `MD0` and `MD1` to GND. The `MD` pins are connected to pull-ups so you have to ground them to get the needed "00" mentioned in the data sheet.

Nothing else was needed. With those connections the board should be detected and work. You might want to bump up `data_rate:` according to the ESP usage doc, I don't have any comprehensive tests of that yet as I want to test the limit when installed rather in my test setup.

ScrotusGobbleBottom, 2024-10-21:
You didnt do anything with the IR pin?

ScrotusGobbleBottom, 2024-10-22:
It seems when the wk2168 is registered in logs, it opens spi_uart_0 thru 3 but the board has 1-4.  Does spi_uart_o equal Uart1 onthe board?

ScrotusGobbleBottom, 2024-10-22:
It seems when the wk2168 is registered in logs, it opens spi_uart_0 thru 3 but the board has screen printed 1-4. Does spi_uart_0 equal Uart1 on the board? Any changes in yaml because of this?

ScrotusGobbleBottom, 2024-10-22:
All good over here,  It wasnt working because the wk2168 repo has a default crystal clock and it didnt match mine. After changing the clock my SPI_uarts are available and working.