esphome/issues #6350: Component fingerprint_grow is marked FAILED

Pytonballoon810, 2024-10-19:
### The problem

after flashing a "good" ie., working config for a fingerprint sensor i get the `[E][component:082]:   Component fingerprint_grow is marked FAILED` error after connecting to the device:
[image]

The yaml config was this:
```yaml
esphome:
  name: xxx
  friendly_name: xxx

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: VERY_VERBOSE

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxx"
    password: "xxx"

captive_portal:
    

uart:
  rx_pin: D1
  tx_pin: D0
  baud_rate: 57600

fingerprint_grow:
  sensing_pin: D2

  on_finger_scan_invalid:
    - homeassistant.event:
        event: esphome.xxx
  on_finger_scan_matched:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
          confidence: !lambda 'return confidence;'
  on_finger_scan_unmatched:
    - homeassistant.event:
        event: esphome.xxx
  on_finger_scan_misplaced:
    - homeassistant.event:
        event: esphome.xxx
  on_enrollment_scan:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
          scan_num: !lambda 'return scan_num;'
  on_enrollment_done:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
  on_enrollment_failed:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'

api:
  encryption:
    key: "xxx"

  actions:
  - action: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - action: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - action: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - action: delete_all
    then:
      - fingerprint_grow.delete_all:
```

[image]

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

TFT AS608

### Example YAML snippet

```yaml
esphome:
  name: xxx
  friendly_name: xxx

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: VERY_VERBOSE

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxx"
    password: "xxx"

captive_portal:
    

uart:
  rx_pin: D1
  tx_pin: D0
  baud_rate: 57600

fingerprint_grow:
  sensing_pin: D2

  on_finger_scan_invalid:
    - homeassistant.event:
        event: esphome.xxx
  on_finger_scan_matched:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
          confidence: !lambda 'return confidence;'
  on_finger_scan_unmatched:
    - homeassistant.event:
        event: esphome.xxx
  on_finger_scan_misplaced:
    - homeassistant.event:
        event: esphome.xxx
  on_enrollment_scan:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
          scan_num: !lambda 'return scan_num;'
  on_enrollment_done:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'
  on_enrollment_failed:
    - homeassistant.event:
        event: esphome.xxx
        data:
          finger_id: !lambda 'return finger_id;'

api:
  encryption:
    key: "xxx"

  actions:
  - action: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - action: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - action: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - action: delete_all
    then:
      - fingerprint_grow.delete_all:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/xxx.yaml...
INFO Starting log output from xxx using esphome API
INFO Successfully connected to fingerprint-haustur @ xxx in 0.037s
INFO Successful handshake with fingerprint-haustur @ xxx in 2.651s
[21:42:50][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 19 2024, 18:27:49
[21:42:50][C][wifi:600]: WiFi:
[21:42:50][C][wifi:428]:   Local MAC: 30:83:98:9E:29:CB
[21:42:50][C][wifi:433]:   SSID: [redacted]
[21:42:50][C][wifi:436]:   IP Address: xxx
[21:42:50][C][wifi:439]:   BSSID: [redacted]
[21:42:50][C][wifi:441]:   Hostname: 'xxx'
[21:42:50][C][wifi:443]:   Signal strength: -57 dB ▂▄▆█
[21:42:50][V][wifi:445]:   Priority: 0.0
[21:42:50][C][wifi:447]:   Channel: 9
[21:42:50][C][wifi:448]:   Subnet: 255.255.255.0
[21:42:50][C][wifi:449]:   Gateway: xxx
[21:42:50][C][wifi:450]:   DNS1: xxx
[21:42:50][C][wifi:451]:   DNS2: 8.8.4.4
[21:42:50][C][logger:185]: Logger:
[21:42:50][C][logger:186]:   Level: VERY_VERBOSE
[21:42:50][C][logger:188]:   Log Baud Rate: 115200
[21:42:50][C][logger:189]:   Hardware UART: UART0
[21:42:50][C][uart.arduino_esp8266:118]: UART Bus:
[21:42:50][C][uart.arduino_esp8266:119]:   TX Pin: GPIO16
[21:42:50][C][uart.arduino_esp8266:120]:   RX Pin: GPIO5
[21:42:50][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[21:42:50][C][uart.arduino_esp8266:124]:   Baud Rate: 57600 baud
[21:42:50][C][uart.arduino_esp8266:125]:   Data Bits: 8
[21:42:50][C][uart.arduino_esp8266:126]:   Parity: NONE
[21:42:50][C][uart.arduino_esp8266:127]:   Stop bits: 1
[21:42:50][C][uart.arduino_esp8266:131]:   Using software serial
[21:42:50][C][fingerprint_grow:537]: GROW_FINGERPRINT_READER:
[21:42:50][C][fingerprint_grow:538]:   System Identifier Code: 0x0000
[21:42:50][C][fingerprint_grow:539]:   Touch Sensing Pin: GPIO4
[21:42:50][C][fingerprint_grow:541]:   Sensor Power Pin: None
[21:42:50][C][fingerprint_grow:546]:   Idle Period to Sleep: Never
[21:42:50][C][fingerprint_grow:548]:   Update Interval: 0.5s
[21:42:50][E][component:082]:   Component fingerprint_grow is marked FAILED
[21:42:50][C][captive_portal:089]: Captive Portal:
[21:42:50][C][mdns:116]: mDNS:
[21:42:50][C][mdns:117]:   Hostname: xxx
[21:42:50][V][mdns:118]:   Services:
[21:42:50][V][mdns:120]:   - _esphomelib, _tcp, 6053
[21:42:50][V][mdns:122]:     TXT: friendly_name = xxx
[21:42:50][V][mdns:122]:     TXT: version = 2024.10.0
[21:42:50][V][mdns:122]:     TXT: mac = xxx
[21:42:50][V][mdns:122]:     TXT: platform = ESP8266
[21:42:50][V][mdns:122]:     TXT: board = d1_mini
[21:42:50][V][mdns:122]:     TXT: network = wifi
[21:42:50][V][mdns:122]:     TXT: api_encryption = xxx
[21:42:50][C][esphome.ota:073]: Over-The-Air updates:
[21:42:50][C][esphome.ota:074]:   Address: xxx.local:8266
[21:42:50][C][esphome.ota:075]:   Version: 2
[21:42:50][C][esphome.ota:078]:   Password configured
[21:42:50][C][safe_mode:018]: Safe Mode:
[21:42:50][C][safe_mode:019]:   Boot considered successful after 60 seconds
[21:42:50][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:42:50][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[21:42:50][C][api:140]: API Server:
[21:42:50][C][api:141]:   Address: xxx:6053
[21:42:50][C][api:143]:   Using noise encryption: YES
[21:42:52][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=607898 (now=667898)
```

drublin, 2024-10-26:
i have the same issue. it randomly happened earlier, now is always dead. The sensor does NOT go to sleep after timeout (this is a ESP action, not R503), though other ESP32 pins work, HA API works. During Enroll the sensor LED glows (with wrong color) but doesn't enroll.

an older similar but dead issue [HERE](https://github.com/esphome/issues/issues/5724)

Pytonballoon810, 2024-10-26:
I actually fixed my issue but I don't know exactly what did it: I switched the RX and TX pins as I think I wired them up incorrectly because RX and TX must be switched up such that the RX pin on the ESP connects to the TX pin on the sensor. Also I chose to use the hardware serial gpio for the communication between the sensor and the ESP as software serial seems to be unstable at such high baud rates (57000). After disabling the logging via the config by setting the baud rate to 0, it worked like a charm.

drublin, 2024-10-28:
i've succeded to make it work by migrating to another uart pins. UART0 and UART2 are not working, although it was surely working on UART2 when i started the prototype.
at some moment R503 was continuously scanning a finger, and thus returning "mismatch" and failing to put the sensor to sleep.
i have two sensors and both behave identically. So i still keep an eye on it as i still consider it unstable.

drublin, 2024-11-11:
UPD: in my case the R503 was responding to Aura LED commands, starting the Enroll action but was not reading the normal and enroll scans. During boot it was also marked as FAILED as OP mention. This showed me that UART config was ok, and the issue was with Wakeup or "sensing pin". Sharing some troubleshooting steps:

1. Connect the ESP (in my case ESP8266 D1 mini) to PC using Putty. Baudrate 74880 during boot.
 I saw boot mode(7,7), meaning the ESP boots in flash mode because one special pin (GPIO15/TX2) was High during boot. Check [HERE](https://esphome.io/components/esp8266.html). Using a 100k pull-down for this pin helped boot normally, but didn't solve the R503 issue.

2. Luckily i had another R503 so i could confirm it was a sensor issue that the pin was High during boot, and the second sensor worked as expected.
 
3. i initially ordered two R503 sensors (proto/final) in two sizes: M22 and M25. I needed M22 for my final setup and unlyckily the M22 was broken. I was able to switch the aluminium casings by carving out the Silicone and then gluing it back with hotglue gun. 
i tested it on the broken sensor and damaged the electronics while carving. Don't go to much below cable connector as the sensor will come out to front anyway. And original silicone isolates the electronic parts from the hot glue.

[image]
[image]

closed 2025-01-07 by Pytonballoon810