esphome/issues #6351: Modbus - problem reading registers above address 8192 (0x2000)

SCiunczyk, 2024-10-20:
### The problem

I am working on a proxy wifi adapter for a modbus device [https://www.creasol.it/evse](https://www.creasol.it/evse). I have a problem reading registers above address 8192 (0x2000). Only the first read is correct, subsequent reads return strange values (stable - not random). Interestingly, the correct readings are registers in the address range 0-11 and 513 (such addresses are provided by the device). Device is OK - checked from PC with USB-RS485 adapter. 

I prepared a sample yaml to show the problem.
In this example we have three registers: 9001 = 12 | 9002 = 4000 | 9003 - 800

Reading with custom_command returns the correct values
The normal read as sensor works well only for the first read.

Interestingly, writing to these high registers works fine. Using this I made a temporary workaround - I defined pairs read with custom function to sensor and write to number. But it's gluing on chewing gum ;-)

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32

### Board

XIAIO ESP32C3 (seeed_xiao_esp32c3)

### Component causing the issue

modbus

### Example YAML snippet

```yaml
substitutions:
  name: "ev-charge"
  friendly_name: ev-charge

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "my secret key ...."

# Enable logging
logger:
  level: DEBUG

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  ap: {}
[... 164 lines cut ...]
[09:22:06][D][modbus_controller.sensor:025]: Sensor new state: 12.00
[09:22:06][D][sensor:094]: 'CUST9001': Sending state 12.00000  with 0 decimals of accuracy
[09:22:06][D][modbus_controller.sensor:025]: Sensor new state: 12.00
[09:22:06][D][sensor:094]: 'REG9001': Sending state 12.00000  with 0 decimals of accuracy
[09:22:06][D][modbus_controller.sensor:025]: Sensor new state: 30806.00
[09:22:06][D][sensor:094]: 'REG9002': Sending state 30806.00000  with 0 decimals of accuracy
[09:22:06][D][modbus_controller.sensor:025]: Sensor new state: 44474.00
[09:22:06][D][sensor:094]: 'REG9003': Sending state 44474.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 4000.00
[09:22:11][D][sensor:094]: 'CUST9002': Sending state 4000.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 800.00
[09:22:11][D][sensor:094]: 'CUST9003': Sending state 800.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 12.00
[09:22:11][D][sensor:094]: 'CUST9001': Sending state 12.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 12.00
[09:22:11][D][sensor:094]: 'REG9001': Sending state 12.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 30806.00
[09:22:11][D][sensor:094]: 'REG9002': Sending state 30806.00000  with 0 decimals of accuracy
[09:22:11][D][modbus_controller.sensor:025]: Sensor new state: 44474.00
[09:22:11][D][sensor:094]: 'REG9003': Sending state 44474.00000  with 0 decimals of accuracy
```

### Additional information

[image]

brambo123, 2025-01-19:
Can you please enable uart debugging to show the actual uart data?

A big difference between the standard registers and the custom command version is the request that is send.
The normal registers will be combined into 1 request of 3 registers.
With the manual version you make 3 separate requests.
Have you also tried to request all 3 registers in 1 request on the computer?

SCiunczyk, 2025-01-19:
Hi,

I will check tomorrow, the device is outside in electrical enclosure:
https://github.com/SCiunczyk/wallbox-DIY but it's dark night here in Poland now 😏

From PC I was able to test one by one register. By using modbus open-source tester.

You mean "enable uart debugging" on ESPHome?

brambo123, 2025-01-19:
> You mean "enable uart debugging" on ESPHome?

https://esphome.io/components/uart.html#debugging
Adding `debug:` to your uart component is probably enough.
With the raw UART data we can see if the Modbus component is making a mistake or if the device is sending incorrect data.

SCiunczyk, 2025-01-20:
Log attached, if you have a few minutes to take a look...

[logs_ev-charge_logs.txt](https://github.com/user-attachments/files/18477169/logs_ev-charge_logs.txt)

brambo123, 2025-01-20:
Your device responds with 1 register to a 3 register request.

**Request:**
e3:03:23:29:00:03:c8:05

> Unit: 0xe3
> Function code: 0x03
> Starting Address: 0x2329
> Quantity of Registers: 0x0003
> CRC: 0xc805

**Response:**
e3:03:02:00:0a:c0:55

> Unit: 0xe3
> Function code: 0x03
> Byte count: 2
> Register value: 0x000a
> CRC: 0xc055

If you add `force_new_range: true` to each modbus sensor it should make separate requests.

SCiunczyk, 2025-01-20:
One minute... I will check with this parameter

SCiunczyk, 2025-01-20:
Wow! Looks that this parameter solved my problem 

[logs_ev-charge_logs_force_new_range_true.txt](https://github.com/user-attachments/files/18477441/logs_ev-charge_logs_force_new_range_true.txt)

SCiunczyk, 2025-01-20:
It seems that the problem is on the side of the device and its response to reading multiple registers in a single query.  Thank you very much for your help @brambo123.

I am closing the topic.

closed 2025-01-20 by SCiunczyk