esphome/issues #6352: 2024.10.0 breaks LVGL touchscreen when screen is rotated

angmac, 2024-10-20:
### The problem

I use my cheap yellow display 90 degree rotated, LVGL works fine in 2024.9.x but in 2024.10.0 I get the following when touch the button on the corner (many lines scrolls up):
[I][lvgl:000]: [Warn] (64.317, +44)    indev_pointer_proc: Y is 272 which is greater than ver. res    (in lv_indev.c line #367)

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32

### Board

Cheap Yellow Display without PSRAM, ESP32-2432S028R

### Component causing the issue

Touchscreen (xpt2046) / LVGL

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    id: my_display
    model: ili9341
    spi_id: lcd
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    dimensions:
      height: 320
      width: 240
    data_rate: 40MHz
    update_interval: never
    auto_clear_enabled: false
    invert_colors: false

touchscreen:
  - platform: xpt2046
    id: my_touchscreen
    spi_id: touch
    cs_pin: GPIO33
    interrupt_pin: GPIO36
    threshold: 400
    transform:
      swap_xy: false
      mirror_x: true
    calibration:
      x_min: 241
      x_max: 3928
      y_min: 200
      y_max: 3883
lvgl:
  displays:
    - my_display
  touchscreens:
    - my_touchscreen
  buffer_size: 25%
```

### Anything in the logs that might be useful for us?

```txt
[10:41:35][D][xpt2046:055]: Touchscreen Update [739, 709], z = 1888
[10:41:35][I][lvgl:000]: [Warn] (257.700, +257700)       indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
[10:41:35][I][lvgl:000]: [Warn] (257.745, +45)   indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
[10:41:35][I][lvgl:000]: [Warn] (257.790, +45)   indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
[10:41:35][I][lvgl:000]: [Warn] (257.838, +48)   indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
[10:41:35][I][lvgl:000]: [Warn] (257.883, +45)   indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
[10:41:35][I][lvgl:000]: [Warn] (257.928, +45)   indev_pointer_proc: Y is 276 which is greater than ver. res    (in lv_indev.c line #367)
```

### Additional information

It works fine in 2024.9.x
Maybe the following patch breaks the LVGL: [touchscreen] Fix coordinates when using rotation #7591

reloc4, 2024-10-21:
Same problem here:

But thanks for pointing out thas it is the rotation. 
Working on my first projekt so I thought It's my fault.
Now it works with rotation: 0

Using: Lilygo TDisplay S3

```
esphome:
  name: lilygo
  friendly_name: lilygo
  platformio_options:
    board_build.f_flash: 80000000L
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi

external_components:
  - source: github://landonr/lilygo-tdisplays3-esphome
    components: [tdisplays3]

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
  flash_size: 16MB
...
display:
  - platform: tdisplays3
    id: disp
    rotation: 270
    auto_clear_enabled: false
    
touchscreen:
  - platform: cst816
    id: tsrc
    display: disp
    interrupt_pin: 16
    reset_pin: 21
    i2c_id: touchscr
    calibration:
      x_min: 0
      x_max: 169
      #x_max: 318
      y_min: 0
      y_max: 318
      #y_max: 169
    transform:
      swap_xy: true
      mirror_x: true
      mirror_y: false
    on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );
          

time:
  - platform: homeassistant
    id: ha_time

lvgl:
  displays:
    - disp
  touchscreens:
    - tsrc
```

Error Message:

```
[23:54:42][I][lvgl:000]: [Warn]	(21.195, +21195)	 indev_pointer_proc: Y is -40 which is smaller than zero 	(in lv_indev.c line #364)
[23:54:42][I][lvgl:000]: [Warn]	(21.236, +41)	 indev_pointer_proc: Y is -84 which is smaller than zero 	(in lv_indev.c line #364)
[23:54:42][I][lvgl:000]: [Warn]	(21.277, +41)	 indev_pointer_proc: Y is -126 which is smaller than zero 	(in lv_indev.c line #364)
[23:54:42][I][lvgl:000]: [Warn]	(21.318, +41)	 indev_pointer_proc: Y is -138 which is smaller than zero 	(in lv_indev.c line #364)
[23:54:43][I][lvgl:000]: [Warn]	(21.359, +41)	 indev_pointer_proc: Y is -144 which is smaller than zero 	(in lv_indev.c line #364
```)

clydebarrow, 2024-10-22:
There is an issue in the current release, which is fixed by[ PR #7591](https://github.com/esphome/esphome/pull/7591). Although that PR has been merged, it does not seem that it will make it into the upcoming patch release.
It only matters though if you use `rotation:` which is usually not necessary in any case.

1. Rather than using `rotation:` in the display config, use `transform:` and set the `dimensions:` to reflect the resulting display size - it's faster and results in the touchscreen getting the right display dimensions.
2. Use `transform:` in the touchscreen config to get the touch working right - it may need some experimentation with different values for x and y mirror as touchscreens are often not oriented in the same way as the display screen!
3. Make sure when using `calibration:` (required only for resistive, i.e. xpt2046) that the min values are less than the max values (not an issue in the configs above, but I have seen it elsewhere.)
4. Do not use `calibration:` for capacitive touchscreens. It's not required and just complicates things.

closed 2024-10-22 by clydebarrow