esphome/issues #6353: ignore_efuse_mac_crc is not supported on ESP32C3

fra87, 2024-10-20:
### The problem

After updating to 2024.10.0, trying to install a working project results in the error in the title.

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

ESP32-C3 Super Mini

### Example YAML snippet

```yaml
---
# Test snippet for ESP32-C3 SuperMini

esphome:
  name: 'esph-test'

esp32:
  board: lolin_c3_mini
  framework:
    type: esp-idf
    version: recommended
    # Custom sdkconfig options
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: false

api:

ota:
  - platform: esphome
    id: component_ota
    password: !secret ota_password
   
wifi:
  id: component_wifi
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Failed config

esp32: None
  
  ignore_efuse_mac_crc is not supported on ESP32C3.
  board: lolin_c3_mini
  framework: 
    version: 4.4.8
    sdkconfig_options: 
      COMPILER_OPTIMIZATION_SIZE: y
    advanced: 
      ignore_efuse_mac_crc: False
      ignore_efuse_custom_mac: False
    components: []
    source: ~3.40408.0
    platform_version: platformio/espressif32@5.4.0
    type: esp-idf
```

### Additional information

I used the configuration part from the docs, together with minimal additional info: https://esphome.io/components/esp32.html#esp-idf-framework

Changing `lolin_c3_mini` to `esp32-c3-devkitm-1` doesn't change anything in the outcome

SOLVED by removing the following lines from the configuration

```
    advanced:
      ignore_efuse_mac_crc: false
```

However I don't know how much this can affect fuctionalities, so please advise if this shall be removed or replaced with something else

randybb, 2024-10-20:
Just curious, why do you need that? Are you using EFUSE for overriding MAC address and the stored MAC is not valid?

kbx81, 2024-10-20:
The MAC CRC check is not supported on the C3 -- so enabling/disabling the check in config accomplishes nothing. Therefore, if included in a config for the C3, a validation error is thrown because it's meaningless on the hardware platform.

closed 2024-10-20 by kbx81

fra87, 2024-10-20:
> Just curious, why do you need that? Are you using EFUSE for overriding MAC address and the stored MAC is not valid?

@randybb No, I'm not using this specific feature. I added it in the configuration following the recommendations on [ESPHome official webpage](https://esphome.io/components/esp32.html#esp-idf-framework), where there is also a specific snippet for this specific board.

> The MAC CRC check is not supported on the C3 -- so enabling/disabling the check in config accomplishes nothing. Therefore, if included in a config for the C3, a validation error is thrown because it's meaningless on the hardware platform.

@kbx81 Ok thanks for the confirmation. I'll leave it out of the config then. Just two remarks

- The [help page](https://esphome.io/components/esp32.html#esp-idf-framework) shall be also updated, since people looking at how to use ESP32 smaller variants will come up with a non-working config
- This is actually a breaking change for 2024.10.0, since configurations built for older versions following the recommendations stop working on 2024.10.0. I suggest to highlight this in the changelog.

Thank you :)

randybb, 2024-10-20:
This is definitely not a recommended option and  you should be always using only options that you really need (especially that option for ignoring a burn-in MAC address with a wrong CRC in eFuse).
In many cases examples are showing various option as people have no clue how to use them.

fra87, 2024-10-20:
> This is definitely not a recommended option and you should be always using only options that you really need (especially that option for ignoring a burn-in MAC address with a wrong CRC in eFuse). In many cases examples are showing various option as people have no clue how to use them.

@randybb Thanks; then I strongly think that the webpage I linked shall be modified, since without any additional explanation or hint when I read that I thought that option was actually needed when using ESP-IDF with ESP32-C3