esphome/issues #6354: "esphome run" hangs

AndrewAkinson, 2024-10-20:
### The problem

When running 

esphome run heating.yaml

Hangs at the place shown in the log below. I

This has been logged before and was closed, incorrectly, as a support request (although to be fair they did also ask for support.)

issue #4497
https://github.com/esphome/issues/issues/4497

Whatever is wrong the process should not hang.
f it has finished shouldn't it close the logs and return to the prompt, preferably giving a success message. 
If it has failed it should exit gracefully hopefully giving a useful error message

(It appears to have worked apart from the hang as the device appears as online in HA, which is on a different computer)

thanks

Andrew

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32 -poe olimex

### Example YAML snippet

```yaml
esphome:
  name: heating
  friendly_name: Heating

esp32:
  board: esp32-poe
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXX"

ota:
  - platform: esphome
    password: "XXXXXXXXX"

wifi:
  ssid: "XXXXXXXXX"
  password: "XXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heating Fallback Hotspot"
    password: "XXXXXXXXX"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Processing heating (board: esp32-poe; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
.
.
.
.
[11:25:53][W][component:157]: Component api set Warning flag: unspecified
[11:25:53][I][app:100]: ESPHome version 2024.10.0 compiled on Oct 20 2024, 10:54:48
[11:25:53][C][wifi:600]: WiFi:
[11:25:53][C][wifi:428]:   Local MAC: C8:F0:9E:6A:0D:7C
[11:25:53][C][wifi:433]:   SSID: 'Goodlife'
[11:25:53][C][wifi:436]:   IP Address: 192.168.0.66
[11:25:53][C][wifi:440]:   BSSID: 1C:61:B4:3D:66:4C
[11:25:53][C][wifi:441]:   Hostname: 'heating'
[11:25:53][C][wifi:443]:   Signal strength: -76 dB ▂▄▆█
[11:25:53][C][wifi:447]:   Channel: 6
[11:25:53][C][wifi:448]:   Subnet: 255.255.255.0
[11:25:53][C][wifi:449]:   Gateway: 192.168.0.1
[11:25:53][C][wifi:450]:   DNS1: 192.168.0.1
[11:25:53][C][wifi:451]:   DNS2: 192.168.0.1
[11:25:53][C][logger:185]: Logger:
[11:25:53][C][logger:186]:   Level: DEBUG
[11:25:53][C][logger:188]:   Log Baud Rate: 115200
[11:25:53][C][logger:189]:   Hardware UART: UART0
[11:25:53][C][captive_portal:089]: Captive Portal:
[11:25:53][C][mdns:116]: mDNS:
[11:25:53][C][mdns:117]:   Hostname: heating
[11:25:53][C][esphome.ota:073]: Over-The-Air updates:
[11:25:53][C][esphome.ota:074]:   Address: heating.local:3232
[11:25:53][C][esphome.ota:075]:   Version: 2
[11:25:53][C][esphome.ota:078]:   Password configured
[11:25:53][C][safe_mode:018]: Safe Mode:
[11:25:53][C][safe_mode:020]:   Boot considered successful after 60 seconds
[11:25:53][C][safe_mode:021]:   Invoke after 10 boot attempts
[11:25:53][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[11:25:53][C][api:140]: API Server:
[11:25:53][C][api:141]:   Address: heating.local:6053
[11:25:53][C][api:143]:   Using noise encryption: YES
[11:26:45][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[11:26:45][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:26:45][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

randybb, 2024-10-20:
What do you expect it will be doing? Whole initiation was executed without problems, boot loop counter stored and that's it. And now it is just doing nothing - there is no finish, because it how MCU works, just looping in main() and doing its job, in your case nothing and printing that nothing.

AndrewAkinson, 2024-10-21:
Yes the logs are do nothing, so at that point the logs are not needed.  I would expect a command from the terminal that did other things other than open the logs, to close the logs when it was finished and return to the prompt. Ideally a success message would be sent, but that would be icing on the cake.  There are other points earlier in the process where the logs pause for a reasonably long time and the process is not finished.
Currently it is just bug that is not very neat and very ungraceful, not one that breaks the command.
thanks
Andrew

closed 2024-10-21 by randybb

AndrewAkinson, 2024-10-21:
> If you don't need `logs`, then just use arguments `compile` and then `upload`.

That is not really a solution to the problem as originally As you have closed it I guess you are happy with the standards that it sets, shame.

Thanks for your time in replying and explaining.

ssieb, 2024-10-21:
What do you mean by "hang"?  What is the problem?
The information you provided in the initial post is also incorrect.  How are you doing `esphome run` in the HA addon?

AndrewAkinson, 2024-10-22:
> What do you mean by "hang"? What is the problem?

The log file is left open, this is not neat, how is the user meant to know that the upload is finished. There are a few times in the process where the process does not produce output for some time. Described in earlier comments are what I would expect to happen

> The information you provided in the initial post is also incorrect. 

If you let me know what is wrong I can attempt to correct it, but you probably mean I have incorrectly selected

>What type of installation are you using?

>Home Assistant Add-on

>How are you doing `esphome run` in the HA addon?

So I have esphome installed using pip in a venv

(I cannot use ESPhome web or HA for upload as I'm using Firefox which does not officially support Webserial)

ssieb, 2024-10-22:
If you're going to run a command, then you need to know what it does.  `esphome run` is literally a shortcut that runs three commands: `esphome compile`, `esphome upload`, and `esphome logs`.  You are complaining that the `esphome logs` command is doing what it is supposed to do.  It connects to the device and displays the logs as they are generated.  It will keep doing so until you tell it to stop.  If you don't like this, then don't run that command.

AndrewAkinson, 2024-10-23:
You do not need to be condescending. In a similar vein, to be able to understand and give a justified reply to the original report you need to understand good quality coding.

When I ran the command I knew full well what it did, plus randybb had already confirmed it in a civilized way, in case I did not. I'm reasonably sure that randybb had a clear understanding of what I was suggesting but either thought its not a problem or not worth the time to fix. I disagreed, however, it is not my position to make these decisions. I thanked randybb for taking the time to look and understand. I was a bit curious so read the code, a reasonably graceful/clearer solution is probably very easy to implement, a truly elegant one is more difficult. That would have been a fine end to it and maybe I would have amended the code.

>If you don't like this, then don't run that command.

This attitude is disappointing when someone takes time to suggest improvement. It has to be said the bug reporting responses in esphome are on the more toxic end than lots of projects. I am happy for people to say that they disagree or do not think that it is worth the time to fix. Please feeding back is a most valuable process that has the power to massively improve software, aggressive condescending responses  just reduces its value. I'm unlikely to file another bug report I'm certainly not going to add any code, this has already wasted too much time. I wonder how many others this sort of thing has put off. It is a real shame as esphome is a great idea.

Anyway, time I stopped wasting my time and got on with something useful.