esphome/issues #6360: Libraries compilation failing

vi350, 2024-10-21:
### The problem

Project was previously compiled with ESPHome version 2024.7.2 in the end of july. I've updated esphome container recently to the 2024.9.2. Device had gone buggy, I decided to reboot it via install (ota update), which led to such and error:

```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/ceiling-light.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ceiling-light (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o
Compiling .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha512/cp/hash_sha512_cp.c.o
/tmp/ccZFnzkL.s: Assembler messages:
/tmp/ccZFnzkL.s:3358: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccZFnzkL.s:3360: Error: unrecognized symbol type ""
/tmp/ccZFnzkL.s:3360: Error: junk at end of line, first unrecognized character is `+'
/tmp/ccZFnzkL.s:3361: Warning: missing closing `"'
/tmp/ccZFnzkL.s:3361: Error: unknown opcode or format name 'crypto_hash_sha256_update'
/tmp/ccZFnzkL.s:3366: Error: bad register name: 	sp
/tmp/ccZFnzkL.s:3366: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccZFnzkL.s:3366: Internal error in demand_empty_rest_of_line at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/binutils/gas/read.c:3788.
Please report this bug.
*** [.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o] Error 1
/tmp/ccupP7qe.s: Assembler messages:
/tmp/ccupP7qe.s:7610: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccupP7qe.s:7612: Error: unrecognized symbol type ""
/tmp/ccupP7qe.s:7612: Error: junk at end of line, first unrecognized character is `+'
/tmp/ccupP7qe.s:7613: Warning: missing closing `"'
/tmp/ccupP7qe.s:7613: Error: unknown opcode or format name 'crypto_hash_sha512_update'
/tmp/ccupP7qe.s:7618: Error: bad register name: 	sp
/tmp/ccupP7qe.s:7618: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccupP7qe.s:7618: Internal error in demand_empty_rest_of_line at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/binutils/gas/read.c:3788.
Please report this bug.
*** [.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha512/cp/hash_sha512_cp.c.o] Error 1
========================= [FAILED] Took 10.43 seconds =========================
```

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

lolin_s2_mini

### Component causing the issue

.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o

### Example YAML snippet

```yaml
esphome:
  name: ceiling-light
  friendly_name: ceiling-light

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

# Enable logging
logger:

api:
  encryption:
    key: "*"

ota:
  - platform: esphome
    password: "*"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "*"
    password: "*"
  power_save_mode: none
  output_power: 8.5

captive_portal:
    

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO16
      mode:
        input: true
        pullup: true
    name: "switch"
    id: wall_switch
    filters:
      - delayed_on: 200ms
      - delayed_off: 200ms
    on_state:
      - light.toggle: ceiling_light

light:
  - platform: cwww
    id: ceiling_light
    name: "light"
    cold_white: gpio_cold
    warm_white: gpio_warm
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3000 K
    constant_brightness: true
    gamma_correct: 3.5
    default_transition_length: 2s
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: rgb
    id: ceiling_light_rgb
    name: "rgb"
    red: gpio_red
    green: gpio_green
    blue: gpio_blue
    gamma_correct: 2.5
    default_transition_length: 2s
    restore_mode: RESTORE_DEFAULT_OFF

output:
  - platform: ledc
    id: gpio_red
    pin: GPIO10
  - platform: ledc
    id: gpio_green
    pin: GPIO8
  - platform: ledc
    id: gpio_blue
    pin: GPIO6
  - platform: ledc
    id: gpio_cold
    pin: GPIO14
  - platform: ledc
    id: gpio_warm
    pin: GPIO13
```

ssieb, 2024-10-21:
You have corrupted files.  Try a clean build and if that doesn't help, rebuild the docker container.  Make sure you back up any yaml files.

vi350, 2024-10-22:
```
Compiling .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/crypto_hash.c.o
Compiling .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o
/tmp/ccySqqsj.s: Assembler messages:
/tmp/ccySqqsj.s:3358: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccySqqsj.s:3360: Error: unrecognized symbol type ""
/tmp/ccySqqsj.s:3360: Error: junk at end of line, first unrecognized character is `+'
/tmp/ccySqqsj.s:3361: Warning: missing closing `"'
/tmp/ccySqqsj.s:3361: Error: unknown opcode or format name 'crypto_hash_sha256_update'
/tmp/ccySqqsj.s:3366: Error: bad register name: 	sp
/tmp/ccySqqsj.s:3366: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccySqqsj.s:3366: Internal error in demand_empty_rest_of_line at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/binutils/gas/read.c:3788.
Please report this bug.
*** [.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o] Error 1
========================= [FAILED] Took 121.68 seconds =========================
```
Still occurs. (removed build from .esphome and cache from .esphome/platformio)
I'm using "latest" image tag. Do I need to use another tag? Or build it on my own?
I think I can try removing esphome installation completely with all yaml files backup. In that case I'll comment within a week.

ssieb, 2024-10-22:
Can you find the `hash_sha256_cp.c` source file?  It comes from:
```
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
```
The path should be:
`build/<name>/.piolibdeps/<name>/libsodium/libsodium/src/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c`
See what's in that file.

vi350, 2024-10-27:
[hash_sha256_cp.txt](https://github.com/user-attachments/files/17535481/hash_sha256_cp.txt)

ssieb, 2024-10-27:
If it's still failing, does that `/tmp` file exist?

vi350, 2024-10-28:
```
Compiling .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha512/cp/hash_sha512_cp.c.o
/tmp/ccCJZAGx.s: Assembler messages:
/tmp/ccCJZAGx.s:3358: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccCJZAGx.s:3360: Error: unrecognized symbol type ""
/tmp/ccCJZAGx.s:3360: Error: junk at end of line, first unrecognized character is `+'
/tmp/ccCJZAGx.s:3361: Warning: missing closing `"'
/tmp/ccCJZAGx.s:3361: Error: unknown opcode or format name 'crypto_hash_sha256_update'
/tmp/ccCJZAGx.s:3366: Error: bad register name: 	sp
/tmp/ccCJZAGx.s:3366: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccCJZAGx.s:3366: Internal error in demand_empty_rest_of_line at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/binutils/gas/read.c:3788.
Please report this bug.
*** [.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/hash_sha256_cp.c.o] Error 1
/tmp/ccYIhHdY.s: Assembler messages:
/tmp/ccYIhHdY.s:7610: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccYIhHdY.s:7612: Error: unrecognized symbol type ""
/tmp/ccYIhHdY.s:7612: Error: junk at end of line, first unrecognized character is `+'
/tmp/ccYIhHdY.s:7613: Warning: missing closing `"'
/tmp/ccYIhHdY.s:7613: Error: unknown opcode or format name 'crypto_hash_sha512_update'
/tmp/ccYIhHdY.s:7618: Error: bad register name: 	sp
/tmp/ccYIhHdY.s:7618: Error: junk at end of line, first unrecognized character valued 0x9
/tmp/ccYIhHdY.s:7618: Internal error in demand_empty_rest_of_line at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/binutils/gas/read.c:3788.
Please report this bug.
*** [.pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha512/cp/hash_sha512_cp.c.o] Error 1
========================= [FAILED] Took 10.53 seconds =========================
```
So, i tried to find this file
```
~/esphome_config % sudo find . -name 'ccmIlUf9.s'
~/esphome_config % sudo find . -name 'tmp'
./.esphome/platformio/cache/tmp
~/esphome_config % cd ./.esphome/platformio/cache/tmp
~/esphome_config/.esphome/platformio/cache/tmp % ll
total 8.0K
drwxr-xr-x 2 root root 4.0K Oct 22 05:27 ./
drwxr-xr-x 5 root root 4.0K Oct 22 05:27 ../
```
So, I will provide other files I've found.
[hash_sha256_cp.c.d.txt](https://github.com/user-attachments/files/17536167/hash_sha256_cp.c.d.txt)
is from .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/cp/, but there is no hash_sha256_cp.c.o file.
[hash_sha256.c.o.txt](https://github.com/user-attachments/files/17536181/hash_sha256.c.o.txt)
is from .pioenvs/ceiling-light/lib895/libsodium/crypto_hash/sha256/.
After this I tried to login into container and managed to capture tmp .s file from /tmp folder.
[ccQbHI2i.s.txt](https://github.com/user-attachments/files/17536231/ccQbHI2i.s.txt)
If those not which requiered, tell me.

closed 2025-01-24 by vi350