esphome/issues #6361: Issue with image command and waveshare 7.50inv2 screen

jobi, 2024-10-22:
### The problem

I recently bought [this](https://www.amazon.com/gp/product/B075R4QY3L/) Waveshare e-ink display with [this](https://www.amazon.com/gp/product/B07M5CNP3B/) ESP32 driver board. I'm trying to use esphome to display screenshots of a home assistant dashboard on the e-ink screen.

I have configured the online image component to periodically download an image, and trigger an update of the display component, to display the downloaded image. I'm testing with a small test image - [image]

My problem is that when I issue the image command, the display briefly seems to show the image correctly, then it seems to flash again and ends up showing what looks like pale, inverse version of the original image. I have tried to capture this in [this video](https://drive.google.com/file/d/1DwPjml46bQo-4-ZFOkHCeNL41gIW62fu/view?usp=drive_link).

Issuing a print command with some text instead seems to work fine.

Below I provide the relevant sections of the configuration file. I have tried a few variations (using some of the other `model` variants, not setting `reset_duration` but so far haven't been able to solve the issue.

Thanks

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

Waveshare Universal e-Paper Driver Board with WiFi/Bluetoth SoC ESP32

### Component causing the issue

display / waveshare_epaper

### Example YAML snippet

```yaml
online_image:
  - url: <the url to the image on a local web server>
    format: png
    id: dashboard
    update_interval: 60s
    on_download_finished:
      component.update: my_display

display:
  - platform: waveshare_epaper
    id: my_display
    cs_pin: 15
    dc_pin: 27
    busy_pin:
      number: GPIO25
      inverted: true
    reset_pin: 26
    model: 7.50inv2
    update_interval: never
    reset_duration: 2ms
    lambda: |
     //it.print(0, 0, id(segoe), "E-Paper is awesome!");
     it.image(0, 0, id(dashboard));
```

jobi, 2024-10-26:
Switching the switch on the ESP32 driver board from B to A seems to have improved things a lot. I still have an issue where the text appears correctly but the image appears to have inverted colors:

[image]

m1212e, 2024-11-21:
Got the exact same problem, noticed that I can display things fine when inverting the colors and using BINARY on the display. But only if there is not more than 50% content on the display. Very strange. Did you manage to resolve this?

gideonyong, 2025-04-03:
I have something similar but slightly different.  If i set this to use image instead of online_image, I have no problem displaying this png but if I use online_image, I am not able to see the picture (not even a faint outline).

`    lambda: |-
      it.image(0,0, id(static_image));     #THIS WORKS!!
#      it.image(0, 0, id(dynamic_image), COLOR_OFF, COLOR_ON);    #THIS DOES NOT WORK

#      it.image(0, 0, id(esphomelogo), COLOR_OFF, COLOR_ON);          

#            it.image(0, 0, id(my_image), COLOR_OFF, COLOR_ON);      

image:    #THIS WORKS!!
  - file: "http://192.168.1.126:5050/"
    type: BINARY
    invert_alpha: true
    id: static_image  

http_request:
    verify_ssl: false

online_image:
  - id: dynamic_image
    url: "http://192.168.1.126:5050/"
    type: BINARY
    format: png
    update_interval: 120s`

TheSmileyCash, 2026-09-18:
> I have something similar but slightly different. If i set this to use image instead of online_image, I have no problem displaying this png but if I use online_image, I am not able to see the picture (not even a faint outline).
> 
> ```
> lambda: |-
>       it.image(0,0, id(static_image));     #THIS WORKS!!
> #      it.image(0, 0, id(dynamic_image), COLOR_OFF, COLOR_ON);    #THIS DOES NOT WORK
> 
> image:    #THIS WORKS!!
>   - file: "http://192.168.1.126:5050/"
>     type: BINARY
>     invert_alpha: true
>     id: static_image  
> 
> http_request:
>     verify_ssl: false
> 
> online_image:
>   - id: dynamic_image
>     url: "http://192.168.1.126:5050/"
>     type: BINARY
>     format: png
>     update_interval: 120s
> ```

Did you ever figure this out? I am facing the same issue