esphome/issues #6362: Touchpad for axs15231 device has changed at 2024.10.0

DMonkey-hub, 2024-10-22:
### The problem

I have a JC3248W535 device (ESP32-S3 3.5-inch capacitive touch IPS module 320 * 480). The touchpad was working fine in 2024.9.2 but since the update to 2024.10.0 the coordinates are all wrong. I have gone back to 2024.9.2 and all works fine again.

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32-IDF

### Board

JC3248W535

### Component causing the issue

display platform: axs15231

### Example YAML snippet

```yaml
A full example can be found here...
https://github.com/DMonkey-hub/esphome/tree/main
with the associated yaml here...
https://github.com/DMonkey-hub/esphome/tree/main/examples/JC3248W535EN
```

### Anything in the logs that might be useful for us?

```txt
no
```

### Additional information

The example has 4 buttons the change the Test Meter reading but using 2024.10.0 the button positions move to the right. For instance button 10 works fine, button 20 can be pressed on button 30 and 30 is pressed just to the right of 40. I know the driver is used for the T-Display_S3_Long device so it may impact those as well.
Everthing works as expected when reverting to 2024.9.2

randybb, 2024-10-22:
There was a fix for touchscreen to fix coordinates https://github.com/esphome/esphome/pull/7591 maybe this has an impact to your component. 

Btw, a touchscreen support for this one was added recently https://github.com/esphome/esphome/pull/7592

DMonkey-hub, 2024-10-22:
Thanks for the info, yes I believe that the update did have an impact on the issue.
I've followed the docs and have made some adjustments and improvements but I now can't get the touchpad to get passed the 320 ymax coord. The device is 320x480 but the effect I now see is as if the touchpad is only seeing 320x320. Strange.

clydebarrow, 2024-10-28:
For issues with external components you'll need to contact the author of the component. If you have a problem with a built-in ESPHome component, here is the right place, but you'll get a faster response if you provide a minimal YAML example in the bug report.

DMonkey-hub, 2024-10-28:
Thanks Clyde,
I fixed the issue with the built in Touchscreen component with a custom one. I don't think the latest update was compatible with this component but all working again now thanks.

closed 2024-10-28 by DMonkey-hub

clydebarrow, 2024-10-28:
The display is  directly supported in ESPHome now as well - in the `dev` branch at least, with proper partial update functionality.