esphome/issues #6365: ESP32-C6 crashes at runtime while using I2C

rvalitov, 2024-10-22:
### The problem

I'm using ESP32-C6 with bh1750 sensor. It needs I2C. Enabling I2C leads to regular continuous runtime crashes. The watchdog restarts the device, but it crashes again.

### Which version of ESPHome has the issue?

2024.10.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

esp32-c6-devkitc-1

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: garage_sensors
  friendly_name: Garage Sensors
  platformio_options:
    board_build.flash_mode: dio 

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y
     
# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: ""

wifi:
  ssid: ""
  password: ""
 
i2c:
  sda: 6
  scl: 7
  scan: true
    
sensor:
[... 194 lines cut ...]
[18:43:35]I (658) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[18:43:35]I (664) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[18:43:35]I (671) spi_flash: detected chip: generic
[18:43:35]I (675) spi_flash: flash io: dio
[18:43:35]W (679) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
[18:43:35]W (692) tsens: legacy driver is deprecated, please migrate to `driver/temperature_sensor.h`
[18:43:35]W (701) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
[18:43:35]W (712) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
[18:43:35]I (723) sleep: Configure to isolate all GPIO pins in sleep state
[18:43:35]I (729) sleep: Enable automatic switching of GPIO sleep configuration
[18:43:35]I (737) coexist: coex firmware version: 4482466
[18:43:35]I (742) coexist: coexist rom version 5b8dcfa
[18:43:35]I (747) main_task: Started on CPU0
[18:43:35]I (750) main_task: Calling app_main()
[18:43:35]I (854) main_task: Returned from app_main()
[18:43:45]E (10868) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[18:43:45]E (10868) task_wdt:  - loopTask (CPU 0)
[18:43:45]E (10868) task_wdt: Tasks currently running:
[18:43:45]E (10868) task_wdt: CPU 0: loopTask
[18:43:45]E (10868) task_wdt: Aborting.
```

### Additional information

https://github.com/esphome/esphome/discussions/6957

rvalitov, 2024-10-22:
Setting `scan: false` helps to avoid the crash.

luca-angemi, 2024-10-23:
I'm not sure this is related but my muse luxe uses i2c and keeps crashing in 2024.10.1 (works fine in 2024.10.0 or 2024.9.2)

I see logs in esphome crashing when starting the microphone.

```
2024-10-22 08:11:28.745 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
File "aioesphomeapi/connection.py", line 833, in aioesphomeapi.connection.APIConnection.send_messages_await_response_complex
TimeoutError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/esphome/assist_satellite.py", line 183, in _update_satellite_config
config = await self.cli.get_voice_assistant_configuration(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/aioesphomeapi/client.py", line 1469, in get_voice_assistant_configuration
resp = await self._get_connection().send_message_await_response(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "aioesphomeapi/connection.py", line 851, in send_message_await_response
File "aioesphomeapi/connection.py", line 837, in send_messages_await_response_complex
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for VoiceAssistantConfigurationResponse after 5s
```

rvalitov, 2024-11-08:
Perhaps related to #6366

CamronBorealis, 2024-11-18:
I am getting the same issue on the ESP32-C6-DEVKITC-1 when attempting to connect to a BMP390 sensor

pdcastro, 2024-12-07:
> _Setting scan: false helps to avoid the crash._

Though I did not test it myself, in https://github.com/esphome/issues/issues/5864#issuecomment-2518815454 @lucaskatayama shared that he _”needed to increase [the I2C] frequency, otherwise it crashes with some watchdog error,”_ and I understand that he got I2C working with `scan: true` as follows:

```yaml
i2c:
  ...
  scan: true
  frequency: 200kHz
```

Note however that his board is the XIAO ESP32-C6 rather than the Espressif ESP32-C6-DevKitC-1. I don’t have either board to confirm it or deny — _*yet.*_ 🙂

dgsardina, 2025-02-11:
> I am getting the same issue on the ESP32-C6-DEVKITC-1 when attempting to connect to a BMP390 sensor

@CamronBorealis  Were you able to get it working?

ximex, 2025-02-25:
maybe related to: https://github.com/esphome/issues/issues/6639

bdraco, 2025-06-14:
Can you post a backtrace ? https://deploy-preview-4995--esphome.netlify.app/guides/troubleshooting

rvalitov, 2025-06-21:
> @rvalitov is it still crashing for you? for me i don't have to set `scan: false` any more with my `esp32c6`

May I know what ESPHome version you use? Do you use the same config as mine?

GrumpyMeow, 2025-06-27:
My Xiao ESP32c6 device with [mmwave](https://github.com/limengdu/MR60BHA2_ESPHome_external_components) also keeps rebooting.  
```
<0x1b>[0;35m[C][safe_mode:080]: There have been 0 suspected unsuccessful boot attempts<0x1b>[0m
<0x1b>[0;36m[D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed<0x1b>[0m
<0x1b>[0;32m[I][app:048]: Running through setup()<0x1b>[0m
<0x1b>[0;35m[C][i2c.idf:021]: Running setup<0x1b>[0m
<0x1b>[0;32m[I][i2c.idf:262]: Performing bus recovery<0x1b>[0m
<0x1b>[0;36m[D][esp-idf:000]: <0x1b>[0;32mI (327) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 <0x1b>[0m<0x1b>[0m
<0x1b>[0;36m[Dwing tasks/users did not reset the watchdog in time:<0x1b>[0m
<0x1b>[0;31mE (5326) task_wdt:  - loopTask (CPU 0)<0x1b>[0m
<0x1b>[0;31mE (5326) task_wdt: Tasks currently running:<0x1b>[0m
<0x1b>[0;31mE (5326) task_wdt: CPU 0: loopTask<0x1b>[0m
<0x1b>[0;31mE (5326) task_wdt: Aborting.<0x1b>[0m

Core  0 register dump:
MEPC    : 0x42027d82  RA      : 0x42028e90  SP      : 0x4082b3e0  GP      : 0x40819ff4  
TP      : 0x4082b500  T0      : 0x00000000  T1      : 0x400009f4  T2      : 0xffffffff  
S0/FP   : 0x00000000  S1      : 0x40825e44  A0      : 0x00000000  A1      : 0x40825f68  
A2      : 0x00000001  A3      : 0x00000800  A4      : 0x00000013  A5      : 0x60004000  
A6      : 0x00000004  A7      : 0x00000014  S2      : 0x00000000  S3      : 0x00000067  
S4      : 0x00000014  S5      : 0x00000053  S6      : 0x000003e8  S7      : 0x00000002  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x40823000  T4      : 0x00000001  T5      : 0x40823000  T6      : 0x00000001  
MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x40825f94  MTVAL   : 0x00000001  
MHARTID : 0x00000004  
```
Adding to i2c resolves the crashing/reboot: 
```
  scan: false 
  frequency: 200kHz
```

ximex, 2025-08-22:
@rvalitov still a problem with 2025.8.0?