esphome/issues #6368: slzb06 does not update

javifly, 2024-10-22:
### The problem

Hi,

I have a SLZB06 from smlight working as bluetooth proxy in home assistant.

Now my home assistant has detected it under the esphome integration and tells me that my firm version is 2024.9.2 and that there is a new one, the 2024.10.1 and from home assistant does not let me update the device indicating that the file for this device is not available via ota.

The manufacturers of the slzb06 tell me that everything is correct on their side. They think it's esphome bug because the firmware has ota component with esphome platform, can check it here https://github.com/smlight-dev/slzb-06-fw-esphome/blob/main/configs/bt.yaml

Kind regards, Taras
[image]

### Which version of ESPHome has the issue?

2024.10.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP8266

### Board

.

### Component causing the issue

.

javifly, 2024-10-29:
Does no one know about this matter?

randybb, 2024-11-13:
It is exactly telling what you have to do. There was a breaking change in [2024.6](https://esphome.io/changelog/2024.6.0#ota-platforms) you need to update your config accordingly. The linked config has it fixed, your local one does not.

closed 2024-11-13 by randybb

javifly, 2024-11-14:
Where should I make these changes? In which file?
`# Old
ota:

# New
ota:
  platform: esphome`

randybb, 2024-11-14:
Can you share your full config?
[image]

javifly, 2024-11-14:
yes, it is:

substitutions:
  name: slzb-06-bluetooth-f7c714
packages:
  SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-tech/testesphome/bt.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Tarik2142, 2024-11-14:
You are using outdated firmware or outdated config. You can find the current config here https://github.com/smlight-dev/slzb-06-fw-esphome/blob/main/configs/bt.yaml

javifly, 2024-11-14:
should I change?
__SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-tech/testesphome/bt.yaml

for this other?
__SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-dev/slzb-06-fw-esphome/blob/main/configs/bt.yaml](https://github.com/smlight-dev/slzb-06-fw-esphome/blob/main/configs/bt.yaml

Tarik2142, 2024-11-14:
__SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-dev/slzb-06-fw-esphome/configs/bt.yaml

javifly, 2024-11-15:
I have tried to change in the file

/homeassistant/esphome/slzb-06-bluetooth-f7c714.yaml

the line:

SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-tech/testesphome/bt.yaml

By the line:

SMLIGHT.SLZB-06(Bluetooth-only): github://smlight-dev/slzb-06-fw-esphome/configs/bt.yaml

I have restarted the Home Assistant virtual machine. But the updates are not working either. I leave screenshots of the errors obtained:
 
[image]

[image]