esphome/issues #6369: Erratic values using bme68x_bsec2_i2c component (but NOT with the bme680 component)

jb1228, 2024-10-23:
### The problem

I have been going crazy trying to get my BME680 sensors working the `bme68x_bsec2_i2c` component. The temperature and humidity values are WAY off (I cannot speak for the other sensor values, since I don't have many other sources for pressure, gas, VOCs, etc.).

After seeing all the complaints online about incorrect/high values, I was about to throw all my BME680s in the trash when I decided to try the _regular_ `bme680` component. Turns out there doesn't seem to be any issues with the sensors themselves... but with the `bme68x_bsec2_i2c` component. 

**The `bme680` component seems to work perfectly well with all values stable and within specified tolerances.**

Here are the problems I am seeing:
- When setting the `bme68x_bsec2_i2c.sample_rate` to **LP**, the temperature begins to fluctuate every few seconds between +2°F and +6°F above actual.
- When using **ULP**, the temp jumps to +20°F above actual after about 5 minutes.
- If I try to mix them by setting the `bme68x_bsec2_i2c.sample_rate` to **ULP** and ANY of the individual sensors (temp/humidity/pressure) to **LP**, I get the following error and the component fails to load:<br>`Communication failed (BSEC2 status: 14, BME68X status: 0)`
- The humidity is off as well. 
  - I did't record any of the specifics, but I can if it helps.

I have tested with several sensors/boards/configurations, all with the same, repeatable results.

- I tested 4 or 5 sensors so far from 2 different vendors
  - [Amazon](https://www.amazon.com/gp/product/B0BXKW2KVC)
  - [AliExpress](https://www.aliexpress.us/item/3256805784068021.html)
- Tested with the following boards:
  - ESP32-WROVER
  - Waveshare ESP32-S3-Zero
  - Olimex ESP32-POE-ISO-EA-IND
- **The sensors are NOT IN ENCLOSURES** (yet)
- The boards/sensors are all in different rooms of my house with good ventilation and away from any heat sources
- All the sensors have been connected with long enough wires to keep them away from the boards and other components
- I have tried with and without other components at the same time (e.g. DHT22, AHT20, APDS9960, LD2410C, SSD1306)
- Tested using both the "recommended" and "latest" versions of the Arduino framework
  - The component does not work on ESP-IDF (yet? 🤞)

I noticed there are newer versions of Bosch libraries, but I haven't had time to delve into the process of using them. 

I can only assume there is an issue with how the heater is being used. There are settings for the heater in the `bme680` component but not in the `bme68x_bsec2_i2c` component.

### Which version of ESPHome has the issue?

2024.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

ESP32-WROVER, Waveshare ESP32-S3-Zero,  Olimex ESP32-POE-ISO-EA 

### Component causing the issue

bme68x_bsec2_i2c

### Example YAML snippet

```yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${comment}
  area: ${area}

esp32:
[... 176 lines cut ...]
[18:45:56][C][bme68x_bsec2.sensor:093]:     Icon: 'mdi:test-tube'
[18:45:56][C][bme68x_bsec2.sensor:094]:   Breath VOC equivalent 'BME680 Breath VOC Equivalent'
[18:45:56][C][bme68x_bsec2.sensor:094]:     State Class: 'measurement'
[18:45:56][C][bme68x_bsec2.sensor:094]:     Unit of Measurement: 'ppm'
[18:45:56][C][bme68x_bsec2.sensor:094]:     Accuracy Decimals: 1
[18:45:56][C][bme68x_bsec2.sensor:094]:     Icon: 'mdi:test-tube'
[18:45:56][C][bme68x_bsec2.sensor:095]:   IAQ 'BME680 Indoor Air Quality'
[18:45:56][C][bme68x_bsec2.sensor:095]:     State Class: 'measurement'
[18:45:56][C][bme68x_bsec2.sensor:095]:     Unit of Measurement: 'IAQ'
[18:45:56][C][bme68x_bsec2.sensor:095]:     Accuracy Decimals: 0
[18:45:56][C][bme68x_bsec2.sensor:095]:     Icon: 'mdi:gauge'
[18:45:56][C][bme68x_bsec2.sensor:096]:   IAQ static 'BME680 IAQ Static'
[18:45:56][C][bme68x_bsec2.sensor:096]:     State Class: 'measurement'
[18:45:56][C][bme68x_bsec2.sensor:096]:     Unit of Measurement: 'IAQ'
[18:45:56][C][bme68x_bsec2.sensor:096]:     Accuracy Decimals: 0
[18:45:56][C][bme68x_bsec2.sensor:096]:     Icon: 'mdi:gauge'
[18:45:56][C][bme68x_bsec2.sensor:097]:   Numeric IAQ accuracy 'BME680 IAQ Accuracy'
[18:45:56][C][bme68x_bsec2.sensor:097]:     State Class: 'measurement'
[18:45:56][C][bme68x_bsec2.sensor:097]:     Unit of Measurement: ''
[18:45:56][C][bme68x_bsec2.sensor:097]:     Accuracy Decimals: 0
[18:45:56][C][bme68x_bsec2.sensor:097]:     Icon: 'mdi:checkbox-marked-circle-outline'
[18:45:56][C][bme68x_bsec2.sensor:100]:   IAQ accuracy 'BME680 IAQ Accuracy'
[18:45:56][C][bme68x_bsec2.sensor:100]:     Icon: 'mdi:checkbox-marked-circle-outline'
[18:45:56][E][component:082]:   Component bme68x_bsec2 is marked FAILED
```

schneich, 2024-11-06:
Hi @jb1228,

I have the same issue. When using ULP, the temperature sensor is obviously reading its measurements at the exact moment, the VOC sensor is heating up... no surprise, temperature readings are too high.

I just wanted to set the temperature sensor to LP, got the error and stumbled across this post. (I wanted to use a median-filter to cover the misreadings...)

```

# i2c-config für BME680-Sensor
i2c:
  sda: GPIO21
  scl: GPIO22
  scan: false
  id: bus_a

bme68x_bsec2_i2c:
  address: 0x77
  model: bme680
  operating_age: 28d
  sample_rate: ULP
  supply_voltage: 3.3V
  temperature_offset: 0
  state_save_interval: 6h

  - platform: bme68x_bsec2
    temperature:
      name: "BME680 Temperature"
      sample_rate: LP
      filters:
        - median:
            window_size: 100
            send_every: 20
            send_first_at: 1
    pressure:
      name: "BME680 Pressure"
      accuracy_decimals: 0
    humidity:
      name: "BME680 Humidity"
      accuracy_decimals: 0
    iaq:
      name: "BME680 IAQ"
      id: iaq
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
      accuracy_decimals: 0
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent original"
      id: voc_source
      internal: false

  - platform: template
    name: "BME680 Breath VOC Equivalent"
    id: BME680_voc_calc
    lambda: |-
      return ( (id(voc_source).state) * 1000) ;
    device_class: volatile_organic_compounds_parts
    state_class: measurement
    unit_of_measurement: ppb
    accuracy_decimals: 0
    update_interval: 5min
    icon: 'mdi:molecule'
```

luar123, 2024-11-07:
You can try the PR with:
```yaml
external_components:
  - source: github://pr#7728
    components: [ bme68x_bsec2 ]
```

schneich, 2024-11-07:
Hi @luar123,

thank you for your response. I will have to do some soldering on the weekend and then will give it a try.

For clarification: did you fix the original problem mentioned by @jb1228 (-> temp-, hum- and VOC-readings beiing useless) or did you fix the point, that one can't mix ULP and LP settings for the sample_rate?

luar123, 2024-11-07:
It should fix the readings. Sample rate mix is still not possible.

jb1228, 2024-11-08:
@luar123 Thank you! I will test the PR as soon as I can.

schneich, 2024-11-09:
> You can try the PR with:
> 
> ```yaml
> external_components:
>   - source: github://pr#7728
>     components: [ bme68x_bsec2 ]
> ```

Dear @luar123,

please take my apologies for the following question: how do I use this in my yaml? Do I add it like this?

```
...

external_components:
  - source: github://pr#7728
    components: [ bme68x_bsec2 ]

bme68x_bsec2_i2c:
  address: 0x77
  model: bme680
  operating_age: 28d
  sample_rate: ULP
  supply_voltage: 3.3V
  temperature_offset: 0
  state_save_interval: 6h

...

```

Well, I just did and get this log, when installing. Is this as intended?

```
INFO ESPHome 2024.10.3
INFO Reading configuration /config/wohnzimmer2.yaml...
INFO Detected timezone 'Europe/Zurich'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-wohnzimmer2 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing git+https://github.com/boschsensortec/Bosch-BSEC2-Library.git#v1.8.2610
INFO Installing git+https://github.com/boschsensortec/Bosch-BSEC2-Library.git#v1.8.2610
git version 2.39.5
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-lw_afitc'...
Note: switching to '24eccb80f1f1cbafcc7121f134b075fe9b37902c'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Library Manager: BSEC2 Software Library@1.8.2610+sha.24eccb8 has been installed!
INFO BSEC2 Software Library@1.8.2610+sha.24eccb8 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Library Manager: Installing BME68x Sensor library @ 1.1.40407
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: BME68x Sensor library@1.1.40407 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- BME68x Sensor library @ 1.1.40407
|-- bsec2 @ 1.8.2610+sha.24eccb8
```

luar123, 2024-11-09:
Yes, when you run it the first there should be:

> INFO ESPHome 2024.11.0-dev
> INFO Reading configuration airq_sensor.yaml...
> INFO Cloning https://github.com/esphome/esphome.git@pull/7728/head

You can check the .esphome/external_components/xyz/ folder (probably the newest) and run git log. Should show latest commit
> Date:   Wed Nov 6 22:43:00 2024 +0100
> Fix temperature for bme680

schneich, 2024-11-10:
The sensor has been running since yesterday night: temperature seems to be accurate. I have an infrared device for measuring temperatures and the sensor is 0.4°C too high, which is okay.

[image]

Humidity is a bit more difficult to say. The BME680 says 69%. I have two other (cheap) sensors (-> DHT11) indicating a humidity of 55% and 61%. I think, 69% is a bit too high.

[image]

Regarding VOC values: well, the values are not comparable to my AirthingsWave Plus values.

[image]

CO2 values are much closer between the BME680 and the AirthingsWave Plus. (I will put the sensors next to each other to get a better picture...)
[image]

The IAQ of the BME680 seems to be okayish compared to the value of the AirthingsWave Plus device. (VOC values of the BME680 improved in that regard that the values do not increase to 1'000'000 ppb anymore.)

schneich, 2024-11-11:
dear @luar123,

there is something downloaded, but I am not sure, if your PR is installed on my ESP32.

```
...

external_components:
  - source: github://pr#7728
    components: [ bme68x_bsec2 ]

bme68x_bsec2_i2c:
  address: 0x77
  model: bme680
  operating_age: 28d
  sample_rate: ULP
  supply_voltage: 3.3V
  temperature_offset: 0
  state_save_interval: 6h

...

```

Is it correct that the external code is labeled ```bme68x_bsec2``` and when I include the component, it is done with ```bme68x_bsec2_i2c```?

When trying ```git log```, it get this:
```
uadmin@ubuntu:~/Docker/HomeAssistant/esphome_config/.esphome/external_components/3f5ea120$ ls -lah
total 200K
drwxr-xr-x 10 root root 4.0K Nov  9 21:38 .
drwxr-xr-x  3 root root 4.0K Nov  9 21:38 ..
-rw-r--r--  1 root root 3.7K Nov  9 21:38 .clang-format
-rw-r--r--  1 root root 6.8K Nov  9 21:38 .clang-tidy
-rw-r--r--  1 root root 3.2K Nov  9 21:38 CODE_OF_CONDUCT.md
-rw-r--r--  1 root root  22K Nov  9 21:38 CODEOWNERS
-rw-r--r--  1 root root  656 Nov  9 21:38 CONTRIBUTING.md
-rw-r--r--  1 root root   34 Nov  9 21:38 .coveragerc
drwxr-xr-x  2 root root 4.0K Nov  9 21:38 .devcontainer
drwxr-xr-x  3 root root 4.0K Nov  9 21:38 docker
-rw-r--r--  1 root root 1.3K Nov  9 21:38 .dockerignore
-rw-r--r--  1 root root  398 Nov  9 21:38 .editorconfig
drwxr-xr-x  5 root root 4.0K Nov  9 21:38 esphome
-rw-r--r--  1 root root  988 Nov  9 21:38 .flake8
drwxr-xr-x  8 root root 4.0K Nov  9 21:38 .git
-rw-r--r--  1 root root   76 Nov  9 21:38 .gitattributes
drwxr-xr-x  5 root root 4.0K Nov  9 21:38 .github
-rw-r--r--  1 root root 1.6K Nov  9 21:38 .gitignore
-rw-r--r--  1 root root  36K Nov  9 21:38 LICENSE
-rw-r--r--  1 root root  181 Nov  9 21:38 MANIFEST.in
-rw-r--r--  1 root root  14K Nov  9 21:38 platformio.ini
-rw-r--r--  1 root root 1.5K Nov  9 21:38 .pre-commit-config.yaml
-rw-r--r--  1 root root 3.6K Nov  9 21:38 pyproject.toml
-rw-r--r--  1 root root  701 Nov  9 21:38 README.md
-rw-r--r--  1 root root  293 Nov  9 21:38 requirements_dev.txt
-rw-r--r--  1 root root   16 Nov  9 21:38 requirements_optional.txt
-rw-r--r--  1 root root  364 Nov  9 21:38 requirements_test.txt
-rw-r--r--  1 root root  781 Nov  9 21:38 requirements.txt
drwxr-xr-x  3 root root 4.0K Nov  9 21:38 script
-rw-r--r--  1 root root  753 Nov  9 21:38 sdkconfig.defaults
drwxr-xr-x  8 root root 4.0K Nov  9 21:38 tests
drwxr-xr-x  2 root root 4.0K Nov  9 21:38 .vscode
-rw-r--r--  1 root root  310 Nov  9 21:38 .yamllint
uadmin@ubuntu:~/Docker/HomeAssistant/esphome_config/.esphome/external_components/3f5ea120$ git log
fatal: detected dubious ownership in repository at '/home/uadmin/Docker/HomeAssistant/esphome_config/.esphome/external_components/3f5ea120'
To add an exception for this directory, call:

        git config --global --add safe.directory /home/uadmin/Docker/HomeAssistant/esphome_config/.esphome/external_components/3f5ea120
```

The logs if my ESP32 still start with:
```
INFO ESPHome 2024.10.3
INFO Reading configuration /config/wohnzimmer2.yaml...
```

luar123, 2024-11-11:
Internally it is the `bme68x_bsec2` component thats correct. And I think the PR is applied because otherwise you would see a much higher temperature offset. I am also not sure about humidity and don't have a reference for VOC/IAQ. I put some measurements in the PR discussion.

schneich, 2024-11-15:
Hi @luar123,

thank you. As far as I know, the humidity values are used in the calculation for AIQ? If I set an offset for humidity, will the offset be included in the IAQ-calculation?

luar123, 2024-11-17:
As far as I know there is no possibility to provide a humidity offset to bsec. If you use esphome sensor filters it won't influence any other sensors.

tabascoz, 2024-11-29:
Hi There, after updating to 2024.11.2 from 11.0 i see temperature values starting to drift out of the blue in my two bme680. I am suspecting the patch  https://github.com/esphome/esphome/pull/7728 might be the cupid (just a rouge guess). 

[image]

Is it happening with anyone else? 

here's the relevant code: 

```
bme68x_bsec2_i2c:
    model: bme680
    address: 0x76
    sample_rate: lp
    state_save_interval: 6h
    operating_age: 4d
```

luar123, 2024-11-29:
That looks really strange, but #7728 was already included in 2024.11.0.

tabascoz, 2024-11-30:
yep, something happened in the new versions. Now i reverted back to bme680_bsec. 
One possibility though is that the two esphomes that i have with this sensor might have not received the fw update from 2024.10. 

PS. changed the sensor from cheap ALI to one i purchased here in EU from Mauser.pt. and having the same results. 
Would be interested to hear if anyone is having the issue and happy to contribute.

luar123, 2024-11-30:
If you want to debug it please provide more information: sensor section of your yaml, logs, etc.
I have updated mine to latest dev version 8 hours ago: so far no issues. How long did it take for this to happen?

You could also try to fully erase esp32 and then flash again to remove the last saved state. Maybe that is not compatible. In 2024.11.0 also the bsec lib was updated.

tabascoz, 2024-12-01:
Hi, Will post tomorrow morning more info, the two devices are inside battery packs  and little hard to do maintenance (but possible). Would it be the same to trigger esphome factory reset? ,Or only physically erasing and reflashing ?

Long story short: I”ve been struggling with these bmes for about three months. Everything started when I changed the espdevices  from dev board to  a 2x  relay board and also the  physical location (from inside to outside the battery box). The  old boards had firmwares from early  2024, bsec2 lib was external from neffs and everything was fine. After the mass change I stated having issues with i2c comm, losing the bmes in random times., from some minutes to some days. Changed power source, cables, bme, tweaked i2c speed but they never were reliable again. 

Only downgrading to bsec it has brought them back to stability , BUT gas readings were not much reliable anymore - started having IAQ crazy high values out of the blue. 

With 2024.11.3, and bsec2, the i2c comm looked solid until my emergency batt temp alarm triggered with 50c ( almost had a heart attack lol) . It took about one hour to happen.

schneich, 2024-12-01:
I am also still struggling with the sensors.
Whenever doing something in our kitchen, this happens in our living room. Is this to be expected?
[image]

schneich, 2024-12-01:
[image]

tabascoz, 2024-12-01:
yep, something happened in the new versions. Now i reverted back to bme680_bsec. 
One possibility though is that the two esphomes that i have with this sensor might have not received the fw update from 2024.10. 

PS. changed the sensor from cheap ALI to one i purchased here in EU from Mauser.pt. and having the same results. 
Would be interested to hear if anyone is having the issue and happy to contribute. 

> Whenever doing something in our kitchen, this happens in our living room. Is this to be expected?

Guess what, tested same sensor with DC power and another esp32 and it's rock solid. So either are thw two esp board or mains noise at i2c bus... will do some other tests and revert

tabascoz, 2024-12-01:
I gave up (lol) 

Tried shield cable - no change... reverted back to serial 10cm flat. 
Connected computer, erased flash and flashed clean-build firmware from today's git. 
I also changed I2C pin to another ones that are on the opposite side of the board ( far from mains) 

In summary, now i was never more able to have the sensor working with bsec2. What is different from by desk's test set is: 

- environment:  mains are present in the board and it's on a Life4po battery pack
- espdevice: This is the device I'm using: https://www.aliexpress.com/item/1005007027676026.html?spm=a2g0o.order_list.order_list_main.5.35e91802w79hmQ
- Presence of DalyBMS component and serial connection active ( to be honest no issues with this connection at all, rock solid 9600bps) 

Although it could be some I2c noise, i tried changing cable and nothing really changed. Also it works well and reliable with Bsec library.

Here's the log with failed bsec2 communcation: 
[logs_dalybms_run.txt](https://github.com/user-attachments/files/17969223/logs_dalybms_run.txt)
code:
[dalybms-bsec2.yaml.txt](https://github.com/user-attachments/files/17969231/dalybms-bsec2.yaml.txt)

And here's the same version of logs with bsec ( not touching hardware ) 
[logs_dalybms_logs.txt](https://github.com/user-attachments/files/17969249/logs_dalybms_logs.txt)
[dalybms-bsec.yaml.txt](https://github.com/user-attachments/files/17969240/dalybms-bsec.yaml.txt)

I do APPRECIATE the support :) many thanks in advance.

tabascoz, 2024-12-01:
Think i found the culprit of bsec2 not connecting: 

`sensor:
  - platform: bme68x_bsec2
#  - platform: bme680_bsec
    temperature:
      name: "bme680 Temperature"
      id: bme680_temperature
      sample_rate: ulp
      filters:
      - timeout: 300s  # sent value will be NaN
      - clamp:
          min_value: 0
          max_value: 70
          ignore_out_of_range: true`

Sample rate "ulp" doesn't work but it works with bsec. Anyway this was just an attempt to play with the parameters and ended up "staying there". Now it is working with bsec2 and I'll give some hours to check if temp drift will happen again

schneich, 2024-12-01:
I am running ESPHome Dashboard in a little Docker Container. So, I went into CLI and renamed folders, to make sure, no old code is used.
```
root@esphome:/config/.esphome# mv bme68x_bsec2/ bme68x_bsec2_old
root@esphome:/config/.esphome/build# mv esphome-wohnzimmer2 esphome-wohnzimmer2_old
```

Then, I re-installed everything on my ESP32.

```powershell
INFO ESPHome 2024.11.2
INFO Reading configuration /config/wohnzimmer2.yaml...
INFO Detected timezone 'Europe/Zurich'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-wohnzimmer2 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.4
INFO Installing esphome/AsyncTCP-esphome @ 2.1.4
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
INFO AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
INFO ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing git+https://github.com/boschsensortec/Bosch-BSEC2-Library.git#v1.8.2610
INFO Installing git+https://github.com/boschsensortec/Bosch-BSEC2-Library.git#v1.8.2610
git version 2.39.5
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-q0pxriss'...
Note: switching to '24eccb80f1f1cbafcc7121f134b075fe9b37902c'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Library Manager: BSEC2 Software Library@1.8.2610+sha.24eccb8 has been installed!
INFO BSEC2 Software Library@1.8.2610+sha.24eccb8 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Library Manager: Installing BME68x Sensor library @ 1.1.40407
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: BME68x Sensor library@1.1.40407 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
[... 537 lines cut ...]
[16:02:10][D][sensor:094]: 'Keller VOC': Sending state 78.00000 ppb with 0 decimals of accuracy
[16:02:10][D][sensor:094]: 'Keller ambient light': Sending state 1.00000 lx with 0 decimals of accuracy
[16:02:10][I][ble_client:043]: [A4:DA:32:B9:5A:A5] Disabling BLE client.
[16:02:10][I][esp32_ble_client:084]: [0] [A4:DA:32:B9:5A:A5] Disconnecting.
[16:02:10][W][component:237]: Component esp32_ble took a long time for an operation (124 ms).
[16:02:10][W][component:238]: Components should block for at most 30 ms.
[16:02:10][D][esp32_ble_client:110]: [0] [A4:DA:32:B9:5A:A5] ESP_GATTC_CLOSE_EVT
[16:02:10][D][esp32_ble_client:172]: [0] [A4:DA:32:B9:5A:A5] ESP_GATTC_DISCONNECT_EVT, reason 22
[16:02:10][W][airthings_wave_base:027]: Disconnected!
[16:02:11][D][sensor:094]: 'BME680 IAQ': Sending state 53.75429 IAQ with 0 decimals of accuracy
[16:02:11][D][sensor:094]: 'BME680 CO2 Equivalent': Sending state 581.57037 ppm with 0 decimals of accuracy
[16:02:11][D][sensor:094]: 'BME680 Breath VOC Equivalent original': Sending state 0.60158 ppm with 1 decimals of accuracy
[16:02:11][D][sensor:094]: 'BME680 Pressure': Sending state 973.33417 hPa with 0 decimals of accuracy
[16:02:11][D][sensor:094]: 'BME680 Temperature': Sending state 23.35253 °C with 1 decimals of accuracy
[16:02:11][D][sensor:094]: 'BME680 Humidity': Sending state 49.21727 % with 0 decimals of accuracy
[16:02:50][D][sensor:094]: 'Uptime Sekunden': Sending state 342.64801 s with 0 decimals of accuracy
[16:03:50][D][sensor:094]: 'Uptime Sekunden': Sending state 402.64899 s with 0 decimals of accuracy
[16:04:50][D][sensor:094]: 'Uptime Sekunden': Sending state 462.64899 s with 0 decimals of accuracy
[16:04:59][D][sensor:094]: 'BME680 Breath VOC Equivalent': Sending state 601.58386 ppb with 0 decimals of accuracy
[16:05:26][D][sensor:094]: 'Keller Taupunkt': Sending state 7.91905 °C with 1 decimals of accuracy
[16:05:50][D][sensor:094]: 'Uptime Sekunden': Sending state 522.64899 s with 0 decimals of accuracy
``` 

So, I should be using `BME68x Sensor library@1.1.40407` and I am running on `ULP`.
@tabascoz are you saying, I should change to `LP`? Well, I don't need a 3sec update intervall and I would like to reduce network traffic...

tabascoz, 2024-12-01:
Hi Schneich 
Let me do some other tests.. think you can limit the rate at sensor object level. So far, i had the temp drift in the "untouched" battery.  The one that i did the most testing is stable so far. I will change the wiring, suspecting some issues interacting with other... Maybe the way the bsec2 code works is to dampen/average readings and one i2c noise could get a weird temp and induce the drift. Just another theory

luar123, 2024-12-01:
@tabascoz: You need to remove sample rate parameters from the sensors. With bsec2 you can only set all to LP or ULP. I think this is a bug in bsec2 and it is probably working with the old bsec component.

tabascoz, 2024-12-01:
thanks Luar... would be awesome if Bosch lib appropriately logged an error. You're right.  Awesome progress today
i think there was some noise in i2c bus that weren't affecting bsec - perhaps because of error checking routines. 

So far so good, no temp drifts with the code below:

```
i2c:
  frequency: 52khz
  sda: GPIO33
  scl: GPIO26

bme68x_bsec2_i2c:
    model: bme680
    address: 0x76
    sample_rate: lp
  # Default: 6h
    state_save_interval: 6h
    supply_voltage: 3.3V
    

sensor:
  - platform: bme68x_bsec2
#  - platform: bme680_bsec
    temperature:
      name: "bme680 Temperature"
      id: bme680_temperature
      sample_rate: lp
      filters:
      - timeout: 300s  # sent value will be NaN
      - clamp:
          min_value: 0
          max_value: 70
          ignore_out_of_range: true
    pressure:
      name: "BME680 Pressure"
      id: bme680_pressure
      sample_rate: lp
    humidity:
      name: "BME680 Relative Humidity"
      id: bme680_humidity
      sample_rate: lp

```

tabascoz, 2024-12-02:
Hi All,

@luar123  thanks for your support. I think it's good now. Both devices are solid, no spikes in any of the sensors and no temperature drift. 

I think main root cause was i2c bus noise inducted by the mains. This small X2relay esp32 board, powered by AC is a handy board, but i learned the inner raw of pins are next to AC and this explains why I tried to test with a 5cm strip cable and was still getting noise ( no matter what i tested). Deciding to change to pins physically located on the edge of the board allowed it to function. 

Second part of the puzzle was the cable itself. It was routed through a small hole to the inside of the battery box. Had to drill another hole 10cm away and reroute the cables. 

I used cat6 shielded UTP with sta+ground and sdl_ground, 3.3v+ground. Another issue as given some characteristics of this cable (unsure what) i hat to tweak i2c bus speed for it to work. Default frequency of 50khz wasn't working in the 1st battery at all.   52khz did the trick. The other battery (I implemented the fix later on) worked with 50khz without problems; 

Not to mention ulp vs lp parameters. I ended up letting them all set to lp; 

After this changes, all looks good, rock solid.  

@schneich I was wrong, bsec2 doesn't accept update_interval parameter. May it be something is worth it to implement in esphome code, given inability to tweak ulp/lp in bsec2 lib. 

Also perhaps worth it a mention in the documentation if someone finds crazy readings, would worth it to double check i2c noise. Also worth to suggest to try different bus speed or even different cable lengths. I read in many places that 1st one to blame would be Chinese low-quality or counterfeit sensors, so i thought it was the case... bought a new "quality" sensor  to find out the very same readings and issues. 

Ah was forgetting - erasing esp32 flash didn't change anything but great to learn that it could be something recommended in case of issues. 

Cheers,

schneich, 2024-12-14:
So, 2 weeks ago, I did the update to the newest version. I still see some spikes in VOC values. I assume, this must be some over-sensitive reaction from the sensor on something we do in the kitchen. (Perhaps its the water boiler, the coffee machine, the pan, the oven - or all of it...) After opening the windows, the values very quickly drop back to normal. Overall indication of IAQ-values seem to be reasonable.

[image]

Temperature is still ~1°C too high. As this is to be expected, I used the components-offset to compensate for that. Unfortunately, this results in higher humidity values. When using the offset, this is to be expected, but humidity values are too high. I used a sensor-filter to reduce the humidity value. Not nice, but good enough for my use cases.

Would I buy the BME680 again. No - definitely no.

As far as I am concerned, this issues can be closed.

sfields-forever, 2025-03-10:
If someone does not find the bugfix (like me), here it is: https://github.com/esphome/esphome/pull/7728

closed 2025-07-20 by jb1228