esphome/issues #6370: WIFI does not connect on remote speaker after last update

itnassol, 2024-10-23:
### The problem

I use round ESP boards based on the Onju Voice, been working beautifully up until the last update for ESP, The speakers now just get into a loop where they are constantly connecting to WIFI but not getting anywhere.

### Which version of ESPHome has the issue?

2024.10.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

itnassol, 2024-10-23:
So this seems to be an API issue, if I reboot any of the speakers it wont connect to the API... Any help would be great

randybb, 2024-10-23:
First format it right (use the code block as this is unreadable) and provide serial logs

itnassol, 2024-10-23:
Yep you're right my bad, rushed it (one of those days) Sorry all formatted now, Logs in the right place... Am fairly new to thi, what are serial logs? Thanks for giving me a prod.

randybb, 2024-10-23:
Finally it is readable :) Thanks.

But serial logs - these one are just telling that the dashboard it is not able to connect to address 192.168.68.52 as port 6053 is not open.

Since it is S3 you can just connect to one of USB ports and check logs when it boots. If it is boot loping we might figure out why it is doing so.

ShohninDmitriy, 2024-10-23:
WIFI logging has been broken again
[logs_jbd-bms_logs_usb.txt](https://github.com/user-attachments/files/17498134/logs_jbd-bms_logs_usb.txt)
[logs_jbd-bms_logs_wifi.txt](https://github.com/user-attachments/files/17498137/logs_jbd-bms_logs_wifi.txt)

randybb, 2024-10-23:
@ShohninDmitriy this is absolutely unrelated to this issue - as you see, your device is working just fine, your esphome dashboard is broken, probably something was updated and is not compatiblke anymore (rebuild addon/docker/venv).

ShohninDmitriy, 2024-10-24:
> @ShohninDmitriy this is absolutely unrelated to this issue - as you see, your device is working just fine, your esphome dashboard is broken, probably something was updated and is not compatiblke anymore (rebuild addon/docker/venv).

After reinstalling, alas, it does not work either

Jansen-AL, 2024-10-24:
Same problem with Respeaker lite kit ans Atom echo. No wifi connection anymore.

itnassol, 2024-10-24:
The latest update seems to have fixed the problem for me, thank you for all the advice.