esphome/issues #6371: Specifying (any) glyphs for a bitmap font leads to empty PROGMEM (due to EStr vs str)

cnschn, 2024-10-23:
### The problem

When I add any `glyphs` array to a bitmap font definition the generated code compiles and runs, but using the font in a `display` component does not draw anything on the page.

After some digging I noticed that the `PROGMEM` array in the generated `.cpp` code is just empty, and some further digging revealed that whenever I specify any glyphs explicitly, they end up as `EStr` instances in the code, while the default glyphs are just `str`.

Now, and I have no idea why, `getmask` returns a zero width mask for any `EStr`! Minimal test in an interactive shell:

```
[...]
>>> f = font.load_bitmap_font('<some bitmap font>')
>>> A = add_class_to_obj('A', ESPForceValue)
>>> type(A)
<class 'esphome.helpers.EStr'>
>>> f.getmask('A', mode='1').size
(10, 20)
>>> f.getmask(A, mode='1').size
(0, 20)
```

A quick fix would be to explicitly cast to `str` before calling Pillow's `getmask` function, i.e.

```
class BitmapFontWrapper:

    def getmask(self, glyph, **kwargs):
        return self.font.getmask(str(glyph), **kwargs)
```

But I'm not sure if this is the right place. I also tried figuring out _why_ it breaks, but it looks like the actual `getmask` implementation is in C, and I don't have any experience with how Python types are handled in native code.

### Which version of ESPHome has the issue?

Tested on dev and 2024.10.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

font

### Example YAML snippet

```yaml
# Works:
font:
  - file: "fonts/foo.pcf"
    id: foo

# Does not work:
font:
  - file: "fonts/foo.pcf"
    id: foo
    glyphs:
      - A
      - B
```

cnschn, 2024-10-24:
One additional observation: The font component docs say

> The extras section only supports OpenType/TrueType files

but that does not appear to be true, the `extras.glyphs` simply show the same issue described above. With the `getmask(str(glyph))` fix, adding additional glyphs via `extras` works perfectly fine for a bitmap font as well.

clydebarrow, 2024-10-29:
As a work-around, specify the glyphs as a string rather than as an array.
```
font:
  - file: MatrixChunky8X.bdf
    glyphs: "Text"
    id: chunky
```

closed 2024-10-29 by clydebarrow