esphome/issues #6373: update via notification/entity updates wrong device

532910, 2024-10-24:
### The problem

With several devices described as follows:

`config/esphome/socket-hall.yaml`:
```Yaml
packages:
  main: !include common/ShellyPlugS.yaml
wifi:
  use_address: socket-hall.mydomain.tld
```

`config/esphome/socket-kitchen.yaml`:
```Yaml
packages:
  main: !include common/ShellyPlugS.yaml
wifi:
  use_address: socket-kitchen.mydomain.tld
```

update via notification/entity updates only the last device:

 `update.socket_kitchen_firmware` works as expected: socket-kitchen gets updated
  `update.socket_hall_firmware` triggers update for socket-kitchen

`UPDATE ALL` button from esphome tab works fine and updates all devices.

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP32

### Board

not related

### Component causing the issue

not related

532910, 2025-02-22:
still an issue

ssieb, 2025-02-22:
You'll need to provide more details.  I don't understand what you're trying to do and what's going wrong.

532910, 2025-02-25:
Have you tried to replicate my setup? I believe it's described completely. What's exactly is unclear?

ssieb, 2025-02-25:
There's no way to replicate your setup.
What do you mean by "update via notification"?
What is `update.socket_hall_firmware`?
What do you see in the addon log?

532910, 2025-02-25:
> What do you mean by "update via notification"?

System -> Updates

>  What is `update.socket_hall_firmware`?

is an entity that should be enabled on device page to be able to update via System -> Updates

>  What do you see in the addon log?

no errors, it just updates wrong device

532910, 2025-02-25:
> There's no way to replicate your setup.

why so? just take two identical devices and create a common config you want