esphome/issues #6374: Log output doesn't work

Daemon256, 2024-10-24:
### The problem

INFO ESPHome 2024.10.2
INFO Reading configuration /config/esphome/1floor.yaml...
INFO Detected timezone 'Europe/Moscow'
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1018, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1005, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 481, in command_logs
    return show_logs(config, args, port)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 383, in show_logs
    from esphome.components.api.client import run_logs
  File "/esphome/esphome/components/api/client.py", line 8, in <module>
    from aioesphomeapi import APIClient
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/__init__.py", line 2, in <module>
    from .api_pb2 import (  # type: ignore[attr-defined] # noqa: F401
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/api_pb2.py", line 17, in <module>
    from . import api_options_pb2 as api__options__pb2
  File "/usr/local/lib/python3.11/dist-packages/aioesphomeapi/api_options_pb2.py", line 52, in <module>
    google_dot_protobuf_dot_descriptor__pb2.MethodOptions.RegisterExtension(needs_setup_connection)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'MethodOptions' has no attribute 'RegisterExtension'

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.10.3

### What platform are you using?

ESP8266

### Board

nodemcu

randybb, 2024-10-24:
The same issue was mentioned in a totally unrelated issue https://github.com/esphome/issues/issues/6370#issuecomment-2433427915 
Looks like you have protobuf lib that is not compatible https://github.com/mlflow/mlflow/issues/11811 / https://groups.google.com/g/protobuf/c/mlGp7ZzqKj8

Just tested 2023.11.1 (latest), 2024.10.2 (yours) and dev docker images and no issues.

Daemon256, 2024-10-27:
What needs to be done to fix it?

ssieb, 2024-10-27:
Try rebuilding your docker container.

Daemon256, 2024-10-28:
There is a caveat, even if I rebuild the current container, then the next time it boots it will automatically update itself to the previous version. Or the next time it is updated, it will also break.

SuperTuxinator, 2024-10-28:
The issue occured for me after updating ESPHome to 2024.10.2. It seems to be related to the update. I am not using docker. I directly installed HA on my device via an image. ESPHome I got from the "Add-ons". So not sure how I could rebuild the docker container.

rmayron, 2024-10-30:
Hi - any update or plans for resolution please?

many thanks

xrxca, 2024-11-01:
Docker esphome/esphome:stable (2024.10.2) had this issue for me, devices got updated properly but I got an error after every wireless install, the latest esphome/esphome:dev (2024.11.0-dev) fixes the issue.

kal40, 2024-11-04:
> Docker esphome/esphome:stable (2024.10.2) had this issue for me, devices got updated properly but I got an error after every wireless install, the latest esphome/esphome:dev (2024.11.0-dev) fixes the issue.

Confirmed. It works for me too with the dev version.

edoapra, 2024-11-04:
> Docker esphome/esphome:stable (2024.10.2) had this issue for me, devices got updated properly but I got an error after every wireless install, the latest esphome/esphome:dev (2024.11.0-dev) fixes the issue.

I believe it is due to be the [protobuf](https://pypi.org/project/protobuf/) version used. Since `RegisterExtension` is longer available in [protobuf](https://pypi.org/project/protobuf/) 5.28.3, [aioesphomeapi](https://github.com/esphome/aioesphomeapi) will throw the error reported at https://github.com/esphome/issues/issues/6374#issue-2611851548. Esphome/aioesphomeapi do work with protobuf version 3.20.3 where  `RegisterExtension`  is available.  
The  `pip list` command shows the protobuf versions used in the docker image discussed in this issue.
```
$ docker run -it --rm --entrypoint=pip  ghcr.io/esphome/esphome:2024.10.2  list | grep protob
protobuf                   5.28.3
$ $ docker run -it --rm --entrypoint=pip  ghcr.io/esphome/esphome:2024.11.0-dev20241103  list|grep protob
protobuf                   3.20.3
```
I have experienced the same error in the homeassistant Pre-Release [2024.11.0b2](https://github.com/home-assistant/core/releases/tag/2024.11.0b2) that does use the same 5.28.3 protobuf version   
https://github.com/home-assistant/core/pull/124936   
https://github.com/home-assistant/core/pull/124936/commits/15f0b8afd12036642e834cff113b1aeb7dc1c92d

Daemon256, 2024-11-08:
version 2024.10.3 fixed this issue

closed 2024-11-08 by Daemon256