esphome/issues #6378: ESP32 (S3) gets stuck with more than one neopixelbus?

pixelwave, 2024-10-25:
### The problem

I am trying to connect more than one addressable RGB strip to my ESP32 controller running ESPHome.

When I have **one neopixelbus** light instance (LED1) it runs well. When I try to add a second (LED2) the ESP32 does not boot properly.

The pins should not be the issue. I tried a number of different ones. Always the same behavior no matter which pin I use for the neopixelbus.

```
substitutions:
  # -> LED Output Definitions
  LED1_pin: GPIO17
  LED1_num_leds: "5"
  LED1_type: "GRB"
  LED1_variant: "WS2812"

   # CRASHES WHEN I ADD THIS: 
  LED2_pin: GPIO18
  LED2_num_leds: "5"
  LED2_type: "GRB"
  LED2_variant: "WS2812"

light:
  - platform: neopixelbus
    type: $LED1_type
    variant: $LED1_variant
    pin: $LED1_pin
    num_leds: $LED1_num_leds
    name: "LED OUT 1" 
    id: led1

   # CRASHES WHEN I ADD THIS: 
  - platform: neopixelbus
    type: $LED2_type
    variant: $LED2_variant
    pin: $LED2_pin
    num_leds: $LED2_num_leds
    name: "LED OUT 2" 
    id: led2
```

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Lilygo T-ETH ELite

### Component causing the issue

neopixelbus

### Example YAML snippet

```yaml
# PC Desk Controller
substitutions:
  # System Definitions
  comment: "Modular ESP32 Controller (LED, Switches & More)"
  device_name: base-desk-controller
  friendly_name: (BASE) Desk Controller
  # Controller Variables

  # -> LED Output Definitions
  LED1_pin: GPIO17
  LED1_num_leds: "5"
  LED1_type: "GRB"
  LED1_variant: "WS2812"

  LED2_pin: GPIO18
  LED2_num_leds: "5"
  LED2_type: "GRB"
  LED2_variant: "WS2812"

  # -> Desk Switch (UP/DOWN)
  SW_UP_pin: GPIO08
  SW_DOWN_pin: GPIO16

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${comment}

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# T-ETH-ELite
ethernet:
  type: W5500
  clk_pin: GPIO48
  mosi_pin: GPIO21
  miso_pin: GPIO47
  cs_pin: GPIO45
  interrupt_pin: GPIO14
  #reset_pin: GPIOXX
  
light:
  - platform: neopixelbus
    type: $LED1_type
    variant: $LED1_variant
    pin: $LED1_pin
    num_leds: $LED1_num_leds
    name: "LED OUT 1" 
    id: led1

  - platform: neopixelbus
    type: $LED2_type
    variant: $LED2_variant
    pin: $LED2_pin
    num_leds: $LED2_num_leds
    name: "LED OUT 2" 
    id: led2

switch:
  # Desk Switches (UP/DOWN)
  - platform: gpio
    pin:
      number: $SW_DOWN_pin
    id: switch_down
    name: "Desk - DOWN"
    icon: "mdi:arrow-expand-down"
  - platform: gpio
    pin:
      number: $SW_UP_pin
    id: switch_up
    name: "Desk - UP"
    icon: "mdi:arrow-expand-up"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.9.2
INFO Reading configuration /config/base-desk-controller.yaml...
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing base-desk-controller (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.6
|-- NeoPixelBus @ 2.7.3
Compiling .pioenvs/base-desk-controller/src/main.cpp.o
Linking .pioenvs/base-desk-controller/firmware.elf
RAM:   [=         ]   9.2% (used 30064 bytes from 327680 bytes)
Flash: [====      ]  35.3% (used 647297 bytes from 1835008 bytes)
Building .pioenvs/base-desk-controller/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/base-desk-controller/firmware.bin"], [".pioenvs/base-desk-controller/firmware.elf"])
Wrote 0xae200 bytes to file /config/.esphome/build/base-desk-controller/.pioenvs/base-desk-controller/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/base-desk-controller/firmware.bin"], [".pioenvs/base-desk-controller/firmware.elf"])
========================= [SUCCESS] Took 5.10 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.126
INFO Uploading /config/.esphome/build/base-desk-controller/.pioenvs/base-desk-controller/firmware.bin (647680 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 4.03 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.178.126 using esphome API
WARNING Can't connect to ESPHome API for base-desk-controller @ 192.168.178.126: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.126', port=6053))]: [Errno 111] Connect call failed ('192.168.178.126', 6053) (SocketAPIError)
INFO Trying to connect to base-desk-controller @ 192.168.178.126 in the background
```

pixelwave, 2024-10-25:
I figured when I add **method: BIT_BANG** it does compile correctly. So it seems there is an issue with the channels when using the default method?

Is that the best method for ESP32-S3 or should I use some RMT/Channel combination?

```
light:
  - platform: neopixelbus
    type: $LED1_type
    variant: $LED1_variant
    pin: $LED1_pin
    num_leds: $LED1_num_leds
    name: "LED OUT 1" 
    id: led1
    method: BIT_BANG

  - platform: neopixelbus
    type: $LED2_type
    variant: $LED2_variant
    pin: $LED2_pin
    num_leds: $LED2_num_leds
    name: "LED OUT 2" 
    id: led2
    method: BIT_BANG
```

ssieb, 2025-02-23:
You should be using the rmt LED component.

pixelwave, 2025-02-23:
I ended up using `fastled_clockless`. Runs stable for months now.

```
light:

  # LED Light Strip 1
  - platform: fastled_clockless
    chipset: $LED1_variant
    pin: $LED1_pin
    num_leds: $LED1_num_leds
    rgb_order: $LED1_type
    name: $LED1_name
    id: light1
    on_turn_on:
      - switch.turn_on: led_relay
    on_turn_off:
      - if:
          condition:
            and:
              - light.is_off: light1
              - light.is_off: light2
          then:
            - switch.turn_off: led_relay

  # LED Light Strip 2
  - platform: fastled_clockless
    chipset: $LED2_variant
    pin: $LED2_pin
    num_leds: $LED2_num_leds
    rgb_order: $LED2_type
    name: $LED2_name
    id: light2
    on_turn_on:
      - switch.turn_on: led_relay
    on_turn_off:
      - if:
          condition:
            and:
              - light.is_off: light1
              - light.is_off: light2
          then:
            - switch.turn_off: led_relay
```

closed 2025-02-23 by pixelwave

ssieb, 2025-02-23:
That is probably the worst option, but up to you.

pixelwave, 2025-02-23:
> That is probably the worst option, but up to you.

In what way? Colors, brightness etc. do work as intended.