esphome/issues #6379: Voice assistant via i2s microphone crashes ESP32-S3 after ESPHome 2024.10.1 update

JustEnoughDucks, 2024-10-25:
### The problem

I am using a ESP32-S3, a PCM5102 as a DAC to my Yamaha receiver, and a MSM261S4030H0R MEMs microphone connected via an interconnect PCB.

The goal (that works) is a simultaneous device that can have full voice assistant satellite support while also being a media player for home assistant/spotify/etc... while also controlling my yamaha rx496-RDS receiver via IR. So far it can do all of this, I just have to put in automatic source switching and better volume control than just home assistant buttons if possible.

After the ESPHome 2024.10.1 update, the esp crashes after enabling the microphone and starting the voice assistant pipeline, thus breaking voice assistant support. I am currently testing with a wake button instead of a wakeword for easy testing, so I don't know if it is a change specifically with button-activated support or not.

| Version  | Voice Assistant |Audio playback (Music Assistant) |
| ------------- | ------------- | ------------- |
| 2024.8.0  | ✅  | ✅ |
| 2024.9.0  | ✅ | ✅ |
| 2024.10.0  | ✅  | ✅ |
| 2024.10.1  | ❌  | ✅ |
| 2024.10.2  | ❌  | ✅ |

### Which version of ESPHome has the issue?

>=2024.10.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

ESP32-S3 

### Component causing the issue

voiceassistant

### Example YAML snippet

```yaml
esp32:
  board: seeed_xiao_esp32s3
  variant: esp32s3
  framework:
    type: arduino
    version: latest
    platform_version: 6.3.2

...

voice_assistant:
  id: va
  microphone: yamaha_mic
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  media_player: media_out
  
  #on_start:

  #on_tts_start:

  on_tts_end:
    - media_player.play_media: !lambda return x;

  on_end:
    - delay: 2s
[... 142 lines cut ...]
[10:23:56][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:23:56][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:23:56][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:23:56][C][api:140]: API Server:
[10:23:56][C][api:141]:   Address: 192.168.1.123:6053
[10:23:56][C][api:143]:   Using noise encryption: YES
[10:23:56][C][audio:225]: Audio:
[10:23:56][C][audio:247]:   External DAC channels: 2
[10:23:56][C][audio:248]:   I2S DOUT Pin: 8
[10:23:56][D][api:103]: Accepted 192.168.1.100
[10:23:56][V][api.connection:1428]: Hello from client: 'Home Assistant 2024.10.3' | 192.168.1.100 | API Version 1.10
[10:23:56][D][api.connection:1446]: Home Assistant 2024.10.3 (192.168.1.100): Connected successfully
[10:24:02][D][binary_sensor:036]: 'PCB_Switch': Sending state ON
[10:24:03][D][binary_sensor:036]: 'PCB_Switch': Sending state OFF
[10:24:32][D][binary_sensor:036]: 'PCB_Switch': Sending state ON
[10:24:33][D][binary_sensor:036]: 'PCB_Switch': Sending state OFF
[10:24:33][D][voice_assistant:510]: State changed from IDLE to START_MICROPHONE
[10:24:33][D][voice_assistant:516]: Desired state set to START_PIPELINE
[10:24:33][D][voice_assistant:222]: Starting Microphone
WARNING esphome-speaker-dongle @ 192.168.1.123: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-speaker-dongle @ 192.168.1.123
WARNING Disconnected from API
INFO Successfully connected to esphome-speaker-dongle @ 192.168.1.123 in 0.002s
INFO Successful handshake with esphome-speaker-dongle @ 192.168.1.123 in 0.056s
```

BigBobbas, 2024-10-29:
This is resolved in 2024.10.2

JustEnoughDucks, 2024-10-30:
@BigBobbas this is not resolved. I have posted that it is tested in 2024.10.2 and it is the exact same behavior as 2024.10.1.

closed 2025-08-13 by JustEnoughDucks