esphome/issues #6382: wb3s and cb3s with tuya MCUI thermostats stopping to responce

Bl00d-B0b, 2024-10-26:
### The problem

I have several beca thermostats converted to ESP home:
https://devices.esphome.io/devices/Beca-Thermostat
with wb3s and cb3s boards. thermostats are hungs aproximatly in 12-48 hours after hardware power on!.

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.10.4

### What platform are you using?

BK72XX

### Board

wb3s,cbs

### Component causing the issue

tuya

### Example YAML snippet

```yaml
https://devices.esphome.io/devices/Beca-Thermostat
```

### Anything in the logs that might be useful for us?

```txt
no, system just hung and no response. 
only power on/off helps
```

### Additional information

n/a

ssieb, 2024-10-26:
Can you ping them?
Can you leave a log running so that you can see the last things before it stops?

Bl00d-B0b, 2024-10-26:
It's a strange behavior. sometime I can ping, and even I can do software restart and/or firmware update, but still tuya do not initialized, only  power on/off helps. but sometime I even can't ping, device is dyed. on LCD screen I see the latest status, and buttons are not working.
If you give me instructions, then possible I can get some events, etc... or I can open TCP port to the device.

Bl00d-B0b, 2024-10-26:
same problem was in all legacy versions from 2024. 
https://github.com/esphome/feature-requests/issues/2603

ssieb, 2024-10-26:
It sounds like the Tuya MCU is getting stuck.  I don't know what esphome can do about that.  You aren't being clear either.  Is it esphome that's not responding or the thermostat interface?

Bl00d-B0b, 2024-10-29:
OK. let's do one by one. I have 6 devices. and those are hooked randomly. I assume it's a TUYA MCU, as device is pingable, and even I can update firmware, This is a log after firmware update on hunged device:
10:38:52][D][debug:035]: ESPHome version 2024.10.2
[10:38:52][D][debug:039]: Free Heap Size: 90744 bytes
[10:38:52][D][debug:016]: LibreTiny Version: 1.5.1
[10:38:52][D][debug:017]: Chip: BK7231T (671a) @ 120 MHz
[10:38:52][D][debug:018]: Chip ID: 0x23A61D
[10:38:52][D][debug:019]: Board: wb3s
[10:38:52][D][debug:020]: Flash: 2048 KiB / RAM: 256 KiB
[10:38:52][D][debug:021]: Reset Reason: SW Reboot
**[10:39:02][E][tuya:464]: Initialization failed at init_state 0**
[10:39:14][D][sensor:093]: 'WiFi Signal': Sending state -50.00000 dBm with 0 decimals of accuracy
[10:39:14][D][sensor:093]: 'Uptime': Sending state 41.15200 s with 0 decimals of accuracy
[10:39:17][E][tuya:464]: Initialization failed at init_state 0
[10:39:32][E][tuya:464]: Initialization failed at init_state 0
[10:39:33][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[10:39:33][D][lt.preferences:104]: Saving 1 preferences to flash...
[10:39:33][D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[10:39:47][E][tuya:464]: Initialization failed at init_state 0
[10:40:02][E][tuya:464]: Initialization failed at init_state 0
[10:40:14][D][sensor:093]: 'WiFi Signal': Sending state -50.00000 dBm with 0 decimals of accuracy
[10:40:14][D][sensor:093]: 'Uptime': Sending state 101.15200 s with 0 decimals of accuracy
[10:40:17][E][tuya:464]: Initialization failed at init_state 0
[10:40:32][E][tuya:464]: Initialization failed at init_state 0

Bl00d-B0b, 2024-10-29:
only hardware power off and helps!

Bl00d-B0b, 2024-10-29:
how to enable debug on uart? is it possible to get logs via wifi? then i can enable and i need to wait until next failure after hardware reset.

Bl00d-B0b, 2024-10-29:
looks like i can add this into config of uart?

  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

Bl00d-B0b, 2024-10-29:
this **debug** does not show any send/receive :(

ssieb, 2024-10-29:
All you should have is the `debug:` line.  It should at least be showing what esphome is sending.

ssieb, 2024-10-30:
esphome is trying, but the device isn't answering.

Bl00d-B0b, 2024-10-30:
Yes, that's I can see. but as i mentioned, even software reboot does not help. only power off/on. 
Original tuya firmware works without any issues for several months without reboot. any help, how it can be fixed.

P.S. As I mentioned also in previous messages, internal temperature sensor need to be calibrated with -7 degree, instead -3 setup from manufacturer by default.

My assumption is tuya is actually need to be requested with less intervals ???? I don't know how to fix it, but device can't be used, if it's require hardware power on/off :(

ssieb, 2024-10-30:
I have no answers for you.  The Tuya MCU is getting stuck and that's why only a power cycle will fix it.  I don't know why it would be getting stuck.  If you can record logs and see what's happening when it gets stuck, maybe that would help.  As for the temperature thing, obviously this device is badly designed.  Maybe the original firmware used some sort of sleep mode, but esphome doesn't support anything like that at this point.

Bl00d-B0b, 2024-10-31:
i know, but all logs I already provided.

ssieb, 2024-10-31:
I mean logs from when it happens, not after.
What "tuya team"?

Bl00d-B0b, 2024-11-08:
Is it possible to fix it? as device is not useful. power on/of every day it's not a solution.
maybe it's possible to have hardware reset on rs port? as software reboot does not help.

Bl00d-B0b, 2024-11-12:
So, i have a question, on debug I see:
[16:27:15][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:27:23][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[16:27:23][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
[16:27:32][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:27:38][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[16:27:38][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
[16:27:48][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:27:53][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[16:27:53][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
[16:28:00][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:28:01][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:28:08][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[16:28:08][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
[16:28:16][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38
[16:28:23][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[16:28:23][D][uart_debug:114]: <<< 55:AA:03:00:00:01:01:04
[16:28:32][D][uart_debug:114]: <<< 55:AA:03:07:00:08:18:02:00:04:00:00:00:AF:DE:55:AA:03:07:00:08:65:02:00:04:00:00:00:00:7C:55:AA:03:07:00:05:24:04:00:01:01:38

As I can see tuya MCU send command every 5 second. I assume this is a problem of heating internal temperature sensor. Is it possible to tune this parameter? For example to setup it in some settings value.

szupi-ipuzs, 2024-11-13:
@Bl00d-B0b ,
There's really no way to stop (or slow down) the reading of temperature because it's not reading, but receiving. TuyaMCU sends it when it sees fit. But, juding from the logs you have shown, it doesn't seem often enough to cause high CPU usage and overheating.

I think what you first need to do is to gather the whole log (with uart_debug enabled) starting from boot, up until the problem (tuyamcu hangup) is noticable. Now, this could be a problem if it's going to take several days. In this case, I recommend to forward the logs to a syslog server.
Here's how:

1. Setup a syslog server somewhere on your network.
2. Add support for syslog to your device, by adding this in your yaml:

```yaml
external_components:
  - source: github://TheStaticTurtle/esphome_syslog
    components: [syslog]

syslog:
    id: syslog_id
    ip_address: "(put here the ip of your syslog server)"
    port: (put here the port of your syslog server, eg. 514)
```

Compile and flash the device - the esphome logger output should be automatically forwarded to the syslog server.
Please post the log here. This way we might be able to actually pinpoint the moment after which the TuyaMCU starts to misbehave and see what might've caused this.

szupi-ipuzs, 2024-11-13:
Regarding the additional "temperature fix" you seem to need - I don't think this is related to your problem. I think the original tuya firmware (that you replaced with esphome) did some additional calibration in the software. And you can also do that in esphome - just use the "offset" filter on the temperature sensor, like so:
```yaml
sensor:
  - platform: "tuya"
    name: "Temperature"
...
    filters:
      - multiply: 0.5
      - offset: -6
```

It is possible of course that each of your device will require a different offset. You can sort it out eg. by using esphome packages and substitutions.

szupi-ipuzs, 2024-11-14:
Ok, i've just read your other post about this where you mention that the temperature offset changes in runtime. So please disregard my comment above, it really DOES look like an overheating problem.
Now, if it is beken chip that overheats (although I doubt it), setting esphome to deep sleep in intervals might help.
Unfortunately on BK72XX platform, this is not yet supported, but can be done by using my [deep_sleep_libretiny](https://github.com/szupi-ipuzs/esphome-external-components) external component.

Still, the full log might show something interesting, so please try to get it first.

Bl00d-B0b, 2024-11-14:
I start syslog on my 6 devices in debug mode with uart. let's see... but it's to much messages. maybe I need to filter?

Bl00d-B0b, 2024-11-15:
takes several hours to catch a hung.
2024-11-15T01:02:37.693+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: <<< 55:AA:03:07:00:08:03:02:00:04:00:00:00:29:43:55:AA:03:07:00:08:66:02:00:04:00:00:00:00:7D" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.249+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'WiFi Signal': Sending state -59.00000 dBm with 0 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.577+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 41" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.578+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 20.50000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.582+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.583+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.584+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.584+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.585+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.586+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.590+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 102 update to 0" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.591+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature (external)': Sending state 0.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:38.702+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: <<< 55:AA:03:07:00:08:03:02:00:04:00:00:00:29:43:55:AA:03:07:00:08:66:02:00:04:00:00:00:00:7D" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.572+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 41" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.574+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 20.50000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.577+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.578+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.579+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.580+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.581+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.582+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:39.589+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 102 update to 0" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
[... cut ...]
2024-11-15T01:10:25.662+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:10:33.041+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Uptime': Sending state 15341.17969 s with 0 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426

Looks like 024-11-15T01:02:39.698+02:00 last success packet from tuya.

Bl00d-B0b, 2024-11-15:
Hello,

Could you please help me with syntax to add deep sleep and yuat component from this project.
https://github.com/szupi-ipuzs/esphome-external-components

I try several ways, but have syntax error. I want to use deep sleep & yuat

Ed

Bl00d-B0b, 2024-11-15:
`2024-11-15T01:02:35.704+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: <<< 55:AA:03:07:00:08:03:02:00:04:00:00:00:29:43:55:AA:03:07:00:08:66:02:00:04:00:00:00:00:7D" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.574+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 41" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.575+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 20.50000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.578+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.578+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.579+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.580+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.580+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.582+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.588+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 102 update to 0" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.600+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature (external)': Sending state 0.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:36.697+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: <<< 55:AA:03:07:00:08:03:02:00:04:00:00:00:29:43:55:AA:03:07:00:08:66:02:00:04:00:00:00:00:7D" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.576+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 41" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.576+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 20.50000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.576+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.579+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.579+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.580+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.581+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.582+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.588+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][tuya:343]: Datapoint 102 update to 0" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:02:37.590+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][sensor:093]: 'Temperature (external)': Sending state 0.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
[... cut ...] rk_udp TRANSPORT=rfc5426
2024-11-15T01:25:10.669+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-15T01:25:25.665+02:00 corridor-thermostat HOST=corridor-thermostat HOST_FROM=corridor-thermostat MESSAGE="[D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
`

szupi-ipuzs, 2024-11-16:
Hmm, there's not much that log can tell us, I was kind of expecting esphome sending something unusual before this happens. But no, nothing unusual.
It seems the last status bytes sent by TuyaMCU are incomplete (55:AA:03 is the header), almost as if it died in the middle of sending it.

So as @ssieb said, there's really nothing you can do from esphome side. The problem is in TuyaMCU. Maybe the original Tuya firmware somehow masked it.

Could you please say what models of beca thermostat do you have? And what was the original Tuya firmware version before you flashed it with esphome? Just to see if other user observe the same problem with them.

szupi-ipuzs, 2024-11-16:
One thing occured to me: maybe this MCU requires the time to be sent. Please add this to your yaml and see if it helps:

```yaml
time:
  - platform: homeassistant
    id: homeassistant_time

tuya:
  time_id: homeassistant_time
```

Bl00d-B0b, 2024-11-16:
I use NTP to set time. And time is set on my devices.

Bl00d-B0b, 2024-11-16:
I use those devices. I do not remember tuya version, as install with tuya cutter esphome and do not have backup

I just found this on AliExpress: 
€56.39 | MOES Tuya WiFi Smart Thermostat, Electric Floor Heating Water/Gas Boiler Temperature Remote Controller for Google Home, Alexa
https://a.aliexpress.com/_EubTuLB

ssieb, 2024-11-17:
I wonder if esphome starts sending before the MCU has finished sending.  I thought something was added to avoid that situation because of a case similar to this, but maybe not, or maybe it's not working for some reason.

Bl00d-B0b, 2024-11-17:
Yes. That's strange. I assume should be protection from such situation.

ssieb, 2024-11-17:
If that is the problem, then it's really a bug in the MCU software.  We're just trying to work around that and it's not possible to guarantee it.

Bl00d-B0b, 2024-11-17:
I can try to catch one more time with syslog, ans let's see, if it's a same issues in log?

Bl00d-B0b, 2024-11-18:
`2024-11-17T23:00:50.085+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 21.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.086+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.089+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.090+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.090+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.091+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:50.203+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][uart_debug:114]: <<< 08:66:02:00:04:00:00:00:00:7D:55:AA:03:07:00:08:03:02:00:04:00:00:00:2A:44:55:AA:03:07:00" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.082+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 42" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.083+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 21.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.086+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.089+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.090+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.090+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:419]:   Current Temperature: 21.00°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.091+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:425]:   Target Temperature: 20.50°C" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:51.203+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][uart_debug:114]: <<< 08:66:02:00:04:00:00:00:00:7D:55:AA:03:07:00:08:03:02:00:04:00:00:00:2A:44:55:AA:03:07:00" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.087+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][tuya:343]: Datapoint 3 update to 42" MSGFORMAT=rfc5424 PROGRAM=tuya SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.087+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][sensor:093]: 'Temperature': Sending state 21.00000 °C with 1 decimals of accuracy" MSGFORMAT=rfc5424 PROGRAM=sensor SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:396]: 'Thermostat' - Sending state:" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:399]:   Mode: HEAT" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:401]:   Action: IDLE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:00:52.088+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][climate:410]:   Preset: NONE" MSGFORMAT=rfc5424 PROGRAM=climate SOURCE=s_network_udp [... cut ...] ANSPORT=rfc5426
2024-11-17T23:01:08.062+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
2024-11-17T23:01:08.182+02:00 wardrobe-thermostat HOST=wardrobe-thermostat HOST_FROM=wardrobe-thermostat MESSAGE="[D][uart_debug:114]: <<< 08:66:02:00:04:00:00:00:00:7D" MSGFORMAT=rfc5424 PROGRAM=uart_debug SOURCE=s_network_udp TRANSPORT=rfc5426
`

wernerhp, 2024-12-01:
I have a similar issue with the `cb2s`.
I'm running this ESPHome firmware https://github.com/wernerhp/esphome/tree/main/bneta-smart-plug

[image]

Attached are my syslogs.  `03:xx` seems more appropriate since I'm in GMT+2 and the syslogs are int UTC, but there was a gap at `05:xx` UTC (that doesn't show up in Home Assistant as it doesn't appear to have caused a reboot), so the logs are included.

[messages-kv-05:xx.log](https://github.com/user-attachments/files/17968294/messages-kv-05.xx.log)

[messages-kv-03:xx.log](https://github.com/user-attachments/files/17968293/messages-kv-03.xx.log)

Bl00d-B0b, 2025-01-14:
Maybe its possible to have a fix? like implement a send/receive lock. just to be sure do not write to tuya when it's receives?

Bl00d-B0b, 2025-01-29:
Just want to add comment about temperature. some of the devices start to be overheated. even i see it on LCD display. tuya chip is very hot. as I mentioned before, I must use -7 degree to align temperature with esphome... tuya version works fine with -3.

but now one device is overheated and shows me temperature +26 insted +20, even my correction is set to -9... 

is any way i can use deep sleep with this chipset?

Ed

Bl00d-B0b, 2025-02-04:
Maybe there is a possibility to off/on uart in esphome.

I assume this one helps.