esphome/issues #6385: API to HomeAssistant is not working after upgrade of HA to 2024.10.4

piknew, 2024-10-27:
### The problem

I am using ESPHome ESP32 board to get data by UART from JK-BMS. Then it is used on HA Dashboard. The case is that on version 2024.10.3 of HA - everything is working correctly. However - after upgrade of HA Core to 2024.10.4 - it seems that HA is getting sensor data only "on init", then refresh is not done. Values of sensors are not changing, HA reports that "last update" is long and longer time ago (at the time of writing - example is 12 minutes - last restart of HA):

[image]

I verified this by restoring HA to 2024.10.3 and it was working perfectly fine. After upgrade - again - the problem.

At the same time I am sure that data is read by ESP32 correctly. So, queries by REST WebServer API is updated. In log - I see updated values. WebServer web page is also "alive". So, it seems that problem is with queries/events by HA API?

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32-IDF

### Component causing the issue

api

### Example YAML snippet

```yaml
substitutions:
  name: esp-uart-jk-bms
  friendly_name: ESPHome UART JK BMS
  device_model: JK-BD4A8S4P
  ssid_name: "***** SECRET *****"
  ssid_password: "***** SECRET *****"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  project:
    name: esphome.${name}
    version: "1.0"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

wifi:
#  use_address: 192.168.10.252
  ssid: ${ssid_name}
  password: ${ssid_password}
  manual_ip:
    static_ip: 192.168.10.252
    subnet: 255.255.255.0
    gateway: 192.168.10.1
    dns1: 192.168.10.1
  ap:
    ssid: ${name}
    password: ${ssid_password}

api:

web_server:
  port: 80

logger:
  baud_rate: 0
  level: DEBUG

ota:
  - platform: esphome
    id: ota_esphome

uart:
  - id: uart_0
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

jk_modbus:
  - id: modbus0
    uart_id: uart_0
    rx_timeout: 50ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus0
    update_interval: 5s

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "Balancing"
    balancing_switch:
      name: "Balancing Switch"
    charging:
      name: "Charging"
    discharging:
      name: "Discharging"
    dedicated_charger_switch:
      name: "Dedicated Charger Switch"
    online_status:
      name: "Online Status"

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "Minimum Cell Voltage"
    max_cell_voltage:
      name: "Maximum Cell Voltage"
    min_voltage_cell:
      name: "Minimum Voltage Cell"
    max_voltage_cell:
      name: "Maximum Voltage Cell"
    delta_cell_voltage:
      name: "Delta Cell Voltage"
    average_cell_voltage:
      name: "Average Cell Voltage"
    cell_voltage_1:
      name: "Cell 1 Voltage"
    cell_voltage_2:
      name: "Cell 2 Voltage"
    cell_voltage_3:
      name: "Cell 3 Voltage"
    cell_voltage_4:
      name: "Cell 4 Voltage"
    cell_voltage_5:
      name: "Cell 5 Voltage"
    cell_voltage_6:
      name: "Cell 6 Voltage"
    cell_voltage_7:
      name: "Cell 7 Voltage"
    cell_voltage_8:
      name: "Cell 8 Voltage"
    power_tube_temperature:
      name: "Power Tube Temperature"
    temperature_sensor_1:
      name: "Temperature Sensor 1"
    temperature_sensor_2:
      name: "Temperature Sensor 2"
    total_voltage:
      name: "Total Voltage"
    current:
      name: "Current"
    power:
      name: "Power"
    charging_power:
      name: "Charging Power"
    discharging_power:
      name: "Discharging Power"
    capacity_remaining:
      name: "Capacity Remaining"
    capacity_remaining_derived:
      name: "Capacity Remaining Derived"
    temperature_sensors:
      name: "Temperature Sensors"
    charging_cycles:
      name: "Charging Cycles"
    total_charging_cycle_capacity:
      name: "Total Charging Cycle Capacity"
    battery_strings:
      name: "Battery Strings"
    errors_bitmask:
      name: "Errors Bitmask"
    operation_mode_bitmask:
      name: "Operation Mode Bitmask"
    total_voltage_overvoltage_protection:
      name: "Total Voltage Overvoltage Protection"
    total_voltage_undervoltage_protection:
      name: "Total Voltage Undervoltage Protection"
    cell_voltage_overvoltage_protection:
      name: "Cell Voltage Overvoltage Protection"
    cell_voltage_overvoltage_recovery:
      name: "Cell Voltage Overvoltage Recovery"
    cell_voltage_overvoltage_delay:
      name: "Cell Voltage Overvoltage Delay"
    cell_voltage_undervoltage_protection:
      name: "Cell Voltage Undervoltage Protection"
    cell_voltage_undervoltage_recovery:
      name: "Cell Voltage Undervoltage Recovery"
    cell_voltage_undervoltage_delay:
      name: "Cell Voltage Undervoltage Delay"
    cell_pressure_difference_protection:
      name: "Cell Pressure Difference Protection"
    discharging_overcurrent_protection:
      name: "Discharging Overcurrent Protection"
    discharging_overcurrent_delay:
      name: "Discharging Overcurrent Delay"
    charging_overcurrent_protection:
      name: "Charging Overcurrent Protection"
    charging_overcurrent_delay:
      name: "Charging Overcurrent Delay"
    balance_starting_voltage:
      name: "Balance Starting Voltage"
    balance_opening_pressure_difference:
      name: "Balance Opening Pressure Difference"
    power_tube_temperature_protection:
      name: "Power Tube Temperature Protection"
    power_tube_temperature_recovery:
      name: "Power Tube Temperature Recovery"
    temperature_sensor_temperature_protection:
      name: "Temperature Sensor Temperature Protection"
    temperature_sensor_temperature_recovery:
      name: "Temperature Sensor Temperature Recovery"
    temperature_sensor_temperature_difference_protection:
      name: "Temperature Sensor Temperature Difference Protection"
    charging_high_temperature_protection:
      name: "Charging High Temperature Protection"
    discharging_high_temperature_protection:
      name: "Discharging High Temperature Protection"
    charging_low_temperature_protection:
      name: "Charging Low Temperature Protection"
    charging_low_temperature_recovery:
      name: "Charging Low Temperature Recovery"
    discharging_low_temperature_protection:
      name: "Discharging Low Temperature Protection"
    discharging_low_temperature_recovery:
      name: "Discharging Low Temperature Recovery"
    total_battery_capacity_setting:
      name: "Total Battery Capacity Setting"
    current_calibration:
      name: "Current Calibration"
    device_address:
      name: "Device Address"
    sleep_wait_time:
      name: "Sleep Wait Time"
    alarm_low_volume:
      name: "Alarm Low Volume"
    total_runtime:
      name: "Total Runtime"
    actual_battery_capacity:
      name: "Actual Battery Capacity"

switch:
  - platform: jk_bms
    charging:
      name: "Charging"
    discharging:
      name: "Discharging"

text_sensor:
  - platform: template
    name: "Device Model"
    lambda: return std::string("${device_model}");
  - platform: jk_bms
    errors:
      name: "Errors"
    operation_mode:
      name: "Operation Mode"
    battery_type:
      name: "Battery Type"
    device_type:
      id: device_type
      name: "Device Type"
    software_version:
      name: "Software Version"
    manufacturer:
      name: "Serial Number"
      filters:
        - lambda: |-
            std::size_t i = x.find(id(device_type).state,0);
            if( i != std::string::npos ) {
              return x.substr(i + id(device_type).state.length());
            }
            return x;
```

piknew, 2024-10-27:
I am not sure, but it seems that update of SOME values are not working. So, it maybe related to esphome-jk-bms (to be confirmed): https://github.com/syssi/esphome-jk-bms/issues/621

dani-rodr, 2024-10-28:
Also experiencing some instability in my esp based presence sensors too. Checked their logs when they are supposed to publish values but is not updating in home assistant

piknew, 2024-10-28:
As suggested in linked issue (comment: https://github.com/syssi/esphome-jk-bms/issues/621#issuecomment-2440067005) it may be related to:

https://github.com/esphome/esphome/pull/7321
https://github.com/esphome/esphome/pull/7547

I have changed to MQTT - so far all values are updated correctly in HA, without delay.

alex-dokienko, 2024-11-17:
got same issues for same project (using jk-bms), but in my case I already using mqtt and values not updated for me, I can see via mqtt explorer topics updated, but on dashboard it shows unknown, when restart ha values refreshed and then same, for another time.

rtdegraaf, 2024-11-19:
I'm having the same issue with a modbus implementation. The device is reporting values every minute when I open the console, but Home Assistant is not always updating the sensor values. When I open the console, Home Asisstant does update the values every singe minute.

No update for a hour:
[image]
Opening console and instant update after sending values from ESP board:
[image]

pgwipeout, 2024-11-20:
Hello there, I suspect that a few issues here are all experiencing the same problem.
(Symptom: Missing or lossed updates to HomeAssistant when using the HA Platform, ESP crashing)

I started having problems recently with my Emporia Vue2 dying after running for a short while. It would start with dropped updates, then no updates, and eventually it would stop responding to pings.
I've since enabled debugging, and I've noticed the following message starts off rarely than becomes increasingly more common until the symptoms manifest:
`[21:53:32][V][api.connection:1298]: Cannot send message because of TCP buffer space`

I suspect something has changed in the HA API that causes it to block messages more often, and it looks to me like the blocked messages aren't getting cleaned up leading to a memory leak:
https://github.com/esphome/esphome/blob/dev/esphome/components/api/api_connection.cpp#L1534

I recommend enabling verbose logging, as this message should be higher than verbose but it isn't. You'll probably want to reduce your component logging back to debug, like mine is below:
```
# Enable logging
logger:
  level: VERBOSE
  logs:
    # by default, every reading will be printed to the UART, which is very slow
    # This will disable printing the readings but keep other helpful messages
    sensor: INFO
    wifi: VERBOSE
    i2c: DEBUG
    emporia_vue: DEBUG
    ```

rtdegraaf, 2024-11-21:
hi @pgwipeout, thanks for your suggestion. I have edited my logging like your example and it seems like it has a positive effect. Updates are coming in every minute now and a graph over the last day seems a whole lot smoother:
[image]
I have edited the logging at the green line, since then the graph line is smooth. Thanks!

pgwipeout, 2024-11-26:
So I've isolated this down even further. It's definitely a bug with the HA API, it seems it's possible to create a race condition between the API and the TCP / Ethernet stack that leads to a soft lock of the board.
In my case, the amount of data being sent combined with a less than ideal wireless signal strength (-59db, bringing it to -57db was enough to fix it) caused the message queue to back up to the point where the board wasn't pingable. I've also made adjustments to my sensors to space out the messages. The heap free goes nuts whenever this starts backing up, you can see here what it looks like before and after the changes:
[image]

I don't think this is new to 2024.10, as I reverted all the way back to 2024.02 and it was still happening. I suspect this bug has been there for a long time (if not from the beginning) and a recent change to Home Assistant itself is making it easier to trigger. 

Looking at the API and the way it handles TCP, it's a miracle this isn't happening more often. It's essentially:
- Check if queue has space, if not, fail.
- Send messages until queue is full, then fail.

There's no cleanup or recovery that I can see. The following changes are necessary:
- Make the message queue full a warn or an info, instead of a verbose.
- Institute some sort of cleanup when a message fails.
- Institute some basic protection for the TCP queue to prevent the board from soft locking when this happens.

rtdegraaf, 2024-11-26:
Nice found @pgwipeout. I can confirm that the problem has been going on for some time and that my ESP board also has a weak Wifi signal. Somewhere around -79db most of the time.

Since changing my log levels, the data is still coming in at a smooth rate. :)