esphome/issues #6389: Deprecated ADC Functions

boezguer, 2024-10-29:
### The problem

When compiling an ESPHome project that uses the ADC component on an ESP32C6 board with the ESP-IDF framework, multiple warnings and errors are generated due to deprecated functions. ESPHome's ADC component still references outdated ADC driver functions, which were deprecated and replaced by new ones in the latest ESP-IDF versions.

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

ESP32C6 

### Example YAML snippet

```yaml
esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: 6.8.1

sensor: 
  - platform: adc
    pin: GPIO2
    name: "Water Level"
    update_interval: 1s
    unit_of_measurement: "%"

output:
  - platform: ledc
    id: pwm_output
    pin: GPIO18
    frequency: 200000 Hz
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/adc/adc_sensor.h:10,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:11:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
src/esphome/components/adc/adc_sensor.h:41:21: error: 'adc2_channel_t' has not been declared
   41 |   void set_channel2(adc2_channel_t channel) {
      |                     ^~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h:80:3: error: 'adc2_channel_t' does not name a type; did you mean 'adc_channel_t'?
   80 |   adc2_channel_t channel2_{ADC2_CHANNEL_MAX};
      |   ^~~~~~~~~~~~~~
      |   adc_channel_t
src/esphome/components/adc/adc_sensor.h:83:3: error: 'esp_adc_cal_characteristics_t' does not name a type
   83 |   esp_adc_cal_characteristics_t cal_characteristics_[SOC_ADC_ATTEN_NUM] = {};
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.h: In member function 'void esphome::adc::ADCSensor::set_channel1(adc1_channel_t)':
src/esphome/components/adc/adc_sensor.h:39:11: error: 'class esphome::adc::ADCSensor' has no member named 'channel2_'; did you mean 'channel1_'?
   39 |     this->channel2_ = ADC2_CHANNEL_MAX;
[... 107 lines cut ...]
  227 |           adc2_config_channel_atten(this->channel2_, ADC_ATTEN_DB_0);
      |                                           ^~~~~~~~~
      |                                           channel1_
src/esphome/components/adc/adc_sensor.cpp:228:30: error: 'class esphome::adc::ADCSensor' has no member named 'channel2_'; did you mean 'channel1_'?
  228 |           adc2_get_raw(this->channel2_, ADC_WIDTH_MAX_SOC_BITS, &raw0);
      |                              ^~~~~~~~~
      |                              channel1_
src/esphome/components/adc/adc_sensor.cpp:238:60: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  238 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                                                            ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:238:19: error: 'esp_adc_cal_raw_to_voltage' was not declared in this scope
  238 |   uint32_t mv12 = esp_adc_cal_raw_to_voltage(raw12, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_12_COMPAT]);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:239:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  239 |   uint32_t mv6 = esp_adc_cal_raw_to_voltage(raw6, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_6]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:240:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  240 |   uint32_t mv2 = esp_adc_cal_raw_to_voltage(raw2, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_2_5]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/adc/adc_sensor.cpp:241:58: error: 'class esphome::adc::ADCSensor' has no member named 'cal_characteristics_'
  241 |   uint32_t mv0 = esp_adc_cal_raw_to_voltage(raw0, &this->cal_characteristics_[(int32_t) ADC_ATTEN_DB_0]);
      |                                                          ^~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/c6-pwm/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
========================= [FAILED] Took 103.58 seconds =========================
```

edwardtfn, 2024-12-09:
Hey @boezguer, do you mind giving PR#7942 a try and let me know the results?
```yaml
external_components:
  - source: github://pr#7940
    components:
      - adc
```

obucklin, 2025-02-04:
> Hey [@boezguer](https://github.com/boezguer), do you mind giving PR#7942 a try and let me know the results?
> 
> external_components:
>   - source: github://pr#7942
>     refresh: 30s
>     components:
>       - adc

Hello,

Not @boezguer but I was having the same problem. I tried this workaround and it compiled fine. Unfortunately, the ADC is not returning accurate values. Instead of a measured 2.0v (LiIon cell with 2:1 voltage divider), I get a continuous 1.086. There is no variation on this value, which makes me suspect it is invalid, although I guess there may be smoothing going on on the backend with a 5s sample rate. 

my yaml is here:

`esphome:
  name: seeed_balcony_controller
  friendly_name: balcony_seeed

esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: "xxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Seed-Balcony Fallback Hotspot"
    password: "xxxxxxx"

sensor:
  - platform: adc
    pin: GPIO0
    name: "battery"
    update_interval: 5s`

Please let me know if you have any suggestions, and I'd be happy to test any changes.
Thanks