esphome/issues #6391: Compile error pipsolar on ESP32

luboshynek, 2024-10-29:
### The problem

Cannot compile pipsolar (https://esphome.io/components/pipsolar.html)

### Which version of ESPHome has the issue?

2024.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.10.4

### What platform are you using?

ESP32

### Board

esp32-s2-saola-1

### Component causing the issue

pipsolar

### Example YAML snippet

```yaml
substitutions:
  name: esphome-invertor2
  friendly_name: ESPHome invertor2
  tx_pin: GPIO17
  rx_pin: GPIO18

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s2.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:
uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

pipsolar:
  -  uart_id: uart_0
     id: inverter2
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.10.2
INFO Reading configuration /config/esphome/esphome-invertotr2.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-invertor2 (board: esp32-s2-saola-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.4
Compiling .pioenvs/esphome-invertor2/src/main.cpp.o
Linking .pioenvs/esphome-invertor2/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pioenvs/esphome-invertor2/src/esphome/core/helpers.cpp.o:(.literal._ZN7esphome7crc16beEPKhtttbb+0x0): undefined reference to `crc16_be'
/data/cache/platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: .pioenvs/esphome-invertor2/src/esphome/core/helpers.cpp.o: in function `esphome::crc16be(unsigned char const*, unsigned short, unsigned short, unsigned short, bool, bool)':
/data/build/esphome-invertor2/src/esphome/core/helpers.cpp:151: undefined reference to `crc16_be'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esphome-invertor2/firmware.elf] Error 1
```

ssieb, 2024-10-30:
It turns out that the S2 library doesn't have any of the big-endian CRC functions.

ssieb, 2024-10-30:
At least for now, anything other than an S2.  Maybe it will be resolved when we move to a newer idf version.

closed 2025-03-24 by kbx81